Koenkk/zigbee2mqtt #713: "platform": "mqtt_json" is deprecated, replace with "schema":"json"

trwolff04, 2018-12-16:
I'm receiving this message under errors after upgrading to the latest version of hassio.  
------------------
Log Details (WARNING)
Sun Dec 16 2018 14:53:45 GMT-0500 (Eastern Standard Time)

"platform": "mqtt_json" is deprecated, replace with "schema":"json"
-----------------------

Any ideas?

danpowell88, 2018-12-16:
Have you updated to the latest, looks like this was fixed a day ago

Koenkk, 2018-12-17:
Has been fixed in https://github.com/Koenkk/zigbee2mqtt/pull/689 (which is in 1.0.0). These are probably old MQTT messages which are still retained by the broker.

closed 2018-12-17 by Koenkk

BergAutomations, 2018-12-17:
Hi. I have the same issue. I have tried to delete mosquitto.db and homeassistant database. Is any retained kept in zigbee2mqtt?

trwolff04, 2018-12-17:
I am on version 0.2.0 ... I dont see an update available for hass.io yet. I have auto update checked.

stratplayer, 2019-02-10:
I had to subscribe to the following

`homeassistant/sensor/+/+/config
homeassistant/light/+/+/config
homeassistant/switch/+/+/config
`
Then sent an empty message to the RETAINED via MQTT.fx and got rid of all but 1 of my errors in the log