Koenkk/zigbee2mqtt #714: Feedback of RGB control does not match the command

mcsSolutions, 2018-12-17:
I am testing control of Sengled Element color bulb.  I can send MQTT message to change brightness and the acknowledge for brighness sets it to a different value than was set.  A simple example is setting brightness to 0.  The bulb does go to what appears as OFF so the response is like as expected.  The MQTT message back shows a brightness level of 78 eventually.

5:23:51 PM Received zigbeeWA/0xb0ce18140001de1b={"state":"ON","brightness":91,"color":{"x":0.4,"y":0.205}}
5:30:04 PM Received zigbeeWA/0xb0ce18140001de1b/set={"brightness":0}
5:30:04 PM Received zigbeeWA/0xb0ce18140001de1b={"state":"ON","brightness":91,"color":{"x":0.4,"y":0.205}}
5:30:05 PM Received zigbeeWA/0xb0ce18140001de1b={"state":"ON","brightness":78,"color":{"x":0.4,"y":0.205}}

When looking at the zigbee2mqtt console the following is seen where the command to 0 brightness is received, but the brightness returned in 91 and then 78.  The x and y values of the color space are unchanged (as expected).  If I get a MQTT message of 78 for brightness then I do not expect the bulb to have 0 brightness as is experienced by looking at the bulb and as commanded.

  zigbee2mqtt:info 2018-12-16 17:30:02 Zigbee publish to '0xb0ce18140001de1b', genLevelCtrl - moveToLevelWithOnOff - {"level":0,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-12-16 17:30:02 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":91,"color":{"x":0.4,"y":0.205}}'
  zigbee2mqtt:info 2018-12-16 17:30:02 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":78,"color":{"x":0.4,"y":0.205}}'

mcsSolutions, 2018-12-17:
Further data that described the behavior...

Control of brightness level produces the incorrect MQTT response for the brightness level unless the brightness control command is repeated after the bulb does reach the commanded brightness.
In the test case below brightness started at 255 with bright bulb.  Command to reduce to 10, but MQTT response was 232.  This implies that the response came before the bulb reached the commanded level.
This was followed by a command back up to 255 and then the MQTT response had no relationship to what the bulb was doing. 
Same illogical MQTT response when commanded back to 0 with the two MQTT messages showing an increasing rather than decreasing level
When the same brightness level was commanded twice then the (2nd) MQTT response was correct

  zigbee2mqtt:info 2018-12-17 12:32:30 Zigbee publish to '0xb0ce18140001de1b', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-12-17 12:32:31 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":223,"color":{"x":0.135,"y":0.04}}'
  zigbee2mqtt:info 2018-12-17 12:32:31 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":255,"color":{"x":0.135,"y":0.04}}'
 
 zigbee2mqtt:info 2018-12-17 12:40:39 Zigbee publish to '0xb0ce18140001de1b', genLevelCtrl - moveToLevelWithOnOff - {"level":10,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-12-17 12:40:39 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":255,"color":{"x":0.135,"y":0.04}}'
  zigbee2mqtt:info 2018-12-17 12:40:40 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":232,"color":{"x":0.135,"y":0.04}}'

  zigbee2mqtt:info 2018-12-17 12:43:38 Zigbee publish to '0xb0ce18140001de1b', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-12-17 12:43:38 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":232,"color":{"x":0.135,"y":0.04}}'
  zigbee2mqtt:info 2018-12-17 12:43:38 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":33,"color":{"x":0.135,"y":0.04}}'

  zigbee2mqtt:info 2018-12-17 12:44:19 Zigbee publish to '0xb0ce18140001de1b', genLevelCtrl - moveToLevelWithOnOff - {"level":0,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-12-17 12:44:19 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":33,"color":{"x":0.135,"y":0.04}}'
  zigbee2mqtt:info 2018-12-17 12:44:19 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":233,"color":{"x":0.135,"y":0.04}}'

  zigbee2mqtt:info 2018-12-17 12:46:39 Zigbee publish to '0xb0ce18140001de1b', genLevelCtrl - moveToLevelWithOnOff - {"level":0,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-12-17 12:46:39 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":233,"color":{"x":0.135,"y":0.04}}'
  zigbee2mqtt:info 2018-12-17 12:46:39 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":0,"color":{"x":0.135,"y":0.04}}'

Koenkk, 2018-12-18:
Can you provide more logging? I'm especially interested in the `set` topics.

Koenkk, 2018-12-19:
Somehow the bulb doesn't publish it's new state. Can you try replacing the current device with (in `devices.js`:

```js
    {
        zigbeeModel: ['E11-N1EA'],
        model: 'E11-N1EA',
        vendor: 'Sengled',
        description: 'Element Plus Color (A19)',
        supports: generic.light_onoff_brightness_colortemp_colorxy().supports,
        fromZigbee: generic.light_onoff_brightness_colortemp_colorxy().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colortemp_colorxy().toZigbee,
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.report('genLevelCtrl', 'currentLevel', 0, 10, 0, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

mcsSolutions, 2018-12-19:
Produces the desired end result with two negative side effects
1. Every 10 seconds No Converter Available message cid genLevelCtrl ...
2. All intermediate brightness levels are reported via MQTT as bulb transitions between original and final.  This behavior does not present an issue, but just excess traffic.

Debug at https://pastebin.com/HUUv6p9A

Non-Debug below

9:29:56 No converter available for 'E11-N1EA' with cid 'genLevelCtrl', type 'attReport' and data '{"cid":"genLevelCtrl","data":{"currentLevel":64}}'
  zigbee2mqtt:warn 2018-12-19 09:29:56 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:info 2018-12-19 09:29:56 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":64,"color":{"x":0.6,"y":0.3},"color_temp":250}'
  zigbee2mqtt:warn 2018-12-19 09:30:06 No converter available for 'E11-N1EA' with cid 'genLevelCtrl', type 'attReport' and data '{"cid":"genLevelCtrl","data":{"currentLevel":64}}'
  zigbee2mqtt:warn 2018-12-19 09:30:06 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:info 2018-12-19 09:30:11 Zigbee publish to '0xb0ce18140001de1b', genLevelCtrl - moveToLevelWithOnOff - {"level":100,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-12-19 09:30:11 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":64,"color":{"x":0.6,"y":0.3},"color_temp":250}'
  zigbee2mqtt:warn 2018-12-19 09:30:11 No converter available for 'E11-N1EA' with cid 'genLevelCtrl', type 'attReport' and data '{"cid":"genLevelCtrl","data":{"currentLevel":65}}'
  zigbee2mqtt:warn 2018-12-19 09:30:11 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:info 2018-12-19 09:30:11 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":65,"color":{"x":0.6,"y":0.3},"color_temp":250}'
  zigbee2mqtt:warn 2018-12-19 09:30:11 No converter available for 'E11-N1EA' with cid 'genLevelCtrl', type 'attReport' and data '{"cid":"genLevelCtrl","data":{"currentLevel":66}}'
  zigbee2mqtt:warn 2018-12-19 09:30:11 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:warn 2018-12-19 09:30:11 No converter available for 'E11-N1EA' with cid 'genLevelCtrl', type 'attReport' and data '{"cid":"genLevelCtrl","data":{"currentLevel":67}}'
  zigbee2mqtt:warn 2018-12-19 09:30:11 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:info 2018-12-19 09:30:11 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":66,"color":{"x":0.6,"y":0.3},"color_temp":250}'
  zigbee2mqtt:info 2018-12-19 09:30:11 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":67,"color":{"x":0.6,"y":0.3},"color_temp":250}'
  zigbee2mqtt:warn 2018-12-19 09:30:11 No converter available for 'E11-N1EA' with cid 'genLevelCtrl', type 'attReport' and data '{"cid":"genLevelCtrl","data":{"currentLevel":76}}'
  zigbee2mqtt:warn 2018-12-19 09:30:11 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:info 2018-12-19 09:30:11 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":76,"color":{"x":0.6,"y":0.3},"color_temp":250}'
  zigbee2mqtt:warn 2018-12-19 09:30:11 No converter available for 'E11-N1EA' with cid 'genLevelCtrl', type 'attReport' and data '{"cid":"genLevelCtrl","data":{"currentLevel":77}}'
  zigbee2mqtt:warn 2018-12-19 09:30:11 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:warn 2018-12-19 09:30:11 No converter available for 'E11-N1EA' with cid 'genLevelCtrl', type 'attReport' and data '{"cid":"genLevelCtrl","data":{"currentLevel":78}}'
  zigbee2mqtt:warn 2018-12-19 09:30:11 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:info 2018-12-19 09:30:11 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":77,"color":{"x":0.6,"y":0.3},"color_temp":250}'
  zigbee2mqtt:info 2018-12-19 09:30:11 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":78,"color":{"x":0.6,"y":0.3},"color_temp":250}'
  zigbee2mqtt:warn 2018-12-19 09:30:12 No converter available for 'E11-N1EA' with cid 'genLevelCtrl', type 'attReport' and data '{"cid":"genLevelCtrl","data":{"currentLevel":84}}'
  zigbee2mqtt:warn 2018-12-19 09:30:12 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:info 2018-12-19 09:30:12 MQTT publish, topic: 'zigbeeWA/0xb0ce18140001de1b', payload: '{"state":"ON","brightness":84,"color":{"x":0.6,"y":0.3},"color_temp":250}'
  zigbee2mqtt:warn 2018-12-19 09:30:12 No converter available for 'E11-N1EA' with cid 'genLevelCtrl', type 'attReport' and data '{"cid":"genLevelCtrl","data":{"currentLevel":85}}'
  zigbee2mqtt:warn 2018-12-19 09:30:12 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:warn 2018-12-19 09:30:12 No converter available for 'E11-N1EA' with cid 'genLevelCtrl', type 'attReport' and data '{"cid":"genLevelCtrl","data":{"currentLevel":86}}'
  zigbee2mqtt:warn 2018-12-19 09:30:12 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
[... cut ...]
  zigbee2mqtt:warn 2018-12-19 09:30:22 No converter available for 'E11-N1EA' with cid 'genLevelCtrl', type 'attReport' and data '{"cid":"genLevelCtrl","data":{"currentLevel":100}}'
  zigbee2mqtt:warn 2018-12-19 09:30:22 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:warn 2018-12-19 09:30:32 No converter available for 'E11-N1EA' with cid 'genLevelCtrl', type 'attReport' and data '{"cid":"genLevelCtrl","data":{"currentLevel":100}}'
  zigbee2mqtt:warn 2018-12-19 09:30:32 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.

Koenkk, 2018-12-20:
Let's reduce the spamming, can you try with: 
```js
    {
        zigbeeModel: ['E11-N1EA'],
        model: 'E11-N1EA',
        vendor: 'Sengled',
        description: 'Element Plus Color (A19)',
        supports: generic.light_onoff_brightness_colortemp_colorxy().supports,
        fromZigbee: generic.light_onoff_brightness_colortemp_colorxy().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colortemp_colorxy().toZigbee,
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.report('genLevelCtrl', 'currentLevel', 2, 3600, 1, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

closed 2019-02-25 by stale[bot]