Koenkk/zigbee2mqtt #715: Requesting support for Sylvania

YamiFrankc, 2018-12-17:
```
  zigbee2mqtt:warn 2018-12-17 01:43:52 Device with modelID 'LIGHTIFY Gardenspot RGB' is not supported.
  zigbee2mqtt:warn 2018-12-17 01:43:52 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:warn 2018-12-17 01:43:52 Device with modelID 'LIGHTIFY Gardenspot RGB' is not supported.
  zigbee2mqtt:warn 2018-12-17 01:43:52 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
```

Koenkk, 2018-12-17:
Please check: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices_on_hassio.html

YamiFrankc, 2018-12-17:
I played a little bit with it and now i get this in the log, from what I gather it may be working
```
  zigbee2mqtt:info 2018-12-17 04:41:33 0x00158d0002726443 (0x00158d0002726443): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
  zigbee2mqtt:info 2018-12-17 04:41:33 0x7cb03eaa00a7242e (0x7cb03eaa00a7242e): E19240LV - Sylvania RGB garden ligths  (Router)
```
But now im confused about auto discovery in hassio, the guide says to edit the `lib/extension/homeassistant.js` file and add the new device to it, under the "mapping section", but not only I dont have that file inside my docker container, it will also revert any changes when the add-on is restarted.

What can I do now?

Koenkk, 2018-12-17:
If you are on =< zigbee2mqtt 1.0.0 it's under `lib/homeassistant.js`.

dbmet, 2018-12-20:
Guess I’ll try and add these later as well cause I have a couple sets of them.

YamiFrankc, 2018-12-21:
I made it work using the same config as "OSRAM Smart+ Gardenpole RGBW", it works alright using that, but the color temperature doesn't seem to work.
The only thing I haven't been able to do is the self discovery thing because I have been busy at work

Koenkk, 2018-12-24:
Does this bulb support color temperature? 

The discovery is something I will add once this device is supported.

dbmet, 2018-12-24:
I have mine connected to deconz conbee and I'm only sending Hue and Saturation no color temp.

 I'll try color temp later when its dark out.

> Does this bulb support color temperature?
> 
> The discovery is something I will add once this device is supported.

YamiFrankc, 2018-12-25:
Doesnt seem to support it, has there is no change when I try to change it using the slider.

Anyway, after some days of it working, now I get this in the log when I try to change its state, and the change doesnt apply: 

```
  zigbee2mqtt:info 2018-12-25 01:10:38 Zigbee publish to '0x7cb03eaa00a7242e', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 2018-12-25 01:10:38 Zigbee publish to '0x7cb03eaa00a7242e', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
```

I tried restarting zigbee2mqtt and rebooting the garden lights but the problem doesnt go away.

dbmet, 2018-12-25:
Agreed Color Temp does nothing with the slider on Deconz either..

Going to attempt to add these later after I figure out what put where in the config files.. haha

> Doesnt seem to support it, has there is no change when I try to change it using the slider.
>
> Anyway, after some days of it working, now I get this in the log when I try to change its state, and the [...]

Koenkk, 2018-12-26:
@dbmet please see https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices_on_hassio.html

closed 2019-02-24 by Koenkk