Koenkk/zigbee2mqtt #716: how to sniff traffic

danpowell88, 2018-12-17:
Is there an easy way to sniff the traffic of a phillips hue bulb when its connected to the original hub.

I have a feeling with the new firmware update theres something being sent/received to the bulb to tell it what colour it should show when it first turns on.

I've updated my bulbs and the feature worked when on the hub but now when they are connected to to zigbee2mqtt the behavior is missing.

Koenkk, 2018-12-17:
This explains it quite well: https://blog.jimmo.id.au/zigbee-packet-capture-724cf9346b47

mcsSolutions, 2018-12-18:
I am also trying to sniff the traffic for RGBGenie support,.  I followed the instructions in the link and all went well until the svn command.  I needed to install svn.  Then later on the make I received the following where the error is around char array digs.  Looking at the source I did not see anything wrong, but then I am not a C programmer.  Is there a compiled binary for the CC2531 to avoid this issue or is there some additional thing I need to do to get it to work.  I started with a new RPi image for DietDI (Debian Stretch Lite) and just followed the Contiki instructions.

rm -f bfd-tmp.h
touch stmp-bfd-h
/bin/bash ./libtool --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I. -I. -I./../include  -DHAVE_asxxxx_vec  -DBINDIR='"/usr/local/bin"'  -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -g -O2 -MT archures.lo -MD -MP -MF .deps/archures.Tpo -c -o archures.lo -DDEFAULT_VECTOR=asxxxx_vec -DSELECT_VECS='&asxxxx_vec' -DSELECT_ARCHITECTURES='&bfd_z80_arch' ./archures.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I. -I. -I./../include -DHAVE_asxxxx_vec -DBINDIR=\"/usr/local/bin\" -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -g -O2 -MT archures.lo -MD -MP -MF .deps/archures.Tpo -c -DDEFAULT_VECTOR=asxxxx_vec "-DSELECT_VECS=&asxxxx_vec" "-DSELECT_ARCHITECTURES=&bfd_z80_arch" ./archures.c -o archures.o
mv -f .deps/archures.Tpo .deps/archures.Plo
/bin/bash ./libtool --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I. -I. -I./../include  -DHAVE_asxxxx_vec  -DBINDIR='"/usr/local/bin"'  -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -g -O2 -MT asxxxx.lo -MD -MP -MF .deps/asxxxx.Tpo -c -o asxxxx.lo asxxxx.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I. -I. -I./../include -DHAVE_asxxxx_vec -DBINDIR=\"/usr/local/bin\" -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -g -O2 -MT asxxxx.lo -MD -MP -MF .deps/asxxxx.Tpo -c asxxxx.c -o asxxxx.o
asxxxx.c:169:19: error: ‘digs’ defined but not used [-Werror=unused-const-variable=]
 static const char digs[] = "0123456789ABCDEF";
                   ^~~~
cc1: all warnings being treated as errors
Makefile:1312: recipe for target 'asxxxx.lo' failed
make[5]: *** [asxxxx.lo] Error 1
make[5]: Leaving directory '/root/sdcc/support/sdbinutils/bfd'
Makefile:1354: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/root/sdcc/support/sdbinutils/bfd'
Makefile:1159: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/root/sdcc/support/sdbinutils/bfd'
Makefile:2511: recipe for target 'all-bfd' failed
make[2]: *** [all-bfd] Error 2
make[2]: Leaving directory '/root/sdcc/support/sdbinutils'
Makefile:838: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/root/sdcc/support/sdbinutils'
Makefile:151: recipe for target 'sdcc-sdbinutils' failed
make: *** [sdcc-sdbinutils] Error 2
root@DietPi:~/sdcc#

danpowell88, 2018-12-21:
Managed to get it working in windows using the ti sniffer and then exported to pcap.

I can't figure out how to find the security header info though so i can view the data.

I ran the capture while doing a reset using the dimmer switch and then made sure the bulb appeared in the hue app and responded to commands and then changed the default power on action.

any ideas?

[sniff.zip](https://github.com/Koenkk/zigbee2mqtt/files/2702237/sniff.zip)

danpowell88, 2018-12-21:
Looking at it a bit more it could be a limitation of the capture tool in windows, I don't suppose you could upload a 2530 compatible sensiff firmware? @kiwinol

danpowell88, 2018-12-21:
Managed to get it somewhat working using https://github.com/homewsn/whsniff

Just havent been able to capture the zigbee key yet

danpowell88, 2018-12-22:
Ok figured it out, for anyone else trying to figure out how to sniff hue traffic, hue uses a private key and not the ZigbeeAlliance09 one, the key can be found here 

https://peeveeone.com/?p=166

Once you've got that you can see Zigbee ZDP frames, filter wireshark on zbee.sec.key and you should have the network key which you need to add to wireshark and voila all of the data now shows

Koenkk, 2018-12-28:
I've written a tutorial on how to do this: https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/how_tos/how_to_sniff_zigbee_traffic.md

closed 2018-12-28 by Koenkk

mcsSolutions, 2018-12-31:
I have never used Ubuntu before but tried to follow the tutorial.  I installed Ubuntu 16.04.2 on RPi.  Followed tutorial to the point of curl command to get the tarball.  It failed due to permissions trying to create the directories and install the LICENSE etc files in /opt/whsniff-1.1.  Tried sudo prefix with same result.  Became root with sudo -s and then was able to complete the install.

During Wireshark install a prompt was asked if unprivledged user should be allowed and I selected the Yes option.

Inserted the flashed CC2531 into RPi.

Tried to start capture and received message "libEGL warninng: DRI2: failed to authenticate
Wireshark (as superuser) windows pops up "Lua: Error during loading: /usr/share/wireshark/init.lua:32: dofile has been disabled due to running wireshark as superuser.  See https://wike.wireshark.org/CapgtureSetup/CapturePrivleges for help in running Wireshark as an unprivledged user."
After clicking OK another Wireshark (as superuser) window "Unrecognized libpcap format or not libpcap data.
The interface enxb827eb9e8372 and bluetooth0 were showing traffic in the timelapse.  nflog, nfqueue, usbmon1 were not showing any traffic.  I do not know what interface the CC2531 is suppose to report.  The LED on the interface was illuminated when initially installed, but was out after Wireshark was running for about a minute.  I did confirm that /dev/ttyACM0 was recognized as a device with ls -l command.

Started a new terminal window and tried to start capture again.  Same libEGL warning.  In the Wireshark window a popup "Couldn't run /usr/bin/dumpcap in child process: Permission denied".  After clicking OK no interfaces were visible in Wireshark window.

Purge and Install Wireshark again, but this time no prompt for type of user allowed.  Attempts at capturing traffic were same as before.

I assume some type of permission issues exist, but dont know how to overcome them.

mcsSolutions, 2019-01-01:
I did have success with a Debian headless install and running Wireshark on Windows via SSH

C:\Program Files\PuTTY\plink -ssh xxxxxxx@192.168.0.200 -pw xxxxxx "whsniff -c 11" | "C:\Program Files\Wireshark\Wireshark.exe" -k -i -

bremy, 2019-02-26:
Bonjour Gentlemen

I have a question on trafic snifing. Is it necessary to have 2 CC2531 dongles (one for the coordinator, one for the sniffer), or just flashing the sniffer firmware on my CC2531, sniffing and flashing back the zigbee2mqtt firmware on the dongle?

mcsSolutions, 2019-02-26:
only need sniffer unless you also want to sniff traffic from your zigbee2mqtt coordinator.  in the later case you would need two devices.

bremy, 2019-02-26:
The goal is to identify the coordiantor_group to associate my basic on/off Tradfri remote to the remote outlet
I saw that must be done in that page [https://www.zigbee2mqtt.io/getting_started/pairing_devices.html](https://www.zigbee2mqtt.io/getting_started/pairing_devices.html)

Koenkk, 2019-02-27:
@bremy I would recommend updating to the latest dev branch and dev firmware (https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin). `coordinator_group` is not required anymore here.

bremy, 2019-02-27:
@Koenkk  That's the one I was flashing yesterday on my cc2531. Outlet is working perfectly at the moment. I will test ASAP with the basic on/off remote including in the bundle outlet+remote from IKEA
[https://www.ikea.com/fr/fr/catalog/products/80364794/](https://www.ikea.com/fr/fr/catalog/products/80364794)

I have a suggestion : when pairing GenOnOff devices. zigbee2mqtt could send one message to switch on and, one second after, another message to switch off. That could be an excellent way to confirm visiualy the success for the pairing process!

timdonovanuk, 2019-02-27:
I'm trying to sniff Xiaomi gateway traffic, but I cannot get the decryption going. I have entered in the "ZigbeeAlliance09" decryption key, but still not seeing decrypted packets. Any ideas? Are xiaomi not using the default key anymore?