Koenkk/zigbee2mqtt #718: Broken link: Alternate flashing method

abhaysbharadwaj, 2018-12-17:
Hello,
In this URL: https://github.com/Koenkk/zigbee2mqtt/blob/master/docs/getting_started/flashing_the_cc2531.md

This URL for "Alternative Firmware Flashing" Seems to be broken: https://github.com/Koenkk/zigbee2mqtt/blob/master/docs/useful_information/alternative-firmware-flashing-methods.md

I had checked it a few days back and it was working. Can you kindly look into this and help me out.

Iv4nS, 2018-12-17:
I think they are trying to move the docs to Github Pages but this is accessible at the moment: https://github.com/Koenkk/zigbee2mqtt/blob/master/docs/information/alternative_flashing_methods.md

Koenkk, 2018-12-17:
Thanks, fixed!

closed 2018-12-17 by Koenkk

timdonovanuk, 2019-02-14:
They are all broken again :(

Koenkk, 2019-02-14:
Documentation has been moved to https://www.zigbee2mqtt.io

FrancYescO, 2019-07-08:
If someone interested on flashing with RPi this method without CCDebbuger is very easy https://notenoughtech.com/home-automation/flashing-cc2531-without-cc-debugger/