Koenkk/zigbee2mqtt #719: Cannot find module zigbee.sheperd

detxm80, 2018-12-17:
Hi,
i've already installed zigbeemqtt by wiki and all gone ok.
A day i've upgraded it and after that arrived problems. My zigbee sniffer seems to work not perfetc. Before led green stay alway on now led a couple of minute stay green and after go off.
This is my log :
`
luigi@ubuntu-server:/opt/zigbee2mqtt$ sudo npm start

> zigbee2mqtt@1.0.0 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2018-12-17 18:18:27 Logging to directory: '/opt/zigbee2mqtt/data/log/2018-12-17.18-18-27'
module.js:550
    throw err;
    ^

Error: Cannot find module 'zigbee-shepherd'
    at Function.Module._resolveFilename (module.js:548:15)
    at Function.Module._load (module.js:475:25)
    at Module.require (module.js:597:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.js:1:81)
    at Module._compile (module.js:653:30)
    at Object.Module._extensions..js (module.js:664:10)
    at Module.load (module.js:566:32)
    at tryModuleLoad (module.js:506:12)
    at Function.Module._load (module.js:498:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.0.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.0.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/luigi/.npm/_logs/2018-12-17T17_18_27_145Z-debug.log

`
Thanks for your help and exscuse me for my bad english :)

Koenkk, 2018-12-18:
After upgrading, did you do: `rm -rf node_modules && npm install`?

Koenkk, 2018-12-18:
Whats the output of `ls node_modules` and `node --version`

detxm80, 2018-12-18:
i Koenkk, thanks for your support. Now i've solved install service by a docker, now it works but by fresh so install (follow the guide) with latest update got errors

closed 2019-02-24 by stale[bot]

robfr, 2019-06-04:
I am encountering the same problem as OP but I can't install via docker (running on ARM) so I don't have a solution.

The zigbee-shepherd folder in node_modules only contains a package.json file which seems but I am not sure if that is the cause of the problem or why it would be empty.

os is a synology armv7 linux (synology_armada38x_ds218j)
NPM version is 6.9.0
Node version is 8.16.0

node_modules folder contents and npm --verbose install are attached.

[node_modules.txt](https://github.com/Koenkk/zigbee2mqtt/files/3254886/node_modules.txt)
[npm-install.txt](https://github.com/Koenkk/zigbee2mqtt/files/3254887/npm-install.txt)

I think 1.3.1 worked previously but there was an OS upgrade and since then I cannot get it working. I've tried 1.4.0->1.1.1.

Any ideas ?