Koenkk/zigbee2mqtt #720: Error: rsp error: 16

Koenkk, 2018-12-17:
This issue happend after updating to the new firmware.

This seems to happen because of limited XDATA: https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/561949.

The new firmware decreased the XDATA size to 0x300 which causes this side effect.

danpowell88, 2018-12-24:
I'm seeing this trying to read the on/off state of a bulb

Koenkk, 2018-12-24:
@danpowell88 that is the startup log right? How do you send that command before starting?

danpowell88, 2018-12-24:
Sorry was just showing what version I was on, command came after that when
I was trying to read the attribute

georg90, 2019-02-09:
Error still persists in 1.1.0 using the OSRAM Smart+ plug.

```
zigbee2mqtt:error 2019-2-9 16:03:43 Zigbee publish to device '0x84182600000fafaa', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: rsp error: 16
```

Should I update my stick? Firmware seems to be a bit old, too:

```
Coordinator firmware version: 'undefined'
```

Koenkk, 2019-02-09:
@georg90 yes, please update your firmware to `20190109`.

georg90, 2019-02-09:
Okay, need to buy the flash device then.. :-)

behave49, 2019-02-17:
same problem with the muller light. what to buy to update firmware myself?

georg90, 2019-02-18:
@behave49 : https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html

SBTephan, 2019-02-21:
Also having this issue, this is for an ikea light (e14 bulb with white/warm white function)
I am using the 20190109 firmware.

Koenkk, 2019-02-21:
@SBTephan can you try with: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin

SBTephan, 2019-02-21:
I will give it a try, do I need to readd the devices? some information that may help..

sometimes this light takes some time to go out. also it is very short range to the coordinator and the other light bulb next to it.

What I will try first is to switch from one to the other and see if the problem travels with the lightbulb.

SBTephan, 2019-02-22:
strange this is, I moved the lights from one to antother, I didn't get the rsp error 16 but a time out.
The light gets the off signal and turns off but zigbee2mqtt thinks it failed to turn off and says "failed with error Error: Timed out after 30000 ms"

SBTephan, 2019-02-23:
So I moved the lightbulb but the problem stays on the same place. Again error 16.
 Looks like it is interference?
Also another light bulb from Ikea at a different place gave error 16 but the light is on, also in domoticz

Koenkk, 2019-02-24:
@SBTephan are you also running the latest dev branch?

SBTephan, 2019-03-12:
> @SBTephan are you also running the latest dev branch?

Hi Koen,

Installed the DEV branch and it seems to be working perfectly now 👍

Koenkk, 2019-03-12:
Awesome!

closed 2019-03-12 by Koenkk