Koenkk/zigbee2mqtt #721: time out on startup cc2530

greynoO, 2018-12-17:
I always get this error with my cc2530:

`zigbee2mqtt:error 2018-12-17 21:31:15 Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2018-12-17 21:31:15 Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2018-12-17 21:31:15 Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}`

The cc2530 does not have a reset button!
Additionally I am using iobroker which also uses this zigbee2mqtt library in an adapter and there is no problem in using the cc2530 module and the connection to zigbee devices. So the problem must lie somewhere else?

I wanted to try to integrate another device (Aqara curtain motor) but now I ain't even able to start the bridge :(

oakbrad, 2018-12-17:
I am getting the same, after having zigbee2mqtt running fine for months. I am not sure if I had it auto-update or what but I can't get the coordinator to start at all anymore.

`
> zigbee2mqtt@0.2.0 start /zigbee2mqtt-0.2.0
> node index.js
		Zigbee2mqtt requires node version >=8.11 10, you are running v8.11.4!
  zigbee2mqtt:info 2018-12-17 14:38:42 Logging to directory: '/share/zigbee2mqtt/log/2018-12-17.14-38-42'
  zigbee2mqtt:info 2018-12-17 14:38:43 Starting zigbee2mqtt version 0.2.0 (commit #unknown)
  zigbee2mqtt:info 2018-12-17 14:38:43 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-12-17 14:38:51 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  zigbee2mqtt:info 2018-12-17 14:39:51 Starting zigbee-shepherd
  zigbee2mqtt:error 2018-12-17 14:39:59 Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2018-12-17 14:39:59 Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2018-12-17 14:39:59 Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/zigbee2mqtt-0.2.0/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/zigbee2mqtt-0.2.0/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}`

This is on HassIO with Addon version 0.2.0, the stable zigbee2mqtt branch.

I tried pressing the buttons, rebooting, etc. I noticed when I rebooted my server (Debian/Proxmox) that it would not boot with the stick inserted - gave "USB 2-2 Read/write error" and as soon as I pulled the stick it continued booting. Made me wonder if my stick has gone bad but it's been working fine for a while..

greynoO, 2019-01-01:
still a problem after trying it with home assistant / hass.io addon. Same error.
The edge release even gets me an exception

```
zigbee2mqtt TypeError: Cannot read property 'close' of undefined
```

Koenkk, 2019-01-01:
This indicates a connection issue, the first post indicates that the device is already in use by another process? Perhaps iobroker?

greynoO, 2019-01-01:
In my first message I already pointed out that - when connected with iBroker - the error message is `Error Resource temporarily unavailable Cannot lock port`, so  this is not the case unfortunately..

Koenkk, 2019-01-01:
Error messages could differ, I would first check if you can get iobroker up and running, if that works try with zigbee2mqtt.

greynoO, 2019-01-01:
No no, ioBroker is running and also uses your `zigbee-shepherd-converters` library, so my stick works, `/dev/ttyUSB0` is correct etc.

I meant that while connected with ioBroker and I start zigbee2mqtt in parallel, the error message is `cannot lock port`. If ioBroker is off, I get the timeout. So ioBroker does not seem to be the issue.

Koenkk, 2019-01-01:
What is your `configuration.yaml`?

Koenkk, 2019-01-01:
If I remember correctly, you need to add 

```
advanced:
  rtscts: false
```

greynoO, 2019-01-01:
Wow, that solves it finally! Where does that come from? Maybe you should add it to the documentation?

Koenkk, 2019-01-02:
Done, thanks!

closed 2019-01-02 by Koenkk