Koenkk/zigbee2mqtt #722: adding support for Nue in wall dimmer

garywatts, 2018-12-18:
Trying to add support for this device below a NUE in wall dimmer model HGZB-02A
https://www.lectory.com.au/p/NUE-Smart-Zigbee-Gadgets-In-Ceiling-Light-Dimmer/NUEROOFLD

It connected ok
'{"type":"device_connected","message":"0x00124b000ae5fd99
 zigbee2mqtt:warn 2018-12-18 15:15:26 Device with modelID 'FB56+ZSC04HG1.0' is not supported.

I then tried to add it to devices.js using a few of the other examples.. here is the start of what I currently have
        zigbeeModel: ['FB56+ZSC04HG1.0'],
        model: 'HGZB-02A',
        vendor: 'Nue',
        description: 'ZigBee in ceiling light dimmer',
 (I have tried various formats of fromzigbee and tozigbee copied from similar devices in the devices.js file but I am not there yet)  

when i try and set the state to on via mqttlens nothing happens on the device. This is an example of a response I get back
  zigbee2mqtt:info 2018-12-18 17:31:30 Zigbee publish to '0x00124b000ae5fd99', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null

I've been using the hass.io add on with the "zigbee_shepherd_devices": true and fiddling with the devices.js that way so I have had very little poking around but any suggestion would be appreciated

Koenkk, 2018-12-18:
Can you provide the contents of your `database.db`?

Koenkk, 2018-12-18:
Lets first try to turn it on/off, can you try with:

```js
 {
        zigbeeModel: ['FB56+ZSC04HG1.0'],
        model: 'HGZB-02A',
        vendor: 'Nue',
        description: 'ZigBee in ceiling light dimmer',
        supports: 'on/off',
        fromZigbee: [],
        toZigbee: [tz.on_off],
        ep: (device) => {
               return {'': 12};
        },
    },
```

If that doesn't work, try with `return {'': 11};` and `return {'': 13};`

garywatts, 2018-12-18:
I just tried above(11,12 and 13) and got same response back when I send on to zigbee2mqtt/0x00124b000ae5fd99/set  

  zigbee2mqtt:info 2018-12-18 23:14:14 Zigbee publish to '0x00124b000ae5fd99', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-12-18 23:14:14 MQTT publish, topic: 'zigbee2mqtt/0x00124b000ae5fd99', payload: '{"state":"on"}'

garywatts, 2018-12-19:
I had no luck with brightness.. this is the output below... let me know if there is any logs deeper that will be of help and I will have a crack at getting to them.

 zigbee2mqtt:info 2018-12-19 16:00:32 Zigbee publish to '0x00124b000ae5fd99', genLevelCtrl - moveToLevelWithOnOff - {"level":19,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - 11
  zigbee2mqtt:info 2018-12-19 16:00:33 MQTT publish: topic 'zigbee2mqtt/0x00124b000ae5fd99', payload '{"state":"ON"}'

Koenkk, 2018-12-19:
You would need to sniff the traffic of the original gateway in order to find out which commands are used (#716)

garywatts, 2018-12-19:
I don't own a nue gateway so I might put this project on the backburner for a while. 

I acquired the device off a mate and it's more of a nice to do than must have thing. I am reluctant to purchase the gateway as it is not supported by home assistant. It would be good to see the nue devices with more support as they are the only Australia approved zigbee in wall devices around at the mo. Thanks for your help. It's a cool project.

highground88, 2018-12-19:
G'day Gary,

I'm in the same boat. See here:

https://github.com/Koenkk/zigbee2mqtt/issues/176

I've asked the distributor of these switches (3asmarthome) if he could assist and he has said that he will look at the project next year.

I don't have them on hand but there are some links in my post which have the SmartThings handlers on them, you may be able to try these. I won't have another chance for a few weeks to try anything new myself but I do have a nue gateway but unsure how to sniff the traffic.

All the best.

garywatts, 2018-12-20:
That promising that he responded that he will look into it. He would sell plenty more if there was some sort of documented api. Keeping my fingers crossed.

highground88, 2019-02-22:
Suggest closing this issue and carrying on here, as I've already done some of the sniffing and this switch is probably next on the list to conquer....  #176

Koenkk, 2019-02-22:
Let's continue in #176 indeed.

closed 2019-02-22 by Koenkk