Koenkk/zigbee2mqtt #723: Adding support for Xiaomi wall switch wireless mode

pataha, 2018-12-18:
I currently have about 20 Xiaomi wall switches double keys model `QBKG12LM` but some of them it just only used one key, in the Mi Home app I can set unused key acts as `Wireless Switch`, the switch does not toggle the relay. Can it be in `ziggbe2mqtt`?
PS, as I remember, double keys `QBKG03LM` has wireless mode too.
Here is ss from Mi Home app settings
[image]

Let me know if something I can help

Koenkk, 2018-12-18:
You need to find out which zigbee commands sets this setting. See #716 on how to sniff the traffic.

pataha, 2018-12-18:
ok, thanks @Koenkk
I'll try to do sniff and report the result here

pataha, 2018-12-18:
sorry, one question: Can I use `zigbee2mqtt` firmware to do sniff or must be the stock firmware?

Koenkk, 2018-12-18:
@pataha a firmware can be found in packet sniffer (not v2) http://www.ti.com/tool/PACKET-SNIFFER

jvmonjo, 2019-02-16:
Support for wireless mode will be awesome. Not much an expert on sniffing though (never done that).

Koenkk, 2019-02-16:
@jvmonjo there is a tutorial here: http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

pataha, 2019-03-11:
@Koenkk I saw that z2m 1.2.0 added support decoupled mode for `QBKG04LM/QBKG03LM` but not for `QBKG12LM/QBKG11LM`. Can you pls add support for those, it should be similar

Koenkk, 2019-03-11:
@pataha I don't have these so cannot check if it works, can you make the following change and check if it does?

Change https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L260 and https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L234
to `toZigbee: [tz.on_off, tz.xiaomi_switch_operation_mode],`

pataha, 2019-03-12:
thanks @Koenkk. Will do and report here

pataha, 2019-03-13:
hi @Koenkk, I tried as your suggestion and it works. Now I can use switch `QBKG12LM` without toggle relay.

Here is the logs, it seems only `click`, `double click` are support, (click on both keys show nothing)
```
zigbee2mqtt:debug 3/12/2019, 6:20:21 PM Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":1}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d0001f27b9f)
zigbee2mqtt:debug 3/12/2019, 6:20:22 PM Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":1}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d0001f27b9f)
zigbee2mqtt:debug 3/12/2019, 6:20:34 PM Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":2}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d0001f27b9f)
zigbee2mqtt:debug 3/12/2019, 6:20:34 PM Received zigbee message of type 'devChange' with data '{"cid":"genMultistateInput","data":{"presentValue":2}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d0001f27b9f)
```

for `long` press, it shows
zigbee2mqtt:debug 3/12/2019, 6:25:40 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65520":{"type":"Buffer","data":[170,16,5,65,135,1,1,16,0]}}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d0001f27b9f)

The last thing is when I set switch channel to `decoupled` the `switch component` still showing, it should become a new sensor

Thanks

Koenkk, 2019-03-13:
@pataha unfortunately it's no possible to update the home assistant dynamically.

Could you make a PR so this can be integrated?

pataha, 2019-03-14:
@Koenkk 
1. can we remove discovery from Home Assistant and add a new component?
2. add a new config to devices.yml for `decoupled mode`
I think #2 is sufficient since people do not need to change that mode frequently

as for PR, will do that soon

Koenkk, 2019-03-14:
@pataha I don't what you mean with seems to be unusable, what we could do is discover an extra click sensor this this device in home assistant.

pataha, 2019-03-15:
I meant without adding new sensor the wireless mode cant be used to trigger automation. So pls add

Koenkk, 2019-03-15:
Added! Assuming that everything is done here, closing.

closed 2019-03-15 by Koenkk

pataha, 2019-03-23:
It shown on HA but the sensor click value was not changed when click on device

in Z2M logs
```
zigbee2mqtt:debug 3/23/2019, 2:02:32 PM Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":1}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d0001f27b9f)
  zigbee2mqtt:debug 3/23/2019, 2:02:34 PM Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":1}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d0001f27b9f)
  zigbee2mqtt:debug 3/23/2019, 2:02:35 PM Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":1}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d0001f27b9f)
  zigbee2mqtt:debug 3/23/2019, 2:02:36 PM Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":1}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d0001f27b9f)
```
but nothing was published to mqtt

Koenkk, 2019-03-23:
Can you check with the latest dev branch?

pataha, 2019-03-24:
Thanks. I’ll try
I can see in your commit, only single click is supported?  Double clicks with `presentValue === 2` should be supported as well

Koenkk, 2019-03-25:
@pataha added!

acelauwc, 2019-03-27:
After getting the dev branch, the switch and click events are reporting in homeassistant as well. However it seems that the left and right button click are reported incorrectly. I need to change L975 in fromZigbee.js file to the following:

`    975                 const mapping = {5: 'left', 6: 'right', 7: 'both'};`

Koenkk, 2019-03-27:
@acelauwc would you mind making a pr?