Koenkk/zigbee2mqtt #727: Zigbeet2mqtt dose not start... status=1/failure

thomas1977mg, 2018-12-18:
Dear Koen,

today I have updated the device and after that I could not get the system running.
Zigbee2MQTT will not start and I don´t have a solution for it. What could it be?

I am getting really nervous, because I have a big system running and need the devices running.
Thanks for your kind support.
Thomas

----

https://pastebin.com/K7Ba6Dw2

----

Koenkk, 2018-12-18:
Can you post your configuration.yaml?

thomas1977mg, 2018-12-18:
Dear Koen,

thanks for your quick reply...
here it is w/o the devices:

# Home Assistant integration (MQTT discovery)
homeassistant: false

permit_join: true

mqtt:
  base_topic: zigbee2mqtt
   server: 'mqtt://192.168.178.32:1883'
broker: '192.168.178.32'

#   server: 'mqtt://192.168.178.32:1900'
#server: 'mqtt://localhost'

serial:
port: >-

/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0014D9DBB6-if00
devices:

advanced:
  channel: 11

I have also installed again node.js to come above 10. BEfore it was 8.xx -
but without success.

Thanks,
Thomas

Koenkk, 2018-12-19:
Is your `port` indented with 2 spaces?

thomas1977mg, 2018-12-19:
Dear Koen,

I am not 100% sure what you mean.
I have on FHEM two broker running. One @ 1883 and a newer one at 1900.
But this were already running since a couple of days.

I used to change it in the yaml forth and back w/o success.

The problem started yesterday after updating the system following your
instruction manual.

Best regards,
Thomas

thomas1977mg, 2018-12-19:
One additional information:

node --version v10.14.2
npm --version 6.4.1

Koenkk, 2018-12-19:
Add 2 spaces before `port` in `configuration.yaml`.

thomas1977mg, 2018-12-19:
Dear Koen...

you are the worlds greatest!

It works again....!

Many thanks...

Thomas

Koenkk, 2018-12-19:
great!

closed 2018-12-19 by Koenkk