Koenkk/zigbee2mqtt #728: Not an issue: Theoretical questions about pairing storage

quarcko, 2018-12-18:
Koenkk,

Just have a few questions regarding the pairing "memory". 
I'm interested in internal workings, because now it seems a bit complicated with lots of layers,
and no clear understanding from my side "where-what-is"...
layers are: CC2531 <-> firmware <-> zigbee-shepherd <-> zigbee-shepherd-converters <-> zigbee2mqtt <-> HA.

1) Does the CC2531 stores any info with paired devices? or is the *.db file inside zigbee-shepherd a pairing "memory"?
in case of *.db file then any stick can pick it up and have all the pairings immediately?
2) Does the CC chip has some unique ID (0x...) or is this an ID generated once by zigbee-shepherd?
And do the paired devices store it inside themselves (end devices)?
Basically here i ask if pairing is two-way process:
Stick stores paired devices and end-device also stores paired coordinator?
3) Is it possible to "clone" whole stick with all information to another one, and do not have to "re-pair" devices? Or there is no need to "clone" nothing, just get a new stick running on same PAN-ID as was old stick?

So lots of questions, but main thing is: are the pairings stored inside firmware or inside *.db file
and if end devices will communicate with new stick with same PAN-ID as old had or something else is needed?

I hope you have some free time to give some answers - if not then i'll close this issue after a while. Thanks!

Koenkk, 2018-12-19:
1. CC2531 stores info of paired devices.
2. The CC chip has a unique ID which is stored by the devices, you can reset it via the CC flashing tool
3. Yes, make sure the channel and the pan-id is the same. When starting the stick for the first time after flashing, make sure it is out of range of your existing zigbee network so that it will get the same pan-id. (it does a scan for existing pan-ids before assigning it).

quarcko, 2018-12-19:
Ok, it is more clear now... so:

1) is it possible to 'dump' firmware + pairing data from the cc2531 to pc with flashing tool?
1a) Somewhere i saw posts that there exists some 'specially' compiled firmware that does not reset device memory, is it possible for you to compile such alternative version? (useful for firmware update without loosing data... i really want to test your new optimized FW, but it is not worth it now, if i loose all the pairings...)

2) Does the flashing tool 'resets' it random or can i assign myself anything i want? any example cmd?

3) So i can start the stick somewhere far away (like at work), and it will get same PAN id, and then power it off and bring it home? or after again power-on it will 'scan' and get different pan-id?
3a) What is the point of zigbee2mqtt 'panid' option in settings of addon? is it any useful to 'force' pan-id?

Ton1965, 2018-12-19:
Hi Guys,

Since somebody has started this theme I would like also to ask a question that bothers me a lot:

When a device is removed from pairing by zigbee2mqtt, is it removed from database only or from CC2531 as well? 

The reason I ask is I read somewhere on this site that people encountered problems with reaching the device count limit when they tried to re-pair the device after first removing as if after each re-pairing a device occupied new memory in CC2531. I am planning to change my channel and am almost certain this would require to re-pair all devices but at the same time I am afraid to get stuck with the device count limit.

danpowell88, 2018-12-20:
I cant answer that question but you can use the network map endpoint to get a list of devices and then use the remove endpoint to remove any that no longer appear in config

Ton1965, 2018-12-20:
@danpowell88 , thanks for the answer.
Can this endpoint removal be done via zigbee2mqtt or do I need some other tool for that?

danpowell88, 2018-12-21:
submit it via the mqtt endpoint

danpowell88, 2018-12-21:
@Koenkk so to upgrade to a newer version of the firmware I would just need to specify panid/channel in the config?

Do I also need to be out of the range of the network, and if so is unplugging the co-ordinator enough to do that?

If I didnt set it originally how do I find out what it needs to be?

Ton1965, 2018-12-21:
Hi @Koenkk ,
Thanks for your hard work!

The problem is I want to use those Livolo zigbee switches that were shown to connect only on channel 26. Is it possible to change the channel without re-paring?
@khzd wrote in corresponding topic that he needed to re-pair only router devices. I tried to switch the channel and all devices disappeared. I did not have enough patience to wait and switched the channel back to 11.

Koenkk, 2018-12-21:
@Ton1965 channel switching requires repairing.

Ton1965, 2018-12-21:
@Koenkk , thanks a lot for your reply.

khzd, 2018-12-22:
> Hi @Koenkk ,
> Thanks for your hard work!
>
> The problem is I want to use those Livolo zigbee switches that were shown to connect only on channel 26. Is it possible to change the channel without [...]

Hi,
it's my bad, i confirmed wrongly that only rooter need to be paired when channel is changed.
I did the experince last night and see that ALL need to be paired again... 50 devices to re-paire, I didnt' finish yet :(

closed 2019-02-27 by stale[bot]