Koenkk/zigbee2mqtt #729: Ecosmart Bulb with model '�P'

boojew, 2018-12-19:
Unsupported device
First check: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html

Bug report
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

Similar issue to #286. Let me know what I can provide to help solve the issue. Logs can be seen here https://pastebin.com/bFWSk18z

Koenkk, 2018-12-19:
Can you provide your `database.db`?

Koenkk, 2018-12-20:
You can use `\u0000\u0002\u0000\u0004\u0012 �P\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u000e` as the `zigbeeModel`.

closed 2019-01-08 by boojew