Koenkk/zigbee2mqtt #733: Device Registry not working for Tradfri Color Lamp (LED1624G9)

ophilips, 2018-12-19:
After the update to the latest version 1.0.1 with support for device registry is almost all of my devices showed up correctly in the integrations panel. Only two Ikea lamps (LED1624G9) are listed as "Entities without devices". I don't know if that's a problem of Homeassistant or Zigbee2mqtt. The strange thing is that these two devices didn't get a "device_id" Homeassistant:

/home/homeassistant/.homeassistant/.storage/core.entity_registry

```

            {
                "config_entry_id": "2dda62983ee04927bd9af0026a2ef758",
                "device_id": "53f469b2fe4549fd9905d80e02a1c90c",
                "disabled_by": null,
                "entity_id": "sensor.0x00158d00029c0276_battery",
                "name": "Batterie",
                "platform": "mqtt",
                "unique_id": "0x00158d00029c0276_battery_zigbee2mqtt"
            },
            {
                "config_entry_id": "2dda62983ee04927bd9af0026a2ef758",
                "device_id": null,
                "disabled_by": null,
                "entity_id": "light.0x000b57fffea7eab1_light",
                "name": "Leseleuchte",
                "platform": "mqtt",
                "unique_id": "0x000b57fffea7eab1_light_zigbee2mqtt"
            },
            {
                "config_entry_id": "2dda62983ee04927bd9af0026a2ef758",
                "device_id": null,
                "disabled_by": null,
                "entity_id": "light.0x000b57fffeae4a78_light",
                "name": "Nachtleuchte",
                "platform": "mqtt",
                "unique_id": "0x000b57fffeae4a78_light_zigbee2mqtt"
            }

```
zigbee2mqtt config:
```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  disable_led: true
  port: /dev/ttyACM0
devices:
  '0x00158d000277246b':
    friendly_name: '0x00158d000277246b'
    retain: false
  '0x000b57fffea7eab1':
    friendly_name: '0x000b57fffea7eab1'
    retain: false
  '0xd0cf5efffe43baaa':
    friendly_name: '0xd0cf5efffe43baaa'
    retain: false
  '0x00158d000272cba9':
    friendly_name: '0x00158d000272cba9'
    retain: false
  '0x00158d00029c0276':
    friendly_name: '0x00158d00029c0276'
    retain: false
  '0x000b57fffeae4a78':
    friendly_name: '0x000b57fffeae4a78'
    retain: false
  '0x00158d00024dd297':
    friendly_name: '0x00158d00024dd297'
    retain: false
  '0x00158d00024e4af5':
    friendly_name: '0x00158d00024e4af5'
    retain: false
```
[image]

Koenkk, 2018-12-19:
I'm having the same issue, don't know what is causing this. @cbrherms do you know more about this?

cbrherms, 2018-12-19:
Looks like https://github.com/home-assistant/home-assistant/pull/19013 didn't make it to rc in time to be included in 0.84.

I think there were some last minute changes to the `light.mqtt` pull that required changes to the discovery pull. Shame. There were intending to have all of the mqtt platforms ready in one release.

Nothing to worry about though. Support for light will be added in 0.85 now as it's already been remerged in to dev, I don't see there being any further changes that would prevent that.

So long as all the other platforms are showing up correctly, and the devices that are created via discovery (including the light devices with the new `schema` format) are working correctly, then there's nothing needed to be done on zigbee2mqtt's end.

Just need to wait it out for 0.85 release of HA scheduled for Jan 9th.

ophilips, 2019-01-10:
Ok. It's working now as expected with Home Assistant 0.85:
[image]

closed 2019-01-10 by ophilips

cbrherms, 2019-01-10:
Fantastic