Koenkk/zigbee2mqtt #735: Reproducible crash while publishing some messages in a sequence

christof62, 2018-12-20:
Crash after publishing multiple messages. This happens also relatively often during normal use cases e.g. setting a light scene with 3 devices. 

Configuration:
zigbee2mqtt on raspberry pi
CC2530 with coordinator firmware

[console.log](https://github.com/Koenkk/zigbee2mqtt/files/2700850/console.log)
[debug.log](https://github.com/Koenkk/zigbee2mqtt/files/2700864/debug.log)

christof62, 2018-12-21:
Found out that issue was due to enabled UART  flow control. Probably it is possible to identify problems in the communication to the zigbee module.

Koenkk, 2018-12-21:
@christof62 what did you change? (this could be useful for others)

christof62, 2018-12-21:
the power 3.3V supply of the USB TTL converter was probably to weak.
Measured around 3V. I exchanged the CH340G by another version (
https://www.ebay.de/itm/CH340G-USB-to-TTL-Adapter-mit-4x-Dupont-Kabel/264090749600)
and the problem was solved.