Koenkk/zigbee2mqtt #737: zigbee2mqtt detecting Aqara cube but not gesture anymore

abdelhakel, 2018-12-21:
Hi, 

My Aqara cube MFKZQ01LM was perfectly working with Zigbee2mqtt : all gestures detected. Automation ok with Yeelight. But since yesterday, no gesture is detected. 
When i open it, and click once, here is what is appearing in logs :
`déc. 21 10:31:59 raspberrypi npm[397]:   zigbee2mqtt:info 2018-12-21 10:31:59 MQTT publish, topic: 'zigbee2mqtt/Cube', payload: '{"battery":99,"voltage":3005,"linkquality":78}'`

So i guess, the cube is detected and paired. It's the only sensor not working.
I receive states from all my others Aqara sensors : temperature, humidity, door...

`sudo systemctl status zigbee2mqtt :`

```
● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-12-21 09:03:35 CET; 1h 42min ago
 Main PID: 397 (npm)
   CGroup: /system.slice/zigbee2mqtt.service
           ├─ 397 npm
           ├─1031 sh -c node index.js
           └─1032 node index.js

déc. 21 10:14:51 raspberrypi npm[397]:   zigbee2mqtt:info 2018-12-21 10:14:51 MQTT publish, topic: 'zigbee2mqtt/chambre enfants', payload: '{"temperature":18.89,"linkquality":65
déc. 21 10:21:34 raspberrypi npm[397]:   zigbee2mqtt:info 2018-12-21 10:21:34 MQTT publish, topic: 'zigbee2mqtt/chambre parents', payload: '{"temperature":18.21,"linkquality":42
déc. 21 10:21:34 raspberrypi npm[397]:   zigbee2mqtt:info 2018-12-21 10:21:34 MQTT publish, topic: 'zigbee2mqtt/chambre parents', payload: '{"temperature":18.21,"linkquality":42
déc. 21 10:21:34 raspberrypi npm[397]:   zigbee2mqtt:info 2018-12-21 10:21:34 MQTT publish, topic: 'zigbee2mqtt/chambre parents', payload: '{"temperature":18.21,"linkquality":42
déc. 21 10:31:59 raspberrypi npm[397]:   zigbee2mqtt:info 2018-12-21 10:31:59 MQTT publish, topic: 'zigbee2mqtt/Cube', payload: '{"battery":99,"voltage":3005,"linkquality":78}'
déc. 21 10:33:12 raspberrypi npm[397]:   zigbee2mqtt:info 2018-12-21 10:33:12 MQTT publish, topic: 'zigbee2mqtt/porte', payload: '{"contact":true,"linkquality":36,"battery":100,
déc. 21 10:33:12 raspberrypi npm[397]:   zigbee2mqtt:info 2018-12-21 10:33:12 MQTT publish, topic: 'zigbee2mqtt/porte', payload: '{"contact":true,"linkquality":36,"battery":100,
déc. 21 10:34:26 raspberrypi npm[397]:   zigbee2mqtt:info 2018-12-21 10:34:26 MQTT publish, topic: 'zigbee2mqtt/chambre enfants', payload: '{"temperature":18.85,"linkquality":73
déc. 21 10:34:26 raspberrypi npm[397]:   zigbee2mqtt:info 2018-12-21 10:34:26 MQTT publish, topic: 'zigbee2mqtt/chambre enfants', payload: '{"temperature":18.85,"linkquality":73
déc. 21 10:34:26 raspberrypi npm[397]:   zigbee2mqtt:info 2018-12-21 10:34:26 MQTT publish, topic: 'zigbee2mqtt/chambre enfants', payload: '{"temperature":18.85,"linkquality":73

khzd, 2018-12-22:
hi @abdelhakel 
i have a pair of them, all working with the current release: see my log
Dec 22 11:34:30 ubuntu59 npm[26515]:   zigbee2mqtt:info 12/22/2018, 11:34:30 AM MQTT publish: topic 'zigbee2mqtt/x_cube_b', payload '{"battery":31,"voltage":2965,"linkquality":115,"action":"shake"}'

restart with debug option and provide your log and the DB record of the cube
Cheers

abdelhakel, 2018-12-22:
> hi @abdelhakel
> i have a pair of them, all working with the current release: see my log
> Dec 22 11:34:30 ubuntu59 npm[26515]: zigbee2mqtt:info 12/22/2018, 11:34:30 AM MQTT publish: topic 'zigbee2mqtt/x_cube_b', payload '{"battery":31,"voltage":2965,"linkquality":115,"action":"shake"}'
> 
> please, restart with debug option and provide your log and the DB record of the cube
> Cheers

Hi, problem fixed.
I just opened the cube it and pushed the pairing button.
Everything is back to normal.
Thank you !!!

closed 2018-12-22 by abdelhakel