Koenkk/zigbee2mqtt #738: OSRAM Smart+ Classic E27 Multicolor

t-buss, 2018-12-21:
Hello there,
I'm trying to get a OSRAM Smart+ RGBW Bulb to work, the succesor of the Lightify product line. On the device, it says "AC03645", which can be found on the [supported devices](https://koenkk.github.io/zigbee2mqtt/information/supported_devices.html) and in the `devices.js` file (still under the name Lightify, but I think the interfaces are not different.

Pairing worked, but the log shows this message:
`Coordinator firmware version: 'undefined'`
`Dec 20 17:13:59 raspberrypi npm[1860]:   zigbee2mqtt:info 12/20/2018, 5:13:59 PM Currently 2 devices are joined:`
`Dec 20 17:13:59 raspberrypi npm[1860]:   zigbee2mqtt:info 12/20/2018, 5:13:59 PM 0x7cb03eaa00b0afc3 (0x7cb03eaa00b0afc3): AB3257001NJ - OSRAM Smart+ plug (Router)` <-- works as expected
`Dec 20 17:13:59 raspberrypi npm[1860]:   zigbee2mqtt:info 12/20/2018, 5:13:59 PM unknown (0x7cb03eaa00aa1507): unkown - undefined unknown (Router)` <-- The new bulb

As you can see, the modelId is "undefined", so I can't follow the guide to support new devices and add an entry to `devices.js`

The corresponding line to that device in `database.db`: https://pastebin.com/kE6s7uDf
Actually, there is more than one line with that device in that file, I'm not sure if this is an issue.

I'm out of ideas how to proceed and would be grateful for some help :)

t-buss, 2018-12-21:
Thanks for the reply.

That model ID is already taken in `devices.js`: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L845
In fact, the entry looks exactly like I would have expected (apart from "Lightify"). Even the model number is the same, "AC03645". Still, the bulb is not treated correctly. Is the device not properly mapped to the definition in zigbee-shepherd-converters?

edyczkowski, 2018-12-22:
Hi
I have those bulb. They are working ok. I had a problem to pair it. I had to pair it twice for some reason. multiple time on and off.

t-buss, 2018-12-24:
Hi,
thanks for the answer @edyczkowski, good to know. I will try to re-flash the chip to undo any pairing (luckily, I only have the 2 devices so far) and do a factory reset of the bulb.
That might work.

edyczkowski, 2018-12-24:
I didn't re-flash it. I was desperate, so I tried to pair again and again..."bit of hammering"
I paired it ones with "unknown - undefined unknown (Router)"
Then I repeated the process. 

With XIAOMI you have to keep pressing reset... I think it is similar with those bulbs

t-buss, 2019-01-09:
I re-flashed the stick and got the same result as before. I also tried "hammering" with no luck :(

Koenkk, 2019-03-13:
Could you try to:

- stop zigbee2mqtt
- remove all entries of this device from `data/database.db`
- start zigbee2mqtt
- re-pair (and check if it get's a modelId now).

whyzermans, 2019-05-09:
Hi Guys. I have exactly the same issue as t-buss explained but with the different model. I have Smart+ E27 Dimmable Classic. Used to go by Classic A60 W clear - LIGHTIFY - and they use the same model name (AC03641)

_zigbeeModel: ['Classic A60 W clear - LIGHTIFY'],
        model: 'AC03641',
        vendor: 'OSRAM',
        description: 'LIGHTIFY LED Classic A60 clear',
        extend: generic.light_onoff_brightness,__

This is the info from the blob. My bulb is the same model (AC03641) but is now not LIGHTIFY but Smart+. Zigbee2mqtt shows it as unknown device (Router). And I have tried everything including removing all entries from data/database.db.

At the same time I bought this:

        zigbeeModel: ['B40 DIM Z3'],
        model: 'AC08562',
        vendor: 'OSRAM',
        description: 'SMART+ Candle E14 Dimmable White',
        extend: generic.light_onoff_brightness

And it works perfectly on the same system.

Any kind of help / hints would be much appreciated.

whyzermans, 2019-05-10:
I managed to fix this by taking entry from the working bulb and paste it with nonrecognized one. It works lika a charm.

closed 2019-07-16 by stale[bot]