Koenkk/zigbee2mqtt #741: Flashing CC2531-USB Variant with 5 pin holes

joshdinsdale, 2018-12-21:
Hi All

I have 2 x CC2531 USB sticks, one is the standard version with the normal pin headers, i have this flashed and working nicely. 
I also have a second CC2531 however this one doesnt have a header, but just 5 pin holes labeled as debug.

Does anyone have a guide for flashing this variant (i would like to use it as a router).? I cant seem to find a pinout for this version...

Thanks!

[https://i.ebayimg.com/images/g/w3YAAOSwaIpb19Ed/s-l1600.jpg](url)

ptvoinfo, 2018-12-22:
@joshdinsdale It is necessary to know the meaning of each pin. You may request it from the seller.

FlorentSueur, 2019-01-10:
Hi @joshdinsdale, I also bought a CC2531-USB variant with 5 pin holes. Did you get the meaning of each pin and did you flash it with the cc debugger ?

Thanks

FlorentSueur, 2019-01-11:
I found this picture in an other case (#790) but it seems to be the CC2540-USB1 as written on the board [
[image]
](url)

andyfitz, 2019-02-26:
also have a stack of this variant . connecting it could prove troublesome

GHPS, 2019-03-14:
I have the same issue - blue CC2531-USB adapter in acrylic case.

I hope that the manufacturer radioland-china.com wasn't mean enough
to change the pinout from the programming adapter that has a
double line (DIL) and a single line (SIL) header. So I checked the
connections:

```
1          5
+----------+
|O O O O O |
+----------+
 | | | | |
 1 2 3 4 7

2         10
+----------+
|O O O O O |
|O O O O O |
+----------+
1          9
```
That's a good sign since it makes sense according to the official documentation

[image]

So the pinout of the SIL connector should be

1: GND
2: Target Voltag Sense
3: DC - Debug Clock
4: DD - Debug Data
5: Reset

On my USB stick pin 1 is the right most pin with the USB connector
facing left. That means that the pins are in reversed order compared
to the picture posted previously by @FlorentSueur.

LordMyschkin, 2019-04-14:
I can confirm that flashing zigbee2mqtt was possible (and suprisingly simple) using the above information following the guide on http://www.zigbee2mqtt.io/information/alternative_flashing_methods.html using a raspberry pi zero w.

GHPS, 2019-04-14:
Thanks for the feedback.

Brandon4466, 2019-05-01:
How did you go exactly about flashing this? How were you able to connect it to the CC Debugger? I tried with just some pin wires but it connects properly.

LordMyschkin, 2019-05-01:
@Brandon4466 since my female-male wires did not fit into the debugging holes, but my male-male did, i just connected the Raspberry to a breadboard, than thee breadboard to the debugging holes by male-male wires. They fit very well, hold themself in place. Then I double checked against short-wiring and the correct Pins, connected to Raspberry and the Board to usb an there we go. No cc-debugger needed for this method, so I‘d prefer this one over any other!

Brandon4466, 2019-05-01:
@LordMyschkin and you had to use all 5? Thank you for your help also!

LordMyschkin, 2019-05-02:
> Am 01.05.2019 um 08:33 schrieb Brandon4466 <notifications@github.com>:
> 
> @LordMyschkin and you had to use all 5? Thank you for your help also!
> 

i did not try with fewer, I used all 5.

Brandon4466, 2019-05-03:
@LordMyschkin Hm, when I do it, ./cc_chipid just returns ID = 0000, All are connected, did you run into this problem?

LordMyschkin, 2019-05-03:
I got cchip_ip=0000 with an error in wiring; it went away after re-checking and correcting the wires.

Brandon4466, 2019-05-05:
@LordMyschkin Is there a way you got the male to male wires to stay in place? They don't fit in the holes on the debug ports at all for me.

zbuh, 2019-05-11:
> I can confirm that flashing zigbee2mqtt was possible (and suprisingly simple) using the above information following the guide on http://www.zigbee2mqtt.io/information/alternative_flashing_methods.html using a raspberry pi zero w.
> Although cc_chipid returned b522 - i'll still have to confirm zigbee2mqtt works....

did you test zigbee2mqtt ?

LordMyschkin, 2019-05-21:
sorry, it took me a while to set up a testing environment. Yes, I succeeded in connecting the stick to a AMD64 Debian Stretch and to pair an Aquara Temperature/Humidity sensor.   It did show some error first "modelID undefined not supported;" but started to work than.

schmurtzm, 2019-06-24:
Hi successfully flashed this one, just followed GHPS's pinout diagram (first pin with a square shape is ground)

 [image](https://DL.dropbox.com/s/kdeceuy83p0k5v1/2019-06-24%2015.29.17.jpg?dl=0)

diagonali, 2019-08-15:
> Hi, successfully flashed this one (acrylic case has been removed), just followed GHPS's pinout diagram (first pin with a square shape is ground)
> 
> [image]

Where do your wires go to from being attached to the USB debug holes? I've got the following: [
[image]
](url)

Can I just cut one of the cables and use it's wires? Or is soldering required? If it''s messy I might just order a naked one with the debug connectors but hopefully can get this one working...

closed 2019-10-21 by stale[bot]