Koenkk/zigbee2mqtt #744: Timeout Problem

matisaul, 2018-12-21:
Hi all, 

This is my first time configuring a zigbee device.
I bought a Zigbee CC2531 Sniffer from [Aliexpress](https://www.aliexpress.com/item/diymore-Wireless-Zigbee-CC2531-Sniffer-Bare-Board-Packet-Protocol-Analyzer-Module-USB-Interface-Dongle-Capture-Packet/32855873520.html?spm=2114.13010708.0.0.7f2e4c4d3E2UB9), and installed the custom firmware from [Koenkk Wiki](https://github.com/Koenkk/zigbee2mqtt/wiki/Getting-started).

The addon connect to MQTT broker without problem, but I get this error when I tried to allow new devices to join:

```
zigbee2mqtt:info 2018-12-13 12:13:04 Logging to directory: '/share/zigbee2mqtt/log/2018-12-13.12-13-03'
  zigbee2mqtt:info 2018-12-13 12:13:08 Starting zigbee2mqtt version 0.2.0 (commit #unknown)
  zigbee2mqtt:info 2018-12-13 12:13:08 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-12-13 12:13:10 zigbee-shepherd started
  zigbee2mqtt:info 2018-12-13 12:13:10 Coordinator firmware version: '20180815'
  zigbee2mqtt:info 2018-12-13 12:13:10 Currently 0 devices are joined:
  zigbee2mqtt:info 2018-12-13 12:13:10 Zigbee: disabling joining new devices.
  zigbee2mqtt:info 2018-12-13 12:13:10 Connecting to MQTT server at mqtt://192.168.0.11:1883
  zigbee2mqtt:info 2018-12-13 12:13:10 zigbee-shepherd ready
  zigbee2mqtt:info 2018-12-13 12:13:10 Connected to MQTT server
  zigbee2mqtt:info 2018-12-13 12:13:10 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-12-13 12:22:33 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2018-12-13 12:22:36 
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/zigbee2mqtt-0.2.0/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/zigbee2mqtt-0.2.0/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}
  zigbee2mqtt:info 2018-12-13 12:32:34 Zigbee: disabling joining new devices.
  zigbee2mqtt:info 2018-12-13 12:32:34 Zigbee: disabling joining new devices.
  zigbee2mqtt:info 2018-12-13 12:32:37 
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/zigbee2mqtt-0.2.0/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/zigbee2mqtt-0.2.0/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}
  zigbee2mqtt:info 2018-12-13 12:32:40 
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/zigbee2mqtt-0.2.0/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/zigbee2mqtt-0.2.0/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}
```

My addon config is:

```
{
  "data_path": "/share/zigbee2mqtt",
  "homeassistant": true,
  "permit_join": false,
  "mqtt_base_topic": "zigbee2mqtt",
  "mqtt_server": "mqtt://192.168.X.X:1883",
  "serial_port": "/dev/ttyACM0"
}

```

Anyone with the same problem?

Thanks in advance

khzd, 2018-12-21:
il faut remplacer  "permit_join": false, -->   "permit_join": true,

gabrielegirardi, 2018-12-29:
Thank you very much for all the help, now everything is working to perfection, wrong in 'inserting the name of the broker, just put there the IP of the device.

matisaul, 2019-01-24:
> change "permit_join": false, --> "permit_join": true,

I get the same error... I can't pair any device... It's very frustrating

gabrielegirardi, 2019-01-24:
Have you done the configuration to automatically pair the devices?

matisaul, 2019-02-05:
I think I figured out what was the problem. It seems there are some troubles with ResinOS (I had installed Hassio on a Intel NUC Pc). Now I have Hasssio over Ubuntu + Docker, and now It's working perfect. I have not tried with HassOS.

closed 2019-02-05 by matisaul