Koenkk/zigbee2mqtt #746: Support phillips hue power on behaviour

danpowell88, 2018-12-22:
In the latest hue light firmware  (SW 1.46.13_r26312) support has been added for allowing the bulb to return to its previous state after a power outage.

I've capture the zigbee traffic when setting a bulb to have the option and the following is the dump.

Can we add support for specifying this.

Setting it to  Power loss recovery (last used colour and brightness) generates the following two packets
[image]

[image]

Setting it to off (light always powers on with warm white full brightness) generates the following two packets
[image]

[image]

danpowell88, 2018-12-23:
I think I've managed to get part of the way there.

Looks like it requires changes in https://github.com/zigbeer/zcl-id/blob/master/definitions/cluster_defs.json

to add the following

to 'genOnOff'
`"powerOnCtrl":{"id":16387,"type":"enum8"}},`

to 'genLevelControl'
"powerOnCtrl":{"id":16384,"type":"uint8"}},

then in zigbee-shepherd-converters something like 
```
configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('genOnOff', 'powerOnCtrl', 255, cb),
				(cb) => device.write('genLevelCtrl', 'powerOnCtrl', 255, cb)
            ];
            execute(device, actions, callback);
        }	
```	

To all hue bulbs.

Unfortunately after doing this I get an error that z2mqtt failed to configure the lights, and I cant see any more debug information, even when running with debug log level and the DEBUG environment variable

Koenkk, 2018-12-25:
Instead of adding it to `configure` I think it's better to have a converters for this. 

Please checkout https://github.com/Koenkk/zigbee-shepherd-converters/tree/hue_power_behaviour

You should be able to control the power on by sending to `zigbee2mqtt/[DEVICE_ID]/set` value:

```
{
  "power_on_behaviour_1": "recover",   // recover or off
  "power_on_behaviour_2": "recover",  // recover or off
}
```

danpowell88, 2018-12-26:
Didnt appear to work

```
[31m zigbee2mqtt:error[39m 2018-12-26 11:03:45 Zigbee publish to '0x00178801035609f4', genOnOff - {

"power_on_behaviour_1": "recover",
"power_on_behaviour_2": "recover",
} - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Unrecognized command
```

Koenkk, 2018-12-27:
I don't understand why this error is thrown, are you on the latest (clean) dev branch?

danpowell88, 2018-12-28:
@Koenkk Tried it again on a clean rapsberry pi with the full dev branch

```
zigbee2mqtt:info 12/28/2018, 7:28:44 AM Starting zigbee2mqtt version 1.0.1 (commit #5e4e6d8)
zigbee2mqtt:info 12/28/2018, 7:28:44 AM Starting zigbee-shepherd
zigbee2mqtt:info 12/28/2018, 7:28:46 AM zigbee-shepherd started
zigbee2mqtt:info 12/28/2018, 7:28:46 AM Coordinator firmware version: '20181024'  
zigbee2mqtt:info 12/28/2018, 7:29:13 AM Zigbee publish to device '0x00178801035609f4', genOnOff - {  "power_on_behaviour_1": "recover",   // recover or off  "power_on_behaviour_2": "recover",  // recover or off} - {} - {"manufSpec":0,"disDefaultRsp":0} - null
zigbee2mqtt:error 12/28/2018, 7:29:13 AM Zigbee publish to device '0x00178801035609f4', genOnOff - {  "power_on_behaviour_1": "recover",   // recover or off  "power_on_behaviour_2": "recover",  // recover or off} - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Unrecognized command
```

I also manually modified the package.json to change the following 

`"zigbee-shepherd-converters": "git+https://github.com/Koenkk/zigbee-shepherd-converters.git#6d49008e91c7606c9067be8c469e898cb2e5a0b2"`

Taking a quick look at toZigbee.js in converts it appeared to have the power on change so I assume this was correct

danpowell88, 2018-12-28:
Just figured it out!

It was the comment in the json that was causing the issue, will test it now

danpowell88, 2018-12-28:
@Koenkk success, tested with recover and off and both worked as expected

danpowell88, 2018-12-28:
> Great! Do you also have color or color temperature bulbs? If yes, can you
> check if that is also recovered?

I'll give it a go

danpowell88, 2018-12-28:
Brightness if maintained but not colour, I'll have to pull out the sniffer again and see if something different is sent for the colour retention

danpowell88, 2018-12-28:
Also just a note along with comments it doesnt like a trailing comma on the final attribute

Koenkk, 2018-12-29:
And what is send for the normal behaviour?

Koenkk, 2019-01-09:
@Chrischi- great, could you make a pr?

Chrischi-, 2019-01-09:
@Koenkk 
any idea to put this into one converter?

if you switch on the behavior, you should have the possibility to set 
the color, color-temperature and brightness. since i don't have 
colored lamps (only color temperature and brightness), i can't 
create a converter for it

Koenkk, 2019-01-09:
I don't think that these converters should be merged together, they each serve a different purpose.

Could you prefix the keys with `hue_`? (as this is hue specific, e.g. `key: 'hue_power_on_color_temperature'],`

Ofcourse all the attributes can be set using 1 MQTT message:

```json
{
  "hue_power_on_behavior": "on",
  "hue_ power_on_level": 200,
  "hue_power_on_color_temperature": 300
}
```

closed 2019-03-11 by Koenkk

bepece1, 2019-10-22:
Also work right in Tuya, Tadfradi or Sonoff lights????

victoriano95, 2020-03-08:
Hi, I'm trying to set this option and it'is not working, these are the details:
Topic: zigbee2mqtt/friendlyName/set
Payload:
```
{
    "hue_power_on_behavior": "recover"
}
```
When I cut the power to the bulb and I restore it again it turns on.
The log output is:`{"ieeeAddr":"0x001788010424184b","type":"Router","networkAddress":14471,"model":"8718696548738","friendly_name":"0x001788010424184b","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Mains (single phase)","modelID":"LTW010","hardwareVersion":1,"softwareBuildID":"1.50.2_r30933","dateCode":"20191218","lastSeen":1583688566569}]}'`

Chrischi-, 2020-03-08:
> Hi, I'm trying to set this option and it'is not working, these are the details:
> Topic: zigbee2mqtt/friendlyName/set
> Payload:
> 
> ```
> {
>     "hue_power_on_behavior": "recover"
> }
> ```
> 
> When I cut the power to the bulb and I restore it again it turns on.

**recover - lamps on after power loss with last state**

victoriano95, 2020-03-08:
I want to recover or power off the bulbs when the power is restored but I when I send the command I can't see nothing on the log and it is not activating the option on the bulb

sjorge, 2020-03-17:
@victoriano95 after upgrading my bulb yesterday with the experimental hue update support.
I too thought it did not work, but it did for me. Before the bulb recovered to the previeus state it had to be without power for a good few seconds.

And indeed, there is no logging for the command.

victoriano95, 2020-03-17:
> @victoriano95 after upgrading my bulb yesterday with the experimental hue update support.
> I too thought it did not work, but it did for me. Before the bulb recovered to the previeus state it had to be without power for a good few seconds.
> 
> And indeed, there is no logging for the command.

Can you explain me what commands are you sending and what is the correct mqtt toppic? I have tried to leave the bulb a few minutes unplugged and it's still on when the power is recovered.
Thanks
@sjorge

sjorge, 2020-03-17:
In my case it was
```json
{
    "hue_power_on_behavior": "recover",
}
```

to `zigbee2mqtt/<friendlyname>/set`

But also pay attemtion to
```
Rules:

    * hue_power_on_behavior value always has to be provided
    * hue_power_on_brightness, hue_power_on_color_temperature and hue_power_on_color can only be provided when hue_power_on_behavior = on
    * hue_power_on_color_temperature and hue_power_on_color cannot be provided together, only one can be set
    * When setting hue_power_on_behavior = on, any not provided values will be reset to their factory defaults

```

victoriano95, 2020-03-17:
When you send this command the bulb turns off? it's ok the comma?
```
{
    "hue_power_on_behavior": "recover",
}
```
I'm sending the same command at the correct toppic, when I send it without comma it's not any output at the log and when I send it with the comma this is the output:
`MQTT publish: topic 'zigbee2mqtt/0x001788010424194b', payload '{"state":"OFF","brightness":0,"color_temp":451}'`

sjorge, 2020-03-17:
The comma was an error, I correct it above. When I sent it there was no response in the logs. That is normal.

I then turned off the bulb, cut the power to the bulb for 15sec and when I restored the power the bulb remained of.

I then turned on the bulb, cut the power to the bulb for 15sec and when I restored the power the bulb turned on again... so I think it is working.

victoriano95, 2020-03-23:
@sjorge Thanks, the problem was my zigbee2mqtt version, I have updated it and it works fine.

kedare, 2021-09-01:
Is there a way to set this using the web interface ? If not it's not very clear how to inject messages in the MQTT ? ('m using home assistant with the embedded mosquitto)

PostLogical, 2021-11-25:
> Is there a way to set this using the web interface ? If not it's not very clear how to inject messages in the MQTT ? ('m using home assistant with the embedded mosquitto)

Just went through this myself. I'm using Home Assistant in a docker container, so apologies if anything is different for HA OS or others, but you should be able to go to your Integrations page in Home Assistant and click on Configure for the MQTT integration. Then there are only a few things on the screen including "Publish a packet." Put in the topic as discussed above zigbee2mqtt/FRIENDLYNAME/set. And then put in the payload below that (see above for options). There isn't any feedback once you click Publish, so to confirm your command reached zigbee2mqtt, you might go to your Zigbee2MQTT interface and open the log and choose debug as log level before publishing the command. If you did it right, the command will appear in the log under debug (and only under debug).