Koenkk/zigbee2mqtt #749: Ikea floalt panel cant turn off tru HA frontend

pedrogense, 2018-12-22:
Ive added a ikea floalt panel 30x30, its reconized as the correct hardware.
In the frontend with auto discovery aswell manualy added to the config you cannot turn it off.

When i publish a mqtt message {"topic":"zigbee2mqtt/0x000b57fffe302f4b/set","payload":"off"} it turns off, when it turned it on tru the frontend in HA if comes on but it will not turn off.

Ive removed the device and readded the device multipe times but is doenst make a difference.

this is the log from zigbee2mqtt:
  zigbee2mqtt:info 2018-12-22 17:49:38 Zigbee publish to '0x000b57fffe302f4b', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-12-22 17:49:38 MQTT publish: topic 'zigbee2mqtt/0x000b57fffe302f4b', payload '{"state":"ON"}'
  zigbee2mqtt:info 2018-12-22 17:49:39 MQTT publish: topic 'zigbee2mqtt/0x000b57fffe302f4b', payload '{"state":"ON"}'
  zigbee2mqtt:info 2018-12-22 17:49:41 Zigbee publish to '0x000b57fffe302f4b', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-12-22 17:49:41 MQTT publish: topic 'zigbee2mqtt/0x000b57fffe302f4b', payload '{"state":"ON"}'

looks like its not given the proper command to turn it off.

im running HA 0.83.2

khzd, 2018-12-22:
hi
please post the 0x000b57fffe302f4b DB record from database.db

pedrogense, 2018-12-22:
{"id":13,"type":"Router","ieeeAddr":"0x000b57fffe302f4b","nwkAddr":54311,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"FLOALT panel WS 30x30","epList":[1],"status":"online","joinTime":1545493844,"endpoints":{"1":{"profId":260,"epId":1,"devId":544,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"10":{},"65533":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"FLOALT panel WS 30x30","dateCode":"20160927","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"65533":1,"nameSupport":0}},"genScenes":{"dir":{"value":3},"attrs":{"65533":1,"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"16387":1,"65533":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"15":0,"16384":254,"65533":1,"currentLevel":254,"remainingTime":0,"onOffTransitionTime":5,"onLevel":255}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{"15":0,"16397":0,"16398":370,"65533":1,"remainingTime":0,"currentX":30138,"currentY":26909,"colorTemperature":370,"colorMode":2,"numPrimaries":0,"enhancedColorMode":2,"colorCapabilities":16,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}},"haDiagnostic":{"dir":{"value":1},"attrs":{"65533":1,"numberOfResets":0,"macRxBcast":0,"macTxBcast":0,"aPSRxBcast":0,"aPSTxBcast":0,"nwkFcFailure":0,"packetBufferAllocateFailures":0,"averageMacRetryPerApsMessageSent":0,"lastMessageLqi":220,"lastMessageRssi":-45}},"lightLink":{"dir":{"value":3},"attrs":{"65533":1}}}}},"_id":"gV6PyuqzKn93UgOO"}
{"id":13,"type":"Router","ieeeAddr":"0x000b57fffe302f4b","nwkAddr":54311,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"FLOALT panel WS 30x30","epList":[1],"status":"online","joinTime":1545493844,"endpoints":{"1":{"profId":260,"epId":1,"devId":544,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"10":{},"65533":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"FLOALT panel WS 30x30","dateCode":"20160927","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"65533":1,"nameSupport":0}},"genScenes":{"dir":{"value":3},"attrs":{"65533":1,"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"16387":1,"65533":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"15":0,"16384":254,"65533":1,"currentLevel":254,"remainingTime":0,"onOffTransitionTime":5,"onLevel":255}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{"15":0,"16397":0,"16398":370,"65533":1,"remainingTime":0,"currentX":30138,"currentY":26909,"colorTemperature":370,"colorMode":2,"numPrimaries":0,"enhancedColorMode":2,"colorCapabilities":16,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}},"haDiagnostic":{"dir":{"value":1},"attrs":{"65533":1,"numberOfResets":0,"macRxBcast":0,"macTxBcast":0,"aPSRxBcast":0,"aPSTxBcast":0,"nwkFcFailure":0,"packetBufferAllocateFailures":0,"averageMacRetryPerApsMessageSent":0,"lastMessageLqi":236,"lastMessageRssi":-41}},"lightLink":{"dir":{"value":3},"attrs":{"65533":1}}}}},"_id":"gV6PyuqzKn93UgOO"}

khzd, 2018-12-22:
Hi 
According to your log, HA is correctly integrated.

pedrogense, 2018-12-22:
I can use it when i publish trough mqtt, its the frontend wich doenst work. No correct status or on/off.
You have any idea where to look?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

pedrogense, 2018-12-23:
its done bij autodiscovery in ha atm, but when i added it it was as a light.

khzd, 2018-12-24:
There is a limitation in ha with light
I manges it by declaring the device as light , it s work fine

pedrogense, 2018-12-24:
What do you mean exactly? Do i need to define it as a switch in ha?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

khzd, 2018-12-25:
No as light in HA: explicitly declare the device as light and fill all Mqtt proprtities. Will post my full implementation soon

ripperle, 2019-01-20:
@pedrogense
I have also a ikea floalt panel (30x60) and I can't even pair it with zigbee2mqtt... I don't receive any message from the lamp.. Can you describe how do you paired it with zigbee2mqtt?

pedrogense, 2019-02-14:
> @pedrogense
> I have also a ikea floalt panel (30x60) and I can't even pair it with zigbee2mqtt... I don't receive any message from the lamp.. Can you describe how do you paired it with zigbee2mqtt?

You need to bring it whitinn 20cm distance tot the zigbee stick

ripperle, 2019-03-10:
After i Updated the coordinator to the latest Firmware (Version january 2019) the panel paired immediately after a reset (switch light on and off 6 times)
It was not necessary to get close to the coordinator... The light is in an another room...

closed 2019-05-16 by stale[bot]