Koenkk/zigbee2mqtt #751: Gledopto GL-C-007 was not recognized

SoSv73, 2018-12-22:
Finally I got my Gledopto GL-C-007 up and running.
After several pairing attempts and following instructions on https://github.com/Koenkk/zigbee2mqtt/issues/608, I had no success to get this running. But that Issue was a good base to start from.

Now, I didn't change the List of endpoints, but changed some values in the entries. I still have Endpoints 11,12,13 active.

After studying https://www.nxp.com/docs/en/user-guide/JN-UG-3091.pdf I found that the devId for EP 12 is wrong. 258 is not a vaild Device ID according to the Zigbee User guide. I changed it to 256 (0x100) for the dimmable Light.
EP 13 was a little bit strange as well. The value for the devId was 57694 (0xE15E). This value is undefined and I corrected it to 2064 (0x810) for a ZLL commissioning device "Colour scene controller". I dont't know if it makes any sense to add the Basic cluster to the in- and outClusterList, but it seemed just reasonable to change it as well.

I copied the Geldopto entry for GL-C-008 and renamed it for the GL-C-007 to GLEDOPTA and added the missing part of the Manufacturer to the entries in the devices.js and made the same change in homeassistant.js.

The changes I made after pairing th GL-C-007:
Database.db
Before:
{"id":18,"type":"Router","ieeeAddr":"0x00124b001ada93d0","nwkAddr":52870,"manufId":0,"epList":[11,12,13],"status":"online","joinTime":1545503232,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{}}},"_id":"gytfse17QDt1Pvg5"}
{"id":18,"type":"Router","ieeeAddr":"0x00124b001ada93d0","nwkAddr":52870,"manufId":0,"epList":[11,12,13],"status":"online","joinTime":1545503232,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"gytfse17QDt1Pvg5"}

After:
{"id":18,"type":"Router","ieeeAddr":"0x00124b001ada93d0","nwkAddr":52870,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTOA","epList":[11,12,13],"status":"online","joinTime":1545503232,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"12":{"profId":260,"epId":12,"devId":256,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"13":{"profId":49246,"epId":13,"devId":2064,"inClusterList":[0,4096],"outClusterList":[0,4096],"clusters":{}}},"_id":"gytfse17QDt1Pvg5"}
{"id":18,"type":"Router","ieeeAddr":"0x00124b001ada93d0","nwkAddr":52870,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTOA","epList":[11,12,13],"status":"online","joinTime":1545503232,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"12":{"profId":260,"epId":12,"devId":256,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":2064,"inClusterList":[0,4096],"outClusterList":[0,4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"gytfse17QDt1Pvg5"}

Devices.js
// Gledopto GL-C-007
    {
        zigbeeModel: ['GLEDOPTOA', 'GL-C-007'],
        model: 'GL-C-007',
        vendor: 'GLEDOPTO',
        description: 'Zigbee LED controller RGB + CCT / RGBW / WWCW / Dimmer',
        supports: generic.light_onoff_brightness_colortemp_colorxy().supports,
        fromZigbee: generic.light_onoff_brightness_colortemp_colorxy().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colortemp_colorxy().toZigbee,
        ep: (device) => {
            if (device.epList.toString() === '11,12,13') {
                return {'': 12};
            } else if (device.epList.toString() === '10,11,13') {
                return {'': 11};
            } else {
                return {};
            }
        },
    },

Homeassistant.js
    'GL-C-007': [configurations.light_brightness_colortemp_colorxy],

Configuration.yaml
  - platform: "mqtt_json"
    name: LED_strip
    state_topic: "zigbee2mqtt/0x00124b001ada93d0"
    availability_topic: "zigbee2mqtt/bridge/state"
    brightness: true
    color_temp: true
    xy: true
    command_topic: "zigbee2mqtt/0x00124b001ada93d0/set"

Now I got the GL-C-007 integrated in my Homeassistant environment. I can change the colour, colour temperature, brightness and controll on/off.

khzd, 2018-12-23:
Hi,
this is a good progress in understanding the DB parameters :)

did you have chance to play with scene ? by HA we can send a scene code ?
ie
...
effect: true
    #Effect list 0=Full, 1=White, 2=RGB, 3=Full hard, 4=Disco, 5=Red Flash, 6=Green Flash, 7=Blue Flash, 8=White Flash]
    effect_list: [0, 1, 2, 3, 4 ,5, 6, 7, 8] 
...
I gived a try but zigbe2mqtt send un error : **AM No converter available for 'effect' (5)**

I you didnt' change the device code, you may just add the bold part
zigbeeModel: ['GLEDOPTO', 'GL-C-008' , **'GLEDOPTOA', 'GL-C-007'**], ?

ArJay60, 2018-12-23:
None of the GLEDOPTO devices is recognized for type. Same is true for GL-S-008Z. Only manual changes to database,db but this wil give GL-C-008 as model. Can something be done for these devices?

SoSv73, 2018-12-23:
Hi khzd,
I was just glad to get the device running. I think HA doesn't support scenes in a Zugbee device. I just chose the devID as Color Scene controller becuase the old DevID (0xE15E) was completely out of the valid range. I am not sure if the Gledopto device will support Colour scenes.

Regards

Sven

zipbee, 2019-01-21:
Hi,

I've recently purchased a GL-C-007 as well and am having issues getting it to work, when first paired I get the following log entries:
`2019-1-21 18:39:08 - info: Device incoming...
2019-1-21 18:39:08 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
2019-1-21 18:39:08 - warn: Device with modelID 'undefined' is not supported.
2019-1-21 18:39:08 - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
2019-1-21 18:39:08 - warn: Device with modelID 'undefined' is not supported. 
2019-1-21 18:39:08 - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html`

And I end up with the following entries in my database.db:
`{"id":6,"type":"Router","ieeeAddr":"0x00124b001ada63c8","nwkAddr":17051,"manufId":0,"epList":[11,12,13,15],"status":"online","joinTime":1548095736,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{}},"15":{"profId":49246,"epId":15,"devId":256,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}}},"_id":"BfeqOslPJkzS78N3"}
{"id":6,"type":"Router","ieeeAddr":"0x00124b001ada63c8","nwkAddr":17051,"manufId":0,"epList":[11,12,13,15],"status":"online","joinTime":1548095736,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}},"15":{"profId":49246,"epId":15,"devId":256,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}}},"_id":"BfeqOslPJkzS78N3"}`

Mine is supposed to be a RGBW controller, and I think it should show up as 2 separate lights, one to control the RGB set of lights and one to control the (warm) white lights, at least that is what it suggests would happen when linked to a Hue Bridge.

My database entries look very similar to the ones from original post, but my epList also has "15" in it.

Any suggestions for how to try and get this to be recognised as 2 lights?

Thanks in advance!

zipbee, 2019-01-23:
Further to my comment above, I've been doing some experimenting, I'm running the latest-dev container.

In database.db I added modelId and then changed the devId for ep12 to 258, devId for ep13 to 2064. That in itself resulted in the device now being recognised when zigbee2mqtt starts and no longer shows them as undefined, but it didn't respond to any commands.

I then updated existing 'GL-C-007' entry in 'zigbee-shepherd-converters/devices.js' to account for my extra endpoint 15 (line 1538):
            if (device.epList.toString() === '11,12,13' || device.epList.toString() === '11,12,13,15') {

Now I can control the lights through Home Assistant, or at least I can change the brightness and RGB colour of the RGB LEDs, but I still have no way to control the separate white LEDs on the strip / controller.

Should I raise this as a separate issue rather than add to this one? It looks like there are a lot of variants of Gledopto's devices and they all behave slightly differently with different endpoints, I think mine must be different from the original comment. 

Thanks

Koenkk, 2019-01-23:
@zipbee for your case, undo the change you made and please add the following

```js
if (device.epList.toString() === '11,12,13,15') {
   return {
       'rgb': 12,
       'white': 13     // if white doesn't work also try with 11 and 15 instead of 13
   };
}
```

You should be able to control the 2 channels with `zigbee2mqtt/[DEVICE_ID]/rgb/set` and `zigbee2mqtt/[DEVICE_ID]/white/set`

zipbee, 2019-01-23:
I've tried that but I'm guessing I might need to take some additional steps to make this work? I get this in the logs:
`2019-1-23 21:01:10 - error: Failed to find device with ieeAddr: '0x00124b001ada63c8/white'`
When running:
`mosquitto_pub -h localhost -t "zigbee2mqtt/0x00124b001ada63c8/white/set" -m '{ "brightness": "255", "state":"on" }'`

I can still control the RGB LEDs using just "zigbee2mqtt/0x00124b001ada63c8/set" as the topic.

Koenkk, 2019-01-24:
@zipbee my bad, you also need to add `white` and `rgb` to https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/devicePublish.js#L9

zipbee, 2019-01-25:
Thank you, only had a few minutes to do some quick testing earlier. 

I can definitely control both independently now which is great, the epId for the RGB LEDs is definitely 12 and the epId for the white LEDs is 15.

The only issue I noticed is that there is sometimes a very long delay between publishing the topic and the actual lights changing. I will need to do some more testing to see what's causing that.

When I have some more time over the next few days I will start from scratch to undo all my changes, re-pair the device, and confirm exactly what is needed to make it work again, I can then submit that as pull requests.

zipbee, 2019-01-28:
I've just been able to do some testing and created some pull requests. Those are the only changes needed to the sources to make the 2 lights work, the only additional thing I had to do after re-pairing was manually add the modelId back into the database.db, I did not have to make any changes to devId etc.

Also the slow response issues I had been seeing were definitely fixed by the clogging fixes previously mentioned.

closed 2019-04-05 by stale[bot]