Koenkk/zigbee2mqtt #752: force device to connect with router instead of coordinator

trekker25, 2018-12-23:
is there a way to have an end device (aqara motion sensor) to connect to the nearest router instead of the coordinator?

The stick is in my utillity closet and next to it is my toilet with a Hue bulb (router) and Aqara motion sensor. The router and sensor are only 30 cm away from each other and it would be logical the sensor connects to the router instead of coordinator (thick wall between). 

Now there is a link quality of 20 - 30 for the motion sensor, and connecting to the router would make more sense.

Battery in and out? Or how to solve this?

lolorc, 2018-12-28:
is pairing through routers working ?

Koenkk, 2018-12-28:
This is hard to say, depends on the router.

h4nc, 2018-12-29:
Does it work with a CC2530 router?

Koenkk, 2018-12-29:
I`m not sure, @ptvoinfo is this supported?

nickrout, 2019-01-10:
Can anyone tell me if it work to pair to a router with a Xiaomi Power Plug ZNCZ02LM?

hvddrift, 2019-01-18:
I also would like to know the answer to this one.

> Can anyone tell me if it work to pair to a router with a Xiaomi Power Plug ZNCZ02LM?

ptvoinfo, 2019-01-19:
@hvddrift Both devices are routers and should be paired with a coordinator. You cannot pair a router to a router.

hvddrift, 2019-01-19:
I am still trying to understand how this should work...

I have the following network
[image]

The Motion sensor is 1 meter away from a Xiaomi Plug (router), and yet it joins back to the controller with a link strenth of 1. Is this intended? Should it try to join the router (I assume the signal strength would be stronger)

Koenkk, 2019-01-19:
@hvddrift I would recommend updating to the latest dev branch, this will show the mesh network better (as one node can be connected to multiple)

hvddrift, 2019-01-19:
Noob, question. Are there instructions for updating to the dev branch?

Thanks
dank je

hvddrift, 2019-01-19:
I worked it out.. Found another thread. 

So, this is my network map now. Has some more information. But the Motion Sensor still bypasses the router. What makes the device connect to a router or direct to the controller?

[image]

Koenkk, 2019-01-19:
@hvddrift the device self decides about this, sometimes it helps to repair the device while having it close to the router you want to pair it to.

hvddrift, 2019-01-20:
And another basic question... To "repair" a device simply means to remove it and repair it?

h4nc, 2019-01-20:
Yes, remove it (you can find how to remove files in the docs) and than pair it again.

closed 2019-03-28 by stale[bot]

tuinstoelen, 2022-12-05:
> Yes, remove it (you can find how to remove files in the docs) and than pair it again.

This comment finally allowed me to solve my Aqara Temperature sensors not connecting to a router, but only to the main hub. It turns out, that if i first explicitly remove the sensor in MQTT, then allow pairing only to a specific router (by using the drop down button), the sensor would connect to the router.

It seems like the sensor remembers which hub/router it was connected to, and will not connect to any other router if you just press the button for 7(?) seconds. First explicitly removing it from MQTT solved this for me.