Koenkk/zigbee2mqtt #755: Question: Zigbee stick new firmware

h4nc, 2018-12-24:
I saw that there is a new firmware version vor cc2531. 

newest version: 20181024
installed version: 20180515

Should I upgrade my firmware?
Do I have to reconnect all devices again when I load a new firmware to the stick? So is the connection stored at the stick or in the hassio addon?

ryanbeaton, 2018-12-24:
You will need to re-pair after upgrade. There is another way with a different compile of the new firmware in one of the issues here. You don’t even need to use the Debugger with that compile 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

ryanbeaton, 2018-12-24:
https://github.com/Koenkk/zigbee2mqtt/issues/543#issuecomment-433951590

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

h4nc, 2018-12-25:
Ok, thanks.

Shipley normal users update their Stick with every new firmware update.

I have a debugger. Is there a benefit not using it or only if you don’t have one?

Ingo1977, 2019-01-14:
I have the same question.
I don't have a Debugger but want to update my stick.
How can I do the upgrade to the latest firmware?
Thanks.

Koenkk, 2019-01-14:
@Ingo1977 https://koenkk.github.io/zigbee2mqtt/information/alternative_flashing_methods.html

closed 2019-03-22 by stale[bot]