Koenkk/zigbee2mqtt #762: Osram Plug Smart+ connected via Philips Hue Bulb as router

chrisdaloa, 2018-12-26:
I have a Osram Plug Smart+ linked to a Philips Hue Bulb that act as router.

When I try to turn on or turn off it It return this error, both when I use the home assistant switch and when I use MQTT topics
 
> Dec 27 00:08:06 hassbian npm[29985]:   zigbee2mqtt:error 27/12/2018, 00:08:06 Zigbee publish to '0x7cb03eaa0a01ead5', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.

The Plug is connected and visibile as shown in this pictures.

[image]

I tried to delete the db, the config and I tried to pair again all devices but without results.

Thanks

Koenkk, 2018-12-27:
Does it work when moving it closer to the hue bulb? Keep in mind that the network map is not 100% accurate. 255 is also suspicious link quality as this 100% (which is practically not possible).

chrisdaloa, 2018-12-27:
The closest attach I have is at lessa than 3 mt. Do you think it's too far?.
 If I put the plug between the bulb and the coordinator, the bulb connect itself to the plug and all works 
[image]
well.

Koenkk, 2018-12-29:
That would be strange, but keep in mind that zigbee devices can also communicate in a star and tree topology. Something which is not shown in the network map. Unless you are having problems I wouldn't care too much about the networkmap.

chrisdaloa, 2018-12-29:
My problem is that it does not work where I need it, but It works in other places so it's not a zigbee2mqtt problem, I think :)
Thank you.

Koenkk, 2018-12-29:
Perhaps there is a lot of interference. https://support.metageek.com/hc/en-us/articles/203845040-ZigBee-and-WiFi-Coexistence explains it quite well.

closed 2019-03-07 by stale[bot]