Koenkk/zigbee2mqtt #763: cc2531 router not staying connected

danpowell88, 2018-12-27:
Has anyone else had an issue with the cc2531 router not staying connected to the co-ordinator?

I can connect them and then connect a xiaomi motion sensor which appears to work for a little bit and shows as connected in the network map, then a few minutes later router is showing 0 connection strength and the xiaomi sensor is showing as online but not connected to anything and is not broadcasting movement events.

The logs also dont show the router doing anything, it checks if its online every minute or so but I dont see the same kind of responses from the router that I did just after the initial pairing.

danpowell88, 2018-12-27:
Router is showing the following from debug

```
46846/0x00158D0001E0A706,CHILD_RFD,Depth:2,Assoc:0
0/0x00124B0014BDEB58,COORD,Depth:0,Assoc:0
46846/0x00158D0001E0A706,CHILD_RFD,Depth:2,Assoc:0
0/0x00124B0014BDEB58,COORD,Depth:0,Assoc:0
46846/0x00158D0001E0A706,CHILD_RFD,Depth:2,Assoc:0
0/0x00124B0014BDEB58,COORD,Depth:0,Assoc:0
46846/0x00158D0001E0A706,CHILD_RFD,Depth:2,Assoc:0

```

Is the 0x for the co-ordinator possibly the issue here?

@ptvoinfo @Koenkk ?

ptvoinfo, 2018-12-27:
@danpowell88 "0x" is expected in the debug output. This output appears only if the router successful paired and connected with the coordinator. The sensor may find a better path to your coordinator. Do you have only three devices in your network (coordinator, router, sensor)?

danpowell88, 2018-12-27:
I've got a few more but the one showing there only was able to connect to
the router when I re pared it after joining the router, still no debug
output showing on the coordinator about the router

danpowell88, 2018-12-27:
> @danpowell88 <https://github.com/danpowell88> What do you expect on the
> coordinator? The router sends a state and some meta information.

I was seeing logs on the coordinator of what devices were connected through
the router, but they've since stopped

ptvoinfo, 2018-12-27:
@danpowell88 I don't have ideas.

orientalsniper, 2019-01-01:
I'm also having problem with my Xiaomi Sensors (they are the farthest) and a CC2531 router, sometimes the sensors are able to send data. It was working fine running stable zigbee2mqtt and firmware, yesterday I flashed the dev coordinator firmware and updated to dev zigbee2mqtt, but there's no dev router firmware.

Is router firmware incompatible with dev coordinator firmware? (https://github.com/Koenkk/Z-Stack-firmware)

closed 2019-01-09 by danpowell88

maymaymay, 2019-01-18:
I'm getting this too:
`AM No converter available for 'CC2530.ROUTER' with cid 'genOnOff', type 'devChange' and data '{"cid":"genOnOff","data":{"onOff":1}}'`

Koenkk, 2019-01-18:
@maymaymay that will be fixed in the next zigbee2mqtt version.