Koenkk/zigbee2mqtt #764: Testing: groups support

Koenkk, 2018-12-27:
This issue is used to gather feedback on the group support feature (#15).

Notes:
- Documentation is available [here](https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/information/groups.md)
- [Integrating with Home Assistant](https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/integration/home_assistant.md#groups)
- At least firmware version `20181224` is required! (available [here](https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator))

dcshoes23, 2018-12-27:
Thanks for this! I'm having one issue, when I call zigbee2mqtt/bedroom_lights/set, I get the following error:

`Failed to find device with ieeAddr: 'bedroom_lights'`

dcshoes23, 2018-12-27:
I used zigbee2mqtt/group/bedroom_lights/set instead and it works! Awesome job!

Koenkk, 2018-12-27:
@dcshoes23 please update to the latest dev branch, ` zigbee2mqtt/group/bedroom_lights/set` is the "old" api (I've changed it a few hours ago).

dcshoes23, 2018-12-27:
@Koenkk Just pulled the latest-dev now and it works with `zigbee2mqtt/bedroom_lights/set`, thanks!

subzero79, 2018-12-27:
This new fw, does it reduce the number of devices the coordinator can hold ?

Koenkk, 2018-12-27:
@subzero79 the CC2531 firmware released together with zigbee2mqtt 1.0.0 had a limit of 20, the new firmware has a limit of 15 (because the stack has to be increased for stability reason). 

In case you **only** have battery powered devices you can use the max devices firmware which has a limit of 44 devices.

dcshoes23, 2018-12-27:
@Koenkk `zigbee2mqtt/bridge/groups/[GROUP_FRIENDLY_NAME]/add` worked with the old version and now it doesn't do anything. There is nothing shown in the logs

Koenkk, 2018-12-27:
@dcshoes23 it's `zigbee2mqtt/bridge/group/[GROUP_FRIENDLY_NAME]/add` (group with **s**)

dcshoes23, 2018-12-27:
@Koenkk that worked, thanks!

trekker25, 2018-12-28:
looks awesome! setting about 15 GU10 spots so groups are welcome!

subzero79, 2018-12-30:
How likely is the fw to be changed again in the future?  
For example to be able to poll an attribute from a device on demand (like xiaomi plugs power, waiting for that feature) when is implemented do you think it will need fw rebuild?
 Having to repair all devices is not something I look forward to.

Koenkk, 2018-12-30:
@subzero79 polling attributes doesn't require a firmware update.

trekker25, 2019-01-01:
How does this integrate in best way with Hass?

Koenkk, 2019-01-01:
@trekker25 the OP contains a link on how to do this

trekker25, 2019-01-01:
Sorry how could I miss that. In 6 weeks my new kitchen is ready and then can work with it. 
Group of 4, another group of 4 and a group of 9 gu10 spots. Curious how this will work out as I only work with wireless switches and sensors and not adding physical switches.

STcraft, 2019-01-01:
Hello,
Can i flash the .hex file (extracted from the zip in https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin/) with the alternative flashing method? (Via Arduino, described here: https://koenkk.github.io/zigbee2mqtt/information/alternative_flashing_methods.html)
Thanks in advance!

Koenkk, 2019-01-01:
@stcraft If I remember correctly, no

STcraft, 2019-01-04:
I've done a bit of research and testing.
It seems removing this line near the bottom of the .hex file:
`:0400000500002DD0FA`
will do the trick to get the file to flash. (otherwise, the script will spit an error out mentioning something about `05`)

I found this out after diff-ing the two hex files (one modded and the other not, Located here: https://github.com/kirovilya/files) 
This was the only change being made.

subzero79, 2019-01-07:
I still haven't flashed the group fw, i have 4 plugs in the attic, overall 24 devices. How does the group works if you want to change brightness or color temp? or it can just control on/off?

Also is it possible to add a group to another group? or can one device be in more than one group?

Koenkk, 2019-01-07:
@subzero79 its the same as controlling a single device, meaning that color and color temperature can also be changed via a group.

A device can be in multiple groups.

dreimer1986, 2019-01-09:
Is it just me or does switching on/off a group not refresh the status of the light bulbs in Home Assistant? If I switch on group "Living Room" the four lights "Living1" - "Living4" still say that they are switched off. When I try to use the switch in Home Assistant on a light in the group then I get these lines from time to time:

2019-1-9 01:12:55 - info: Zigbee publish to device '0x7cb03eaa00a1184a', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-1-9 01:12:58 - error: Zigbee publish to device '0x7cb03eaa00a1184a', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout

In the other cases it just changes the switch to the now correct setting (ON) and I can switch the light off after the extra step.

Koenkk, 2019-01-09:
@dreimer1986 this is indeed not updated, added this as a todo to the OP.

dreimer1986, 2019-01-10:
Thx for fixing it in advance ^^Did not test if color values etc are synced, but I guess not.

timota, 2019-01-16:
unfortunately it wont work for me. After stopping zigbee2mqtt need to unplug CC2531 and press reset button to start it again. even with that it can start only with clear config without devices. 
With clean config (without devices) i managed to add 3 xiaomi enddevices. but after restart wont start again.

log below

> zigbee2mqtt@1.0.1 start /opt/zigbee2mqtt-dev
> node index.js

  zigbee2mqtt:info 2019-1-16 20:19:52 Logging to directory: 'data/log/2019-01-16.20-19-52'
  zigbee2mqtt:debug 2019-1-16 20:19:52 Removing old log directory 'data/log/2019-01-16.20-08-26'
  zigbee2mqtt:debug 2019-1-16 20:19:53 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11],"precfgkey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]},"dbPath":"/opt/zigbee2mqtt-dev/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 2019-1-16 20:19:53 Loaded state from file /opt/zigbee2mqtt-dev/data/state.json
  zigbee2mqtt:info 2019-1-16 20:19:53 Starting zigbee2mqtt version 1.0.1 (commit #28e3288)
  zigbee2mqtt:info 2019-1-16 20:19:53 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-1-16 20:19:54 zigbee-shepherd started
  zigbee2mqtt:info 2019-1-16 20:19:54 Coordinator firmware version: '20181224'
  zigbee2mqtt:debug 2019-1-16 20:19:54 zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0018ecde53","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20181224},"startTime":1547669994,"joinTimeLeft":0}
  zigbee2mqtt:info 2019-1-16 20:19:54 Currently 3 devices are joined:
  zigbee2mqtt:info 2019-1-16 20:19:54 0x00158d000274fe84 (0x00158d000274fe84): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
/opt/zigbee2mqtt-dev/node_modules/q/q.js:155
                throw e;
                ^

TypeError: Cannot read property 'replace' of undefined
    at Object.findByZigbeeModel (/opt/zigbee2mqtt-dev/node_modules/zigbee-shepherd-converters/index.js:20:46)
    at Controller.getDeviceStartupLogMessage (/opt/zigbee2mqtt-dev/lib/controller.js:193:54)
    at devices.forEach (/opt/zigbee2mqtt-dev/lib/controller.js:82:30)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeStarted (/opt/zigbee2mqtt-dev/lib/controller.js:81:17)
    at zigbee.start (/opt/zigbee2mqtt-dev/lib/controller.js:149:26)
    at shepherd.start (/opt/zigbee2mqtt-dev/lib/zigbee.js:64:17)
    at /opt/zigbee2mqtt-dev/node_modules/q/q.js:2055:17
    at runSingle (/opt/zigbee2mqtt-dev/node_modules/q/q.js:137:13)
    at flush (/opt/zigbee2mqtt-dev/node_modules/q/q.js:125:13)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.0.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.0.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-01-16T20_19_54_639Z-debug.log

is it something with dongle ? or with fw or edge version ? 

Thanks

Koenkk, 2019-01-17:
After updating to the dev branch, did you do a `rm -rf node_modules && npm install`?

timota, 2019-01-17:
what i did: 
- completely remove zigbee2mqtt directory (prod and dev)
- git clone DEV
- npm install & and all necessary steps described on the wiki
- reflash CC2531
- reboot NUC

and after all these steps it starts working again.  
i see 2 issues: 
   linux kernel somehow stuck with module, didnt check if it was loaded before reboot. 
   incorrectly and fully switch dev and prod for zigbee2mqtt

glentakahashi, 2019-01-23:
So I'm running the latest dev branch (as of 21:00 1/22) using the standard cc2531 firmware, and I started noticing some strange behavior with groups.

On an ungrouped light, I can toggle the light on/off as fast as humanly possible and zigbee2mqtt + cc2531 will chug along happily. However, if I take the same light and add it to a 1 device group, i can barely toggle the light 4 times before the cc2531's buffer will fill up (Keeps throwing rsp error: 17) and stop taking any more requests.

Any ideas on how to improve this?

```
zigbee2mqtt      | 2019-01-23T02:05:54.965Z serialport:bindings read                                                                                                                                              [153/83079]
zigbee2mqtt      | 2019-01-23T02:05:54.966Z serialport:unixRead Starting read
zigbee2mqtt      | Wed, 23 Jan 2019 02:05:54 GMT cc-znp:SRSP <-- AF:dataRequestExt, { status: 17 }
zigbee2mqtt      | 2019-01-23T02:05:54.967Z zigbee-shepherd:request RSP <-- AF:dataRequestExt, status: 17
zigbee2mqtt      |   zigbee2mqtt:error 2019-1-23 02:05:54 Zigbee publish to group '2', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: rsp error: 17
zigbee2mqtt      | 2019-01-23T02:05:54.973Z serialport:unixRead waiting for readable because of code: EAGAIN
zigbee2mqtt      | 2019-01-23T02:05:54.973Z serialport:poller Polling for "readable"
zigbee2mqtt      |   zigbee2mqtt:debug 2019-1-23 02:05:55 Received MQTT message on 'zigbee2mqtt/office_lights/set' with data '{"state": "ON"}'
zigbee2mqtt      |   zigbee2mqtt:info 2019-1-23 02:05:55 Zigbee publish to group '2', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
zigbee2mqtt      | 2019-01-23T02:05:55.110Z zigbee-shepherd:request REQ --> AF:dataRequestExt, transId: 186
zigbee2mqtt      | Wed, 23 Jan 2019 02:05:55 GMT cc-znp:SREQ --> AF:dataRequestExt, { dstaddrmode: 1, dstaddr: '0x0000000000000002', destendpoint: 255, dstpanid: 0, srcendpoint: 1, clusterid: 6, transid: 186, options: 32, radius: 30, len: 3, data: <Buffer 01 b9 01> }
zigbee2mqtt      | 2019-01-23T02:05:55.115Z serialport:main _write 28 bytes of data
zigbee2mqtt      | 2019-01-23T02:05:55.115Z serialport:bindings write 28 bytes
zigbee2mqtt      | 2019-01-23T02:05:55.117Z serialport:unixWrite Starting write 28 bytes offset 0 bytesToWrite 28
zigbee2mqtt      | 2019-01-23T02:05:55.118Z serialport:unixWrite write returned null 28
zigbee2mqtt      | 2019-01-23T02:05:55.119Z serialport:unixWrite wrote 28 bytes
zigbee2mqtt      | 2019-01-23T02:05:55.120Z serialport:unixWrite Finished writing 28 bytes
zigbee2mqtt      | 2019-01-23T02:05:55.122Z serialport:main binding.write write finished
zigbee2mqtt      | 2019-01-23T02:05:55.124Z serialport:poller received "readable"
zigbee2mqtt      | 2019-01-23T02:05:55.124Z serialport:bindings read
zigbee2mqtt      | 2019-01-23T02:05:55.125Z serialport:unixRead Starting read
zigbee2mqtt      | 2019-01-23T02:05:55.125Z serialport:unixRead Finished read 6 bytes
zigbee2mqtt      | 2019-01-23T02:05:55.126Z serialport:main binding.read finished
zigbee2mqtt      | Wed, 23 Jan 2019 02:05:55 GMT cc-znp { sof: 254,
zigbee2mqtt      |   len: 1,
zigbee2mqtt      |   type: 'SRSP',
zigbee2mqtt      |   subsys: 'AF',
zigbee2mqtt      |   cmd: 'dataRequestExt',
zigbee2mqtt      |   payload: { status: 17 },
zigbee2mqtt      |   fcs: 118,
zigbee2mqtt      |   csum: 118 }
zigbee2mqtt      | 2019-01-23T02:05:55.131Z serialport:main _read reading
zigbee2mqtt      | 2019-01-23T02:05:55.131Z serialport:bindings read
zigbee2mqtt      | 2019-01-23T02:05:55.132Z serialport:unixRead Starting read
zigbee2mqtt      | Wed, 23 Jan 2019 02:05:55 GMT cc-znp:SRSP <-- AF:dataRequestExt, { status: 17 }
zigbee2mqtt      | 2019-01-23T02:05:55.135Z zigbee-shepherd:request RSP <-- AF:dataRequestExt, status: 17
zigbee2mqtt      |   zigbee2mqtt:error 2019-1-23 02:05:55 Zigbee publish to group '2', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: rsp error: 17
zigbee2mqtt      | 2019-01-23T02:05:55.142Z serialport:unixRead waiting for readable because of code: EAGAIN
zigbee2mqtt      | 2019-01-23T02:05:55.144Z serialport:poller Polling for "readable"
zigbee2mqtt      |   zigbee2mqtt:debug 2019-1-23 02:05:56 Check online 0x00158d000237dd9b 0x00158d000237dd9b
zigbee2mqtt      |   zigbee2mqtt:debug 2019-1-23 02:05:56 Check online 0x00158d000237dc1a 0x00158d000237dc1a
zigbee2mqtt      |   zigbee2mqtt:debug 2019-1-23 02:05:56 Check online 0x00158d000237ddb6 0x00158d000237ddb6
zigbee2mqtt      |   zigbee2mqtt:debug 2019-1-23 02:05:56 Check online 0x00158d000237dc52 0x00158d000237dc52
zigbee2mqtt      | 2019-01-23T02:05:56.466Z zigbee-shepherd:request REQ --> ZDO:nodeDescReq
zigbee2mqtt      | Wed, 23 Jan 2019 02:05:56 GMT cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 9722, nwkaddrofinterest: 9722 }
zigbee2mqtt      | 2019-01-23T02:05:56.470Z serialport:main _write 9 bytes of data
zigbee2mqtt      | 2019-01-23T02:05:56.470Z serialport:bindings write 9 bytes
zigbee2mqtt      | 2019-01-23T02:05:56.471Z zigbee-shepherd:request REQ --> ZDO:nodeDescReq
zigbee2mqtt      | 2019-01-23T02:05:56.472Z zigbee-shepherd:request REQ --> ZDO:nodeDescReq
zigbee2mqtt      | 2019-01-23T02:05:56.473Z zigbee-shepherd:request REQ --> ZDO:nodeDescReq
zigbee2mqtt      | 2019-01-23T02:05:56.475Z serialport:unixWrite Starting write 9 bytes offset 0 bytesToWrite 9
zigbee2mqtt      | 2019-01-23T02:05:56.475Z serialport:unixWrite write returned null 9
```

kirovilya, 2019-01-23:
Has anyone tested the work of groups with OSRAM devices (lamps and plugs)? we can't get them to work through groups. https://github.com/ioBroker/ioBroker.zigbee/issues/153

subzero79, 2019-01-23:
I have 4 osram gu10 paired and grouped by zigbee2mqtt.

kirovilya, 2019-01-23:
@subzero79 and you can control these lamps through zigbee2mqtt groups?

sehraf, 2019-01-23:
just added two Osram Smart Plugs to my hallway group and both are working (controlled through mqtt hallway group)

subzero79, 2019-01-23:
@kirovilya yes I can control them
No problem. On/off and brightness.

glentakahashi, 2019-01-24:
@lenisko i'm hitting the rsp: 17 errors a lot as well. I spent a few hours sniffing zigbee traffic to see what was up, but honestly I'm not sure what might be causing the hang ups. out of curiosity, how many routers do you have in your network? One thing I noticed is that when you send a command to a group, all the routers in the network repeat the broadcast as well. I'm not sure how broadcasts+groups+coordinators work, but one thing I'm wondering is if the coordinator gets overflowed with requests from each of the routers as well.

lenisko, 2019-01-24:
@glentakahashi updated my message facts and added questions at the end. Currently it's one coordinator on cc2530. I don't have sniffer device yet so checks on my side have to wait a bit.

glentakahashi, 2019-01-24:
@lenisko yea you should definitely be able to, if you basically repeat the steps at https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/alternatives/max_devices but do it for CC2530, you should be able to recompile with more devices. 

`FIRMWARE_CC2530_MAX_DEVICES` doesn't exist, so you'll have to create it, but it should be fairly easy. Copy the `FIRMWARE_CC2530_DEFAULT` and change add `#define MAX_NEIGHBOR_ENTRIES 1` (or 0 if you have literally no routers) and then keep bumping up `#define NWK_MAX_DEVICE_LIST 15` until it stops building, and then go back to the last successful one.

And interesting, so even without any other routers in your network, you're still seeing the same issues. There must be something about the size or timeout of group broadcasts vs normal commands.

After reading http://ww1.microchip.com/downloads/en/AppNotes/Atmel-42687-ZigBee-Broadcast-Message-Handling-and-Implementation-in-BitCloud-SDK_AT14615_ApplicationNote.pdf, I wonder if this has to do with the size of the Broadcast Transaction Table filling up the buffer very quickly. 

Also in reading the z-stack code and http://www.deyisupport.com/cfs-file.ashx/__key/communityserver-discussions-components-files/104/5468.Application_2D00_Level-Tuning-of-Z_2D00_Stack.pdf, it seems that `BCAST_DELIVERY_TIME` is how long (in 100ms increments - ZDNwkMgr.h:92) a broadcast should be stored in the table. The default in `ZGlobals.h` is 30, which means that it's keeping around these broadcasts for 3 seconds each. `MAX_BCAST`  (number of entries stored) is also by default set to 4, so this means that you can only have 4 active broadcasts at a given time it seems.

Maybe tomorrow (or this weekend) I'm going to retry building a firmware that increases `MAX_BCAST` to  say, 16, and see if I can send out more group commands at a faster rate without hitting rsp: 17 error. I'm still not even sure rsp:17 would be thrown if `MAX_BCAST` is full, but I'll find out.

The obvious downside here is that the network could start getting congested if you're sending out lots of broadcasts, which is why i presume they have such conservative defaults, but I'll see from my testing if I experience any more flakiness.

kirovilya, 2019-01-24:
@subzero79 @sehraf Thanks for the info, we found a problem - we didn’t update the coordinator’s firmware :(

subzero79, 2019-01-24:
Don’t worry not the only one.  on my first attempt I didn’t realize the fw was on the dev branch. I repaired 23 devices including 4 plugs in the attic (40 degrees in there) to realize after that I flashed the wrong fw.

lenisko, 2019-01-24:
@glentakahashi sorry for misinformation. Like @Koenkk pointed out I HAVE already 6 routers in my network (IKEA LED bubls). I wish you luck with new firmware :-) keep us informed.

lenisko, 2019-01-25:
Freeze and lack of ability to start `zigbee2mqtt` again after few minutes

Steps:
- Restarted Hass at 2:44
- I wasn't able to get attention from any devices connected to zigbee network after 2:44:33
- Stopped zigbee2mqtt at 2:47:19
- Wasn't able to bring it up, had to plug out&in and wait some time to get it working back

I had same problem before, so it's not one time

log/debug https://hastebin.com/malipusole

alinelena, 2019-01-26:
one coordinator standard firmware. 
3 groups... one with 4 (gu10)one with 3(gu10) and other with 2(e26)  all seems to work fine.

the big issue is pairing devices I still have few xiaomi sensors refusing to pair and 4 ikea lights.

lenisko, 2019-01-26:
@alinelena keep those sensors close to router/coordinator that helped me for most xiaomi aqara devices with one exception (one of two door sensor is not pairing at all)

alinelena, 2019-01-27:
@lenisko of course i keep them close. i paired 5 door/window sensors... unfortunately one of them is stuck in 1/26/2019, 8:57:14 PM - info: door_kitchen (0x00158d0002333401): unkown - undefined unknown (EndDevice)
funny bit is xiaomi sensors in past where easy like a breeze... only a motion sensors gave me a hard time

empirically i noticed i can get the first 10 sensors, bulbs ikea/xioami without any issues. 10-20 you need to swat a little bit after 20 seems to be a pain. 
My thought is: we may need to look on more reliable hw for larger networks.

a picture of the network is here. 
[image]

alinelena, 2019-01-27:
is there any need for a specific router firmware... i see the one in branch dev is sep 2018.

lenisko, 2019-01-27:
that's for coordinator @BudBundi. Router isn't mentioned in there

alinelena, 2019-01-27:
@lenisko @BudBundi seems both master and dev have the same for router.. flashed that one... and at least connects. groups seem to work reliable I am more worried of latency not that network grew...
would be nice if groups can be highlighted somehow in the map.

[image]

alinelena, 2019-01-30:
ok so indeed it seems this firmware is somehow troublesome.

pros:
  when it works groups work fine and everything is brilliant
cons:
  both yesterday and today things stopped working... yesterday was enough to restart and everything re-paired... today things did not work seems nothing re-pairs.

trekker25, 2019-02-01:
for group support in Hass i need to define it as a light?

like this:
light:    
  - platform: mqtt
    schema: json
    name: light_spots_kozijn
    command_topic: "zigbee2mqtt/group_spots_kozijn/set"
    state_topic: "zigbee2mqtt/group_spots_kozijn/set"
    brightness: true

correct?

Koenkk, 2019-02-01:
@trekker25 yes that's correct (documentation: https://koenkk.github.io/zigbee2mqtt/integration/home_assistant.html#groups)

trekker25, 2019-02-01:
> @trekker25 yes that's correct (documentation: https://koenkk.github.io/zigbee2mqtt/integration/home_assistant.html#groups)

yes i got a little confused as the word light: was not in front of it and was looking into the Hass group documentation. But it's logical it's acting as a single light.

Already have an idea how to implement the state of bulbs? I assume than you need the members of the group also in the zigbee2mqtt config yaml. Or other smart ideas?

lenisko, 2019-02-01:
@Koenkk I know it's not something big, but it's just me
```
  - platform: mqtt
    schema: json
    name: "Salon Light"
    command_topic: "zigbee2mqtt/salon_light/set"
    state_topic: "zigbee2mqtt/salon_light/set"
    retain: true
    color_temp: true
    brightness: true
```
that `color_temp` and `brightness` are not working on `hass/dev` ?
[image]

Koenkk, 2019-02-02:
This is only visible after turnin the device on.

lenisko, 2019-02-02:
@Koenkk
Some feedback

Threr's 25 devices in the network, including 5 routers (IKEA LEDs)

I'm using [latest](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2530/bin/CC2530ZNP-Prod_20190109.zip) fw on CC2530 

__zigbee2mqtt version 1.1.1 (commit #c0326ec)__
Broken flow `Start zigbee2mqtt -> Restart Hass -> ???`
As you can see `office_switch_desk` is responding correctly, after restarting hass zigbee2mqtt is stopping responding to this and any other device in the network (was trying to click things after 4:32 - no luck, waiting for devChange from temp sensors didn't bring anything as well). 
```
  zigbee2mqtt:debug 2/2/2019, 4:31:17 PM Received MQTT message on 'zigbee2mqtt/salon_light/set' with data '{"state": "OFF"}'
  zigbee2mqtt:error 2/2/2019, 4:31:17 PM Failed to find device with ieeAddr: 'salon_light'
  zigbee2mqtt:debug 2/2/2019, 4:31:17 PM Mounted the cieApp (epId 11)
  zigbee2mqtt:debug 2/2/2019, 4:31:20 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.sensor_switch.aq2' (0x00158d0002134f7d)
  zigbee2mqtt:info 2/2/2019, 4:31:20 PM MQTT publish: topic 'zigbee2mqtt/office_switch_desk', payload '{"click":"single","linkquality":78}'
  zigbee2mqtt:debug 2/2/2019, 4:31:20 PM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.sensor_switch.aq2' (0x00158d0002134f7d)
  zigbee2mqtt:debug 2/2/2019, 4:31:28 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.sensor_switch.aq2' (0x00158d0002134f7d)
  zigbee2mqtt:info 2/2/2019, 4:31:28 PM MQTT publish: topic 'zigbee2mqtt/office_switch_desk', payload '{"click":"single","linkquality":86}'
  zigbee2mqtt:debug 2/2/2019, 4:31:38 PM Received MQTT message on 'hass/status' with data 'online'
  zigbee2mqtt:error 2/2/2019, 4:31:54 PM Failed to configure salon_switch_1 (0x0017880104ad39e6) ('undefined')
  zigbee2mqtt:error 2/2/2019, 4:32:00 PM Failed to configure salon_switch_2 (0x0017880104ad3c67) ('undefined')
```
Stopping `npm` and trying to start it again is failing as well
```
  zigbee2mqtt:info 2/2/2019, 4:33:53 PM Starting zigbee2mqtt version 1.1.1 (commit #c0326ec)
  zigbee2mqtt:info 2/2/2019, 4:33:53 PM Starting zigbee-shepherd
  zigbee2mqtt:info 2/2/2019, 4:33:56 PM Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  zigbee2mqtt:info 2/2/2019, 4:34:56 PM Starting zigbee-shepherd
  zigbee2mqtt:error 2/2/2019, 4:34:59 PM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2/2/2019, 4:34:59 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2/2/2019, 4:34:59 PM Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/home/hass/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:273:13)\n    at CcZnp.emit (events.js:182:13)\n    at Timeout.<anonymous> (/home/hass/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
```
Stopping and trying to start it again is producing same error. To fix it, I had to unplug and plug again device.

alinelena, 2019-02-02:
things somehow started to stabilise... it run for two days without issues. responsive... I cannot pair end devices.. i had no issued adding more bulbs... I still have to pair a xiaomi swtich and aqara Tph sensor... never had issues pairing these before... to make things worst asking for the map results in having the Tph sensor and button reported as as connected and online... while routers i know are online... ikea builbs are listed as offline.

[image]

BudBundi, 2019-02-03:
Not sure if it is the right place or just timing.
I have added an 'TRADFRI bulb GU10 WS 400lm' and after sending the message to add it to a group, I get this message:
`  zigbee2mqtt:info 2/3/2019, 7:50:44 PM Zigbee publish to device '0x90fd9ffffe8f027e', genGroups - add - {"groupid":"3","groupname":""} - null - null
  zigbee2mqtt:info 2/3/2019, 7:50:44 PM Successfully add 0x90fd9ffffe8f027e to Flur
  zigbee2mqtt:warn 2/3/2019, 7:50:44 PM No converter available for 'LED1537R6' with cid 'genGroups', type 'devChange' and data '{"cid":"genGroups","data":{"65533":1,"nameSupport":0}}'
  zigbee2mqtt:warn 2/3/2019, 7:50:44 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.`

Database entries for the bulb:
`{"id":20,"type":"Router","ieeeAddr":"0x90fd9ffffe8f027e","nwkAddr":26410,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb GU10 WS 400lm","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":49246,"epId":1,"devId":544,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"10":{},"65533":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI bulb GU10 WS 400lm","dateCode":"20170331","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":3},"attrs":{"65533":1,"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"16387":1,"65533":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"15":0,"16384":254,"65533":1,"currentLevel":254,"remainingTime":0,"onOffTransitionTime":5,"onLevel":255}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{"15":0,"16397":0,"16398":370,"65533":1,"remainingTime":0,"currentX":30138,"currentY":26909,"colorTemperature":370,"colorMode":2,"numPrimaries":0,"enhancedColorMode":2,"colorCapabilities":16,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}},"haDiagnostic":{"dir":{"value":1},"attrs":{"65533":1,"numberOfResets":0,"macRxBcast":0,"macTxBcast":0,"aPSRxBcast":0,"aPSTxBcast":0,"nwkFcFailure":0,"packetBufferAllocateFailures":0,"averageMacRetryPerApsMessageSent":0,"lastMessageLqi":84,"lastMessageRssi":-79}},"lightLink":{"dir":{"value":3},"attrs":{"65533":1}}}}},"_id":"V21GCc1cD5vMyF0D"}
{"id":20,"type":"Router","ieeeAddr":"0x90fd9ffffe8f027e","nwkAddr":26410,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb GU10 WS 400lm","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":49246,"epId":1,"devId":544,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"10":{},"65533":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI bulb GU10 WS 400lm","dateCode":"20170331","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":3},"attrs":{"65533":1,"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"16387":1,"65533":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"15":0,"16384":254,"65533":1,"currentLevel":254,"remainingTime":0,"onOffTransitionTime":5,"onLevel":255}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{"15":0,"16397":0,"16398":370,"65533":1,"remainingTime":0,"currentX":30138,"currentY":26909,"colorTemperature":370,"colorMode":2,"numPrimaries":0,"enhancedColorMode":2,"colorCapabilities":16,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}},"haDiagnostic":{"dir":{"value":1},"attrs":{"65533":1,"numberOfResets":0,"macRxBcast":0,"macTxBcast":0,"aPSRxBcast":0,"aPSTxBcast":0,"nwkFcFailure":0,"packetBufferAllocateFailures":0,"averageMacRetryPerApsMessageSent":0,"lastMessageLqi":84,"lastMessageRssi":-79}},"lightLink":{"dir":{"value":3},"attrs":{"65533":1}}}}},"_id":"V21GCc1cD5vMyF0D"}
`

Koenkk, 2019-02-03:
@BudBundi thanks, will be ignored in the next version.

alinelena, 2019-02-03:
since dev is missing maybe instructions shall be updated...

today i turned off all routers to try to pair the troubled xiaomi sensors left... they still did not I started the routers again, ikea bulbs ikea plug and xiaomi plug... nothing wanted to come back or repair... 
I had to reflash since simply I could not start again. 
Repared all.. first xiaomi sensors then the bulbs and worked... I still have some door/window sensors to repair... i dread it. 
I reflased with 20190109 firmware and master 
up to now things seem to be ok.

glentakahashi, 2019-02-04:
Sorry for the super late reply @lenisko , but finally got around to testing a custom firmware that altered bcast settings.

upping `MAX_BCAST` did in fact reduce the rate of `rsp: error 17` errors, however i still noticed that my bulbs in groups would get clogged after a few consecutive requests regardless. I did some sniffing and it looks as if the coordinator is properly broadcasting the on/off commands, so I determined this to mean that is my bulbs themselves that were the bottleneck. Given that i have cheap Sengled classic bulbs, it's very possible that they cannot handle the amount of broadcast requests that come through. Given how fragile even the CC2531 is with regards to memory, i assume that the Sengled bulbs have memory issues when receiving broadcasts and not freeing it fast enough. I can still use normal ZCL commands directly to the bulbs to turn them on/off as fast as possible, so it must not have the same issue memory issues with those directly, which would make more sense since they can immediately free up memory after receiving an ack.

If you would like to try this in your own network, rebuild the standard CC2531 firmware using the `firmware.patch` as default, except lower the amount of either `MAX_NEIGHBOR_ENTRIES` or `NWK_MAX_DEVICE_LIST` to free up space (reduces # of routers and # of end devices attached to cc2531 respectively) so that you can increase `MAX_BCAST` By changing `MAX_NEIGHBOR_ENTRIES` to 8, i was able to set `MAX_BCAST` to 23, which allowed me to send exactly 23 group commands in a 10 second window before hitting `rsp: error 17`, which is what is to be expected with the default `BCAST_DELIVERY_TIME` being 100.

If you do the above, please let me know how your network/bulbs perform! I am curious to see if you can get higher throughput than me, and my theory about my bulbs being cheap is correct.

kiall, 2019-02-05:
> If this is not possible or I have overseen something it would be nice to have a MQTT message that shows all groups and their members.

Yea, being able to see that would be awesome.

It would also be nice if, and I know this isn't something Home Assistant can do (yet?), ZigBee groups could be discovered by HA and treated as real group (i.e. in the group domain, rather than defining a light that controls the group).

Bruceforce, 2019-02-05:
> It would also be nice if, and I know this isn't something Home Assistant can do (yet?), ZigBee groups could be discovered by HA and treated as real group (i.e. in the group domain, rather than defining a light that controls the group).

This depends on the group members I think. If it's a group of switches only, it should be added to the group domain. If it's a group of lights it should be added to the light domain and become a light group, so that you can control features like RGB color. Home assistant wouldn't let you handle RGB for a group in the group domain iirc.

alinelena, 2019-02-09:
added report true to my bulbs... ikea...unfortunately does not seem to have any reaction. Do the ikea bulbs support it?
I am on 20190109 firmware and dev branch

[212 more replies]

closed 2019-12-31 by stale[bot]