Koenkk/zigbee2mqtt #765: Implement binding

Koenkk, 2018-12-27:
Binding allows Zigbee devices to directly control other Zigbee devices and groups without the intervention of zigbee2mqtt or home automation software.

Adding this feature makes it (probably) possible to supported the following devices:
- TRADFRI dimmer (already supported but not really usable as the interval of reports is very low)
- TRADFRI motion sensor
- TRADFRI five button remote

trekker25, 2018-12-28:
is it possible to add binding without the bridge? Would be awesome if you could bind Hue dimmer remote to Tradfri groups etc

Koenkk, 2018-12-29:
What do you mean with without the bridge?

Koenkk, 2018-12-30:
Implemented, see #782

closed 2018-12-30 by Koenkk

dmak, 2019-01-04:
Am I right saying that correct binding procedure for Tradfri groups usecase is ([see here](https://community.home-assistant.io/t/ikea-tradfri-gateway-zigbee/14788/140)):
> (1) Peer the remote control with the base/gateway/hub
> (2) Then you go with the prepared remote control to the light and peer that with the remote control.

Koenkk, 2019-01-04:
First pair it to zigbee2mqtt, then bind it to the bulb

msadeghz, 2019-03-06:
@Koenkk is it possible to bind a device to hub?

Koenkk, 2019-03-06:
@msadeghz if by hub you mean coordinator, yes, many `configure` sections for devices do this.

msadeghz, 2019-03-07:
@Koenkk what do you mean about configure sections? where are them?

Koenkk, 2019-03-07:
Via MQTT it's not possible, why you want to do this?

msadeghz, 2019-03-09:
> Via MQTT it's not possible, why you want to do this?

the manufacture said I should bind to receive the device status when I push the physical button on the device.

Koenkk, 2019-03-09:
@msadeghz this is something that can be done in the `configure` sections, example: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L532