Koenkk/zigbee2mqtt #767: permit_join not working

fourierreeks, 2018-12-28:
I have a RPi3B+ and I am trying to pair a Xiaomi MiJia door & window contact sensor (MCCGQ01LM). This is the first zigbee device I have. I followed the intructions from the wiki but I am getting stuck with pairing the device. 

From the log info it looks like its an issue with the permit_join value. In the config file it is set to true but from it seems be false (preventing me from pairing a new device).

Configfile:
```
pi@raspberrypi:/opt/zigbee2mqtt $ cat data/configuration.yaml
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost:1883'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM0

advanced:
  log_level: debug
```

Log output:
```
pi@raspberrypi:/opt/zigbee2mqtt $ sudo journalctl -u zigbee2mqtt.service -f
-- Logs begin at Thu 2016-11-03 18:16:43 CET. --
Dec 28 15:01:51 raspberrypi npm[989]:   zigbee2mqtt:info 12/28/2018, 3:01:51 PM Connecting to MQTT server at mqtt://localhost:1883
Dec 28 15:01:51 raspberrypi npm[989]:   zigbee2mqtt:info 12/28/2018, 3:01:51 PM zigbee-shepherd ready
Dec 28 15:01:51 raspberrypi npm[989]:   zigbee2mqtt:info 12/28/2018, 3:01:51 PM Connected to MQTT server
Dec 28 15:01:51 raspberrypi npm[989]:   zigbee2mqtt:info 12/28/2018, 3:01:51 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Dec 28 15:01:51 raspberrypi npm[989]:   zigbee2mqtt:debug 12/28/2018, 3:01:51 PM Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data ''
Dec 28 15:01:51 raspberrypi npm[989]:   zigbee2mqtt:info 12/28/2018, 3:01:51 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[]}'
Dec 28 15:01:51 raspberrypi npm[989]:   zigbee2mqtt:debug 12/28/2018, 3:01:51 PM Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'false'
Dec 28 15:01:51 raspberrypi npm[989]:   zigbee2mqtt:info 12/28/2018, 3:01:51 PM Zigbee: disabling joining new devices.
Dec 28 15:01:51 raspberrypi npm[989]:   zigbee2mqtt:info 12/28/2018, 3:01:51 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"debug","permit_join":false}'
Dec 28 15:01:51 raspberrypi npm[989]:   zigbee2mqtt:debug 12/28/2018, 3:01:51 PM Mounted the cieApp (epId {epId})
Dec 28 15:06:50 raspberrypi npm[989]:   zigbee2mqtt:debug 12/28/2018, 3:06:50 PM Saving state to file /opt/zigbee2mqtt/data/state.json
```

Does anyone have a idea why permit_join stays false?

sehraf, 2018-12-28:
> Dec 28 15:01:51 raspberrypi npm[989]:   zigbee2mqtt:debug 12/28/2018, 3:01:51 PM Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'false'

Something is telling it to disable joining.

fourierreeks, 2018-12-28:
But what could that be? I did not do anything else than the steps on the wiki. I did install the Domoticz plugin, could that have something to do with it?

sehraf, 2018-12-28:
i'm unfamiliar with Domoticz so here are just some guesses:
* restart your MQTT server
* send `zigbee2mqtt/bridge/config/permit_join` with payload `true` through MQTT
* use Domoticz to enable joining (no idea if that's possible)

fourierreeks, 2018-12-28:
It turned out that the permit-join false was send from domoticz. After setting "zigbee pairing" to active permit_join stays true. Now I will check if I can pair the door sensor.

fourierreeks, 2019-01-02:
Today I also bought a Ikea lightbulb but this one also does not connect. For both the ikea lightbulb and the Xiaomi MiJia sensor I followed the procedure on: https://koenkk.github.io/zigbee2mqtt/getting_started/pairing_devices.html (tried at least 5 times for each devices). For the xiaomi I also replaced the battery. 

The logfile shows no new entries after I try the pairing procedures. Is there a way I can check if my USB sniffer is working correctly?

fourierreeks, 2019-01-03:
For the Ikea bulb I do see a small flicker about 2-3 seconds after the 6th on, so I assume the reset works correctly. I did this reset about 10 times today with no result. 

For the MiJia door sensor I do see a short blink every time I press the button to keep it alive and wile I keep pressing I see a longer blink about every 5 seconds, which I think indicates no pairing. 

These 2 items are the only zigbee devices I have. With both devices not being able to pair I suspect it might be some other hardware/software that prevents the setup from working properly. I do have some more Xiaomi devices on its way, but no clue when they will arrive.

sehraf, 2019-01-03:
OK this sounds indeed like a hardware/software problem. 
Stupid question but did you try to restart the system, re-plugin the CC2531 or flash it again?

fourierreeks, 2019-01-03:
I did replug the cc2531 and rebooted the pi. The lamp is <2m from cc2531 and for the door sensor I tried everything from 5m to 5cm. 

I bought the cc2531 pre flashed so I don't have the equipment to flash it again. Is there a way to check if the cc2531 is flashed properly without the "flash" equipment?

sehraf, 2019-01-03:
That's beyond my knowledge :disappointed:

closed 2019-03-11 by stale[bot]