Koenkk/zigbee2mqtt #768: Unsupported Device Help: Bosch Motion Sensor ISW-ZPR1-WP13

finity69x2, 2018-12-28:
I'm trying to add a currently unsupported device.

It is a Bosch PIR Motion Detector Model #ISW-ZPR1-WP13

I've added this to the devices.js:

```
// Bosch
{
    zigbeeModel: ['ISW-ZPR1-WP13'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ISW-ZPR1-WP13', // Vendor model number, look on the device for a model number
    vendor: 'Bosch', // Vendor of the device (only used for documentation and startup logging)
    description: 'Bosch Motion Sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'Motion', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
},
```
Here are the logs:

https://pastebin.com/Evi2dx1f

I'm not sure what do from this point.

From research online I think it is an IAS device but I'm not sure of that.

Here is a working configuration that someone else came up with for it to work on Smartthings:

https://raw.githubusercontent.com/tomasaxerot/SmartThings/master/devicetypes/tomasaxerot/bosch-motion-detector.src/bosch-motion-detector.groovy

Please let me know if there is anything else I can provide.

I've got 4 of these that have been sitting around for over a year waiting for zha in HA to support them. And I know there are several more people in the same situation who would appreciate getting these working if possible.

Koenkk, 2018-12-28:
Please try with:

```js
// Bosch
{
    zigbeeModel: ['ISW-ZPR1-WP13'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ISW-ZPR1-WP13', // Vendor model number, look on the device for a model number
    vendor: 'Bosch', // Vendor of the device (only used for documentation and startup logging)
    description: 'Bosch Motion Sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'Motion', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];
            execute(device, actions, callback);
        },
},
```

Koenkk, 2018-12-29:
Did you also repair the device?

finity69x2, 2018-12-30:
I pulled the batteries last night and re-inserted.

Doesn't look like there is any change.

I added the logs for the first 10 minutes after the batteries were changed to the above link.

Koenkk, 2019-01-01:
`-e DEBUG=*` should be good but indeed the debug logging is not showing, could you try with `-e DEBUG="*"`?

finity69x2, 2019-01-02:
After much mucking around I finally figured out that I couldn't get the logs in the persistent directory (/app/data/logs...) to actually show any debug information past what was already shown above which was kind of sparse.

But I did realize that there were logs for the docker container that I could view using Portainer and those logs are actually capturing the debug logs, too, and in great detail.

I rebuilt a new container, edited the devices.js (which I haven't been able to successfully persist yet, but that's another story...), factory reset the sensor and allowed it to try to re-pair with the controller.

since the other log was getting long and provided little useful information I decided to create a new pastebin post for the new container debug logs.

Here is the link:

https://pastebin.com/U7KdLyr2

be warned its pretty long at over 4000 lines in just several seconds of capturing data.

I hope you find what you need. If not let me know what to do next.

Koenkk, 2019-01-02:
I need to update the logging a little bit to find out what is going wrong here. I will come back you in a few days because I dont have access to my laptop currently.

finity69x2, 2019-01-02:
Ok thanks again.

Koenkk, 2019-01-06:
Can you try again with the latest dev branch?

finity69x2, 2019-01-07:
is there a dev docker image to use? if so what is the image to use?

I tried koenkk/zigbee2mqtt:dev and it wasn't happy with that.

Koenkk, 2019-01-07:
Use `latest-dev`

Koenkk, 2019-01-09:
Can you post your `database.db`?

finity69x2, 2019-01-09:
https://pastebin.com/2gY5ARnG

Koenkk, 2019-01-09:
Can you try with:

```js
// Bosch
{
    zigbeeModel: ['ISW-ZPR1-WP13'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ISW-ZPR1-WP13', // Vendor model number, look on the device for a model number
    vendor: 'Bosch', // Vendor of the device (only used for documentation and startup logging)
    description: 'Bosch Motion Sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'Motion', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 5);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];
            execute(device, actions, callback);
        },
},
```

finity69x2, 2019-01-12:
I made the changes as requested above.

I added the log files after container rebuild and pairing the device to the end of this log file:

https://pastebin.com/ArN68rVu

Koenkk, 2019-01-13:
Still the same error which shouldn't happen we the above converter, can you provide your `database.db` again?

finity69x2, 2019-01-14:
added the new db information to the end of the previous file

https://pastebin.com/2gY5ARnG

just as another question...

is it possible that we would see the same errors if the device itself is faulty?

I've only ever tried to pair one device to anything (HA or zigbee2mqtt). Maybe I'll try a different one to see if I get the same results.

Koenkk, 2019-01-14:
That shouldn't produce this error, can you post the whole content of your `devices.js`?

finity69x2, 2019-01-15:
I honestly have no idea how to do that.

I can console into the container with sh or ash and I can cd to the correct file but i have to use vi as an editor (nano or vim is "not found") and I just can't figure out how to copy the entire text of the file use vi to the system clipboard in order to copy it to another document outside of the vi editor/console.

Koenkk, 2019-01-15:
On line 2156 I see you are still using the old code, can you use the code from https://github.com/Koenkk/zigbee2mqtt/issues/768#issuecomment-452841059

finity69x2, 2019-01-16:
Wow, Sorry about that. I must have accidentally copied the same config again.

I fixed that (and verified it). the config still fails but at least there is a different error this time.

added to the end here:

https://pastebin.com/ArN68rVu

Koenkk, 2019-01-16:
The error is different now: `2019-1-15 22:14:49 - error: Failed to configure 0x00155f00782d42af 0x00155f00782d42af ('Error: AF data request fails, status code: 240. MAC transaction expired.')`.

This indicated that the device is sleeping, can you repair the device, this makes sure the device is awake when the configure is executed.

finity69x2, 2019-01-20:
I just tried it again and I got the same error ("MAC transaction expired") but I'm not really sure why that would be the case.

According to the docs the device should go to pairing mode for 3 minutes after the batteries are inserted.

From the logs you can see that the time from when the controller received the first message from the device (pretty much as soon as I reinserted the batteries) till I get the timeout is about 1 minute. The device should still be in pairing mode then.

Unless as soon as the we get the "connected" message the device goes out of pairing mode and the timeout occurs then?

there is no other way to "wake up" the device. there is a tamper switch but I tried it both ways (with and without the tamper switch pressed) and I still get the same error.

Koenkk, 2019-01-21:
Can you try with 

```js
// Bosch
{
    zigbeeModel: ['ISW-ZPR1-WP13'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ISW-ZPR1-WP13', // Vendor model number, look on the device for a model number
    vendor: 'Bosch', // Vendor of the device (only used for documentation and startup logging)
    description: 'Bosch Motion Sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'Motion', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 5);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
            ];
            execute(device, actions, callback);
        },
},
```

kstone5, 2019-01-23:
I'm a one month nubie with Home Assistant and bought a couple of these too.  Incorrectly presumed "it's Zigbee, it'll work!:  
Reading up on zigbee2mqtt in general and this in particular.  I'm following this!
Thanks to you both and good luck!
K

finity69x2, 2019-01-23:
I feel your pain.

I literally did exactly the same thing.

Hopefully Koenkk can get these running for us.

Koenkk, 2019-01-23:
@finity69x2  did you try https://github.com/Koenkk/zigbee2mqtt/issues/768#issuecomment-456026714 ?

finity69x2, 2019-01-23:
Not yet. I just saw the comment and got it added to the container but I'm not at home to actually try to pair a device until later.

finity69x2, 2019-01-24:
new logs added to the end. But not really necessary since it is the same end result.

https://pastebin.com/ArN68rVu

Koenkk, 2019-01-26:
And with:

```js
// Bosch
{
    zigbeeModel: ['ISW-ZPR1-WP13'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ISW-ZPR1-WP13', // Vendor model number, look on the device for a model number
    vendor: 'Bosch', // Vendor of the device (only used for documentation and startup logging)
    description: 'Bosch Motion Sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'Motion', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 5);
            const actions = [
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];
            execute(device, actions, callback);
        },
},
```

finity69x2, 2019-01-30:
same...

https://pastebin.com/HsVmvVSX

Koenkk, 2019-01-30:
Let's check if we can at least get the temperature working:

```js
// Bosch
{
    zigbeeModel: ['ISW-ZPR1-WP13'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ISW-ZPR1-WP13', // Vendor model number, look on the device for a model number
    vendor: 'Bosch', // Vendor of the device (only used for documentation and startup logging)
    description: 'Bosch Motion Sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'Motion', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 5);
            const actions = [
                (cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
                (cb) => device.report('msTemperatureMeasurement', 'measuredValue', 1, 3, 0, cb),
            ];
            execute(device, actions, callback);
        },
},
``

finity69x2, 2019-02-01:
nope...

newest logs at the end...

https://pastebin.com/HsVmvVSX

Koenkk, 2019-02-02:
I'm kinda lost on how to continue here, as I don't have this device is very hard to debug what going on. Perhaps somebody else can continue with this

finity69x2, 2019-02-02:
thank you for trying. These stupid things have been a thorn since I bought them.

I hope this isn't too forward but would it help if I sent a device to you to work with? I don't know if you are in the US or not but if you are I doubt if the shipping would be too much.

I would be willing to do that so it might help everyone else in my same situation if you are open to it..

Koenkk, 2019-02-02:
I'm open to that, however I live in the Netherlands so I expect high shipping costs.

Koenkk, 2019-02-02:
My telegram is Koenkk

finity69x2, 2019-02-02:
sorry i don't have any idea what that means... (ignorant American here, remember? ;) )

finity69x2, 2019-02-02:
nevermind... i figured it out. sent you a message.

sraue, 2019-02-15:
Koenkk, you got such a device? can send you such one too if it is to expensive to send from US...

Koenkk, 2019-02-15:
@sraue Hi Stephan! nice to see you here :smile:

I don't have the device, if you want you can send it over (no guarantees that I get it working though). Please contact me on telegram (Koenkk).

kstone5, 2019-02-16:
@Koenkk I'm also in the U.S. but would be happy to donate something to the cause.  Do you have a method in place I could donate a few $'s (e.g., 'buy me coffee')?
Thanks to you and others that advance these projects!

Koenkk, 2019-02-16:
@kstone5 paypal donate link can be found here: https://github.com/Koenkk/zigbee2mqtt/blob/master/README.md

ReneHonig, 2019-02-19:
@Koenkk , I have this device and live in NL. Do you still require it?

Koenkk, 2019-02-19:
@ReneHonig no, @sraue already send me one (on its way)

Koenkk, 2019-03-03:
This device is supported now!

The reason of the `240` error was that the IAS configure has to succeed right after pairing, probably the configure wasn't there when you initially paired the device. 

If you already paired the device, remove the device from `database.db` and pair it again by factory resetting it:

[image]

finity69x2, 2019-03-03:
That's great! I knew I could have faith in you getting this done. 

I'll try it as soon as I can and report back with the results.

When will this be entered into the master branch?

Koenkk, 2019-03-04:
@finity69x2 yes, once the next version of zigbee2mqtt is released.

dgz, 2019-04-10:
I have 2 of these and was able to pair them today with the most recent released version of the hass.io addon so this is looking good. I followed the reset instructions above first since they were previously paired with SmartThings and then started from the beginning of the pairing instructions provided with them. 

Both work as expected, just note to those that haven't used them that they report occupancy for about 1 second and then immediately report clear and ignore motion for 3 minutes. Most people will want to program an ignore period in their automations or use a template sensor to account for this behavior.

closed 2019-06-16 by stale[bot]

bseishen, 2020-01-07:
@dgz Just ran into this when setting up one of these sensors. I'm wondering if the sticky state should be part of zigbee2mqtt or handle it in automation software. I might should add that tid bit to the device page, since i thought it was a hardware/configuration issue.