Koenkk/zigbee2mqtt #769: Unsupported Bosch Motion Sensor RFDL-ZB-MS

pfischbach, 2018-12-28:
Trying to setup a Bosch Motion Sensor TriTech ZigBee HA1.2 Radar PIR Sensor
Pairing worked out well. I added a device copied from  VMS_ADUROLIGHT
Seems to send messages when motion is detected. Tamper contact works, but motion messages are false instead of true.

See log and device info from pairing.

[Log](https://pastebin.com/TXPmAnHq) 

Would be nice if we can get them working.
[Could be similar to #768 Bosch Motion Sensor ISW-ZPR1-WP13 request](https://github.com/Koenkk/zigbee2mqtt/issues/768)

Koenkk, 2018-12-28:
What configuration do you use know? (`devices.js`)

pfischbach, 2018-12-29:
Did some more testing with the basic config from the last post 
looks like zoneStatus 33 = motion true and zoneStatus 32 = motion false
ias_zone_motion_dev_change function looks at bit 1 Alarm 2 for motion and not bit 0 ??
I think motion is on bit 0 for the Bosch sensor
bit 2 work with tamper contact.
I get 37 / 36 when I open the tamper contact and activate motion

Koenkk, 2018-12-29:
Dont use the ias_zone_motion_dev_change, you can create a new converter for this case

pfischbach, 2019-01-03:
I was able to create a new converter looking at the correct bit 0. All works well.
Now I am trying to create a converter to set the sensitivity value. Looking at the example JTQJBF01LMBW_sensitivity I can't figure out the correct parameters based on the data from my Bosch  database entry.
` "ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":13,"zoneStatus":0,"iasCieAddr":"0x0000000000000000","zoneId":255,"numZoneSensitivityLevelsSupported":7,"currentZoneSensitivityLevel":5}}` 

@Koenkk  any hint to get me going?

Koenkk, 2019-01-04:
Need to check if that is possible. Did you set the sensitivity level before? (using a different hub)

pfischbach, 2019-01-10:
No, I do not have a commercial hub. I thought if there is a parameter in the data base I could read/write it.
"numZoneSensitivityLevelsSupported":7,"currentZoneSensitivityLevel":5

Koenkk, 2019-01-11:
Ok, let's check if we can write it.

Can you add the following actions to `configure()`?

```js
(cb) => device.write('ssIasZone', 'currentZoneSensitivityLevel', 7, cb),
(cb) => device.read('ssIasZone', 'currentZoneSensitivityLevel', cb),

```

pfischbach, 2019-01-18:
I am confused now I re-paired the device with 
`    {
        zigbeeModel: ['RFDL-ZB-MS'],
        model: 'RADON TriTech ZB',
        vendor: 'Bosch',
        description: 'Motion sensor',
        supports: 'occupancy and temperature',
        fromZigbee: [fz.generic_temperature, fz.ignore_temperature_change, fz.STS_PRS_251_battery, fz.ignore_power_change, fz.bosch_ias_zone_motion_status_change],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
                (cb) => device.write('ssIasZone', 'currentZoneSensitivityLevel', 7, cb),
                (cb) => device.read('ssIasZone', 'currentZoneSensitivityLevel', cb),
         
            ];

            execute(device, actions, callback);
        },
    },`
No I don't see the temperature and battery voltage anymore.

`  zigbee2mqtt:debug 1/17/2019, 6:27:17 PM Mounted the cieApp (epId {epId})
  zigbee2mqtt:error 1/17/2019, 6:27:21 PM Failed to configure 0x000d6f000f59add9 0x000d6f000f59add9
  zigbee2mqtt:debug 1/17/2019, 6:27:24 PM Received zigbee message of type 'devChange' with data '{"cid":"ssIasZone","data":{"zoneStatus":32}}' of device 'RFDL-ZB-MS' (0x000d6f000f59add9)
  zigbee2mqtt:warn 1/17/2019, 6:27:24 PM No converter available for 'RADON TriTech ZB' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"zoneStatus":32}}'
`
With only the fromZigbee converters I had everything working but as soon as I paired a second device it would not work anymore.
What is the best procedure to start from scratch. 
The bosch converter for the motion sensor seems to work.
[bosch converter](https://pastebin.com/U1rbaYea)

Do I always have to reflash to clear the firmware? What is stored in the coordinator after power off?
The sensor would not repair after deleting the data base entry, only after reflash.

Koenkk, 2019-01-18:
Let's get back to the basics.

Could you first check if the occupancy works and that no `No converter available messages..` are published:

```js
{
        zigbeeModel: ['RFDL-ZB-MS'],
        model: 'RADON TriTech ZB',
        vendor: 'Bosch',
        description: 'Motion sensor',
        supports: 'occupancy and temperature',
        fromZigbee: [fz.bosch_ias_zone_motion_status_change, fz.ignore_iaszone_devchange],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

With the following converters:
```js
    bosch_ias_zone_motion_status_change: {
        cid: 'ssIasZone',
        type: 'statusChange',
        convert: (model, msg, publish, options) => {
            const zoneStatus = msg.data.zoneStatus;
            return {
                occupancy: (zoneStatus & 1) > 0, // Bit 1 = Alarm 1: Presence Indication
                tamper: (zoneStatus & 1<<2) > 0, // Bit 2 = Tamper status
                battery_low: (zoneStatus & 1<<3) > 0, // Bit 3 = Battery LOW indicator (trips around 2.4V)
            };
        },
    },
    ignore_iaszone_devchange: {
        cid: 'ssIasZone',
        type: 'devChange',
        convert: (model, msg, publish, options) => null,
    },
```

pfischbach, 2019-01-20:
Thank you! Did a test with one sensor, see log 
[Log](https://pastebin.com/htiexcPC)
Sensor works for motion, motion is detected and reset. cyclic temperature and genPowerCfg message causes a no converter found using your suggested converter and device entry. Temperature and batt values look OK.
Tested a second sensor and it works as well. Not sure why one sensor has the battery in the MQTT message and the other one not. I did a re-flash and deleted the data base before the new pairing.
Good news  are the motion detection works.

Koenkk, 2019-01-21:
Great, could you make a PR?

pfischbach, 2019-01-22:
Yes I can try, but there are still two issues in the log above.
Should I put the converters for battery and temperature back in?

zigbee2mqtt:debug 1/22/2019, 1:55:16 PM Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryVoltage":30}}' of device 'RFDL-ZB-MS' (0x000d6f000f59ac4e)
  zigbee2mqtt:warn 1/22/2019, 1:55:16 PM No converter available for 'RADON TriTech ZB' with cid 'genPowerCfg', type 'attReport' and data '{"cid":"genPowerCfg","data":{"batteryVoltage":30}}'
  zigbee2mqtt:warn 1/22/2019, 1:55:16 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.

  zigbee2mqtt:debug 1/22/2019, 1:53:00 PM Received zigbee message of type 'attReport' with data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":1859}}' of device 'RFDL-ZB-MS' (0x000d6f000f59ac4e)
  zigbee2mqtt:warn 1/22/2019, 1:53:00 PM No converter available for 'RADON TriTech ZB' with cid 'msTemperatureMeasurement', type 'attReport' and data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":1859}}'
  zigbee2mqtt:warn 1/22/2019, 1:53:00 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.

Koenkk, 2019-01-22:
For the temperature you can use the following converters `fz. generic_temperature`. 

What type of battery does this device contain?

pfischbach, 2019-01-22:
Battery is a 3V Lithium CR123A

Koenkk, 2019-01-23:
You can use the `fz.generic_batteryvoltage_3000_2500`, if everything works ok, could you make a PR?

pfischbach, 2019-01-24:
Yes, I will test multiple sensors and make a PR

pfischbach, 2019-01-28:
@Koenkk 
Tested with 3 sensors,  had to add to devices
                (cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
                (cb) => device.report('msTemperatureMeasurement', 'measuredValue', 30, 600, 1, cb),
to get the temperature reported. Also added fz.ignore_temperature_change. Works now, can you explain what the  values 30 and 600 mean in "measuredValue', 30, 600"?

Koenkk, 2019-01-28:
Minimum and maximum report interval of the temperature. I would recommend setting it to this value: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L8

closed 2019-04-05 by stale[bot]

Styx85, 2019-11-25:
Hi, I just tested a similar Bosch product with the model Number [RFPR-ZB-MS](https://us.boschsecurity.com/en/products/intrusionalarmsystems/premiseswireless/zigbee/radionpirzbwirelessmotion_1/radionpirzbwirelessmotion_1_36394), I just added the model number to the devices.js and it works out of the box. Guess one has to add all published versions? eg. RFPR-ZB-MS, RFPR-ZB, [RFDL-ZB](https://us.boschsecurity.com/en/products/intrusionalarmsystems/premiseswireless/zigbee/radiontritechzbwirelessmo_1/radiontritechzbwirelessmo_1_36397)).

there are more modell numbers mentioned in the instruction manual, but you cannot find the modell on the bosch homepage. [link1](https://usermanual.wiki/Bosch-Security-Systems/PR-ZB) [link2](https://usermanual.wiki/Bosch-Security-Systems/DL-ZB)

`zigbeeModel: ['RFDL-ZB', 'RFDL-ZB-MS', 'RFPR-ZB-MS', 'RFPR-ZB', ] `

alternatively all versions mentioned in the instruction manuals:
`zigbeeModel: ['RFDL-ZB', 'RFDL-ZB-EU', 'RFDL-ZB-H', 'RFDL-ZB-K', 'RFDL-ZB-CHI', 'RFDL-ZB-MS', 'RFDL-ZB-ES', 'RFPR-ZB', 'RFPR-ZB-EU', 'RFPR-ZB-CHI', 'RFPR-ZB-ES', 'RFPR-ZB-MS', ] `

by the way the difference between RFDL and RFPR series seems only the added radar with RFDL

Koenkk, 2019-11-25:
@Styx85 thanks, added!

cinemarene, 2020-05-08:
I bought one of those Bosch motion detectors. Unfortunately I can't pair it properly.
It is recognized, but I do not receive any status updates.
Mine has the product code "RFPR-ZB-SH-EU".

Is it relevant that it has an "Install Code"?
And is it relevant that it has Zigbee "EUI-64" as a designation?

Ltek, 2024-08-27:
Warning to Users : Z2M does not work/pair properly with this. When you can get it to paid (after like 10-15 times) it usually falls off hours or days later.