Koenkk/zigbee2mqtt #770: Bitron Smart Plug with Metering - seMetering support

simonrath, 2018-12-28:
I'm trying to figure out how to get the information on power consumption for the Bitron Smart Plug with metering 16A.

Because the device is not supported, I have already done the following steps.

Added the Device to devices.js
```
{
        zigbeeModel: ['902010/25'],
        model: '902010/25',
        vendor: 'Bitron Video',
        description: 'Bitron Video Plug with metering',
        supports: 'on/off, power measurement',
        fromZigbee: [fz.generic_state_change, fz.bitron_seMetering, fz.bitron_seMeteringatt],
        toZigbee: [tz.on_off],
},
```
Added converters to fromZigbee.js

```
	bitron_seMetering: {
		cid: 'seMetering',
		type: 'attReport',
		convert: (model, msg, publish, options) => {
			console.log(msg.data)
			return {
				power: msg.data.data['instantaneousDemand'],
				current: msg.data.data['currentSummDelivered'],
			};
		},
	},
	bitron_seMeteringatt: {
		cid: 'seMetering',
		type: 'devChange',
		convert: (model, msg, publish, options) => {
			return {
				power: msg.data.data['instantaneousDemand'],
				current: msg.data.data['currentSummDelivered'],
			};
		},
	},
```
After this, iam able to turn the plug on and off, but i still get no message about the power consumption.

Database.db
https://pastebin.com/6X4Pb6Y2
Log
https://pastebin.com/yr1fcDLN

And here the Datasheet for the device (Page 5)
http://media.bitronvideo.eu/bitronhome/Bitron_Home_ZigBee_Clusters_and_Attributes_-_en_0814.pdf

Koenkk, 2018-12-28:
Please try with:

```js
{
        zigbeeModel: ['902010/25'],
        model: '902010/25',
        vendor: 'Bitron Video',
        description: 'Bitron Video Plug with metering',
        supports: 'on/off, power measurement',
        fromZigbee: [fz.generic_state_change, fz.bitron_seMetering, fz.bitron_seMeteringatt],
        toZigbee: [tz.on_off],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            execute(device, [(cb) => device.report('seMetering', 'instantaneousDemand', 10, 60, 1, cb)], callback);
        },
},
```

simonrath, 2018-12-28:
Thank you. Know it works as intended. I also get the state message, when pressing the on/off button on the device.
```
{
        zigbeeModel: ['902010/25'],
        model: '902010/25',
        vendor: 'Bitron Video',
        description: 'Bitron Video Plug with metering',
        supports: 'on/off, power measurement',
        fromZigbee: [fz.generic_state_change, fz.bitron_seMetering],
        toZigbee: [tz.on_off],
		configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
				(cb) => device.report('seMetering', 'instantaneousDemand', 10, 60, 1, cb),
				(cb) => device.bind('genOnOff', coordinator, cb),
			];
			
            execute(device, actions, callback);
        },
    },
```
```
bitron_seMetering: {
		cid: 'seMetering',
		type: 'attReport',
		convert: (model, msg, publish, options) => {
			return {
				// Power in Watt
				power: parseFloat(msg.data.data['instantaneousDemand']) / 10,
			};
		},
	},
```

I still dont know for what the attributes "currentSummDelivered" and "currentSummReceived" are. They are always 0

Koenkk, 2018-12-29:
These can be ignored (not applicable for these kind of devices). Could you make a PR?

simonrath, 2018-12-30:
Done 
Koenkk/zigbee-shepherd-converters#181

In addition to my implementation, i needed to put "manufacturerName":"Bitron Video","modelId":"902010/25" in database.db manually. As mentioned in #682 

Do i need to change the doc somehow?

Koenkk, 2018-12-30:
@simonrath why did you need to do this?

closed 2019-03-07 by stale[bot]

martin3000, 2019-07-08:
I have a smart plug where current_summ_delivered is filled. It is Watt*seconds, the power consumption.