Koenkk/zigbee2mqtt #771: Update to Home Assistant 0.84.x & Zigbee2mqtt 1.0.x breaks all sensors

MikeVRT, 2018-12-28:
Workable configuration on HASS 0.82.1 & zigbee2mqtt 0.2.0 
updated as usual to HASS 0.84.6 & zigbee2mqtt 1.0.1

can't work at all: no any sensors,
Integrations->MQTT reports "This integration has no devices."

MikeVRT, 2018-12-28:
> Do not use the yaml definition!
But we talking about update. Could you describe right update way step by step please?

MarsWarrior, 2018-12-29:
What I did was the following:

- Remove the mqtt definition in configuration.yaml
- Add mqtt as integration. This enables the device registry for autodiscovery
- Enable autodiscovery for mqtt in configuration.yaml (https://www.home-assistant.io/docs/mqtt/discovery/)
- in your Zigbee2mqtt configuration.yaml set homeassistant: true

Restart zigbee2mqtt and after that home assistant.
If things go well, you will see the devices in your mqtt integrations. Note that there is still a problem in hass for lights (Tradfri, Hue for example): these will show up as devices without entities. Should be fixed with version 0.85 which should be released around januari 9th.

My zigbee Aqara sensors all show up in the integrations panel.

Koenkk, 2018-12-29:
@MarsWarrior not sure what you mean with pair them once again, but on the zigbee2mqtt nothing has to be done.

MarsWarrior, 2018-12-29:
@Koenkk I had to re-pair all my aqara sensors again for some reason. But since then no problems anymore, even after reflashing the CC2531 sticks.

Maybe some problem on my end 😄

closed 2019-02-27 by Koenkk