Koenkk/zigbee2mqtt #774: Osram flex outdoor connectivity issues - only connects to coordinator (no router)?

h4nc, 2018-12-29:
I'd like to mount an Osram flex outdoor in my garden. Unfortunately I'm facing connectivity issues. 

I have several enddevices. The most of them work great. I have one coordinator and two routers in my network. 
The controller of the led stripe would sit 5 meters next to a router with one wall in between. Nevertheless I cannot control the led stripe.

When I put it in the same room as the coordinator it works. Putting it one room further make it loose the connection. My other Zigbee devices don't have these issues (except ikea dimmer).
The other devices are sensors (mostly aqara sensors). So having a lamp is new to my zigbee2mqtt network.

I kept my hue bridge (because of philips ambilight+hue) so I keep my hue lights seperated from the zigbee2mqtt network. This means I have to zigbee networks running. 
I also tried to connect the stripe to the hue bridge. This way the stripe works a little further. But again not in the garden (next bulb sits 2 walls away, approx. 10 meters).

I'd really like to use this light and don't understand why other deives work fine and this one not.

I don't know if its a bug but link quality of my routers (the value that is reported to Home assistant) is 0 most of the time for both routers. However as I said before, my other devices work. And the zigbee map tells me that the devices are connected to the router. 

Any advice would be helpful.

h4nc, 2018-12-29:
Ok, maybe someone here will help us find a solution.

Koenkk, 2018-12-30:
I would recommend adding a CC2530 as the antenna is much more powerful than the one on your zigbee bulb

Koenkk, 2018-12-30:
Let me know if that works

h4nc, 2018-12-30:
I already tried it. It seems like the cc2530 router is not able to add devices to the network, but I don't know it, could be something else. So don't take this as bulletproof.

As @dbmet has the same issues (not connecting to router only to coordinator) it seems like there is different problem.
But really like to get rid of this. I'm glad to test further suggestions.

h4nc, 2019-02-28:
Should stay opened, as this is not solved

h4nc, 2019-04-29:
Should stay open.

h4nc, 2019-06-28:
Stay open

h4nc, 2019-08-27:
Stay open

Koenkk, 2019-08-28:
I'm not sure if you issue will ever be solved, this seems to be an issue with the device itself and doesn't look like it can be fixed from zigbee2mqtt.

h4nc, 2019-08-28:
Ok, so than let’s close it.

closed 2019-08-28 by h4nc