Koenkk/zigbee2mqtt #775: Testing: device availability

Koenkk, 2018-12-29:
This allows zigbee2mqtt to mark individual devices as offline. E.g. when you switch a bulb off using the regular switch. 

The device availability is published on `zigbee2mqtt/[FRIENLDY_NAME]/availability` with possible payloads being `offline` and `online`.

Home Assistant integration has been updated, devices will be shown as `unavailable` when they are offline.

Device availability is only checked for AC powered routers.

Todos:
[ ] Check device availability for new paired devices (without having to restart zigbee2mqtt)

How to test:
Update zigbee2mqtt to the latest dev branch (edge for hassio users). Stop zigbee2mqtt, add to zigbee2mqtt `configuration.yaml`

```yaml
experimental:
  availablility_timeout: 60 # Availability check interval
```

start zigbee2mqtt and restart Home Assistant.

Most of the credits of this feature go to @ugrug

wimpie007, 2018-12-30:
doesnt work with hassio/zigbee2mqtt-edge.
Can't see anything different. (when is zigbee2mqtt-edge updated to most recent source?)

Koenkk, 2018-12-30:
Should be, can you post your startup log?

Koenkk, 2018-12-30:
@wimpie007 I see that your are running #2a197a4, this was introduced in a later commit #afeed4f372bced737206bd91f8e62ce65b127eb9 (this can be seen in the startup log). 

Note that also a typo has just been fixed `availablility` -> `availability`, I've updated the configuration in the OP. Wait at least one hour to give the hassio addon some time to regenerate.

wimpie007, 2018-12-31:
I stopped/started the  zigbee2mqtt-edge addon in hassio this morningn (auto-update is on), but still on commit #2a197a4.
In hassio, i can't do more to 'force' the update?

Koenkk, 2018-12-31:
I dont understand why, the image has been updated 14 hours ago which makes sense (https://hub.docker.com/r/dwelch2101/zigbee2mqtt-edge-armhf/tags). @danielwelch could you help?

danielwelch, 2018-12-31:
I don’t think starting and stopping alone should work, as this doesn’t pull the new image. You need to uninstall and re-install.

wimpie007, 2018-12-31:
Thanks! uninstall/install did the trick, bit annoying, but ok!:) will report later on....

wimpie007, 2018-12-31:
doesnt seem to do anything...
here is the start-log.
https://pastebin.com/u5kANWCm

Koenkk, 2018-12-31:
Can you post a bit more of you startup log? I'm especially interested in the `homeassistant/` topic it publishes to.

wimpie007, 2019-01-01:
sure!
https://pastebin.com/WmXvQ2Qz
PS Happy 2019!!:)

Koenkk, 2019-01-01:
Happy 2019! Somehow the functionallity is not enabled yet (because it still publishes zigbee2mqtt/bridge/state as availabillity topic). Can you post the experimental part of your configuration.yaml again?

wimpie007, 2019-01-01:
core-ssh:/share/zigbee2mqtt# cat configuration.yaml
advanced:
  channel: 13
  log_level: debug
devices:
  '0x000b57fffea710af':
    friendly_name: '0x000b57fffea710af'
    retain: false
...snip...
  '0x84182600000fcc42':
    friendly_name: '0x84182600000fcc42'
    retain: false
experimental:
  availablility_timeout: 60
homeassistant: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://homeassistant:1883
permit_join: false
serial:
  disable_led: true
  port: /dev/ttyACM0
core-ssh:/share/zigbee2mqtt#

wimpie007, 2019-01-01:
YES! we are getting somewhere!
for the lights it works! show up as "unavailable" when powered-off.
BUT... also for al the sensors.
**Sensors show all up as unavailable as they do not respond to ping...**
"Device availability is only checked for AC powered routers.": not true!:)

Koenkk, 2019-01-01:
Can you post your log_level: debug log?

wimpie007, 2019-01-01:
https://pastebin.com/bbu0MEnD
see for example device:
0x00158d00029c076d (0x00158d00029c076d): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)

Koenkk, 2019-01-01:
Found the bug, should be fixed

martinrosenauer, 2019-01-03:
@Koenkk I can also confirm that it works for me on a bunch of different Trådfri bulbs as well as some Philips Hues. I don't see any updates for my battery driven sensors/buttons as expected..

It would be great if you also manage the state of the lights send to Home Assistant as mentioned by @wimpie007.  

Under all circumstances - **thanks for doing/maintaining this brilliant project  !**

wimpie007, 2019-01-04:
Another improvement would be:
 - faster convergence to state online by using "Received zigbee message"..."with data"... as 'online-trigger'
Now, even when the "received zigbee message" is received, the device remains "unavailable" until the next ping cycle...

WiwiWillou, 2019-01-06:
Great feature !

Iv4nS, 2019-01-11:
Question.. I have the E11-G13 - Sengled Element Classic (A19) (EndDevice) bulbs, these do not report when they are turned ON by the wall switch nor their status periodically.
The thing is, the type of this device is "EndDevice" so this code will not actually try pinging these bulbs. Should EndDevice be added or should I be using another method to detect if they are available or not?

sehraf, 2019-01-19:
My OSRAM Smart+ plug are detected (falsely) as offline from time to time. Does anybody else has observed this behavior?

Koenkk, 2019-01-29:
For the 1.1 release I made this option non-experimental, please move `availability_timeout` from `experimental` to `advanced` (https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/configuration/configuration.md)

sehraf, 2019-01-29:
I've disabled device availability since i home assistant won't let me control the smart plug when it is marked as offline (which makes sense if it would be offline ... which it isn't)
So a blacklist might come in handy.

Koenkk, 2019-01-29:
@sehraf I see, added this to the OP

Koenkk, 2019-01-31:
Thanks to @jbmbn devices will now instantly marked as online when they are powered again.

Koenkk, 2019-02-02:
> Koen, thanks!
> it works now!
> state of the lamps doesnt reflect reality after power off/on, but that is another issue...
> a big step for this project! nice!

Implemented (can be tested in the dev branch).

Koenkk, 2019-02-02:
@sehraf `availability_blacklist` has been implemented (documentation: https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/configuration/configuration.md)

timstanley1985, 2019-02-06:
Hi, I have authored PR #1030 to work around the issue of availablity of non AC devices.  The change allows home assistant to decide whether a device is available based on how long it has been since the most recent state update was received

kiall, 2019-02-07:
One thing that's a little annoying once this is enabled - theres lots of logspam when your devices aren't always on:

     zigbee2mqtt:error 2019-2-7 19:05:53 Failed to ping 0x0017880102ebca62
     zigbee2mqtt:info 2019-2-7 19:05:53 MQTT publish: topic 'zigbee2mqtt/living_ceiling_light_2/availability', payload 'offline'

This will repeat every so often for each offline device.. Could these messages only be logged upon transitioning from online to offline? (or offline -> online again..) This I guess is more annoying when using via hassio, and the log history available is severely limited.

Koenkk, 2019-02-09:
@kiall implemented!

tb-killa, 2019-03-09:
@Koenkk Get this on pairing mode if device isnt fully online:
```
 zigbee2mqtt:error 2019-3-9 21:29:53
        {"message":"Cannot read property 'ieeeAddr' of undefined","stack":"TypeError: Cannot read property 'ieeeAddr' of undefined\n    at DeviceAvailability.isPingable (/opt/zigbee2mqtt/lib/extension/deviceAvailability.js:33:44)\n    at DeviceAvailability.onZigbeeMessage (/opt/zigbee2mqtt/lib/extension/deviceAvailability.js:142:28)\n    at extensions.filter.forEach (/opt/zigbee2mqtt/lib/controller.js:134:31)\n    at Array.forEach (<anonymous>)\n    at Controller.onZigbeeMessage (/opt/zigbee2mqtt/lib/controller.js:134:14)\n    at Zigbee.onMessage (/opt/zigbee2mqtt/lib/zigbee.js:209:18)\n    at ZShepherd.emit (events.js:180:13)\nat ZShepherd.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:99:14)\n    at ZShepherd.emit (events.js:180:13)\n    at /opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/event_handlers.js:229:18"}
```

Bruceforce, 2019-04-04:
I started zigbee2mqtt with 3 ikea tradfri gu 10 turned on, after 5 minutes I turned them off, and only one of them became unavailable in home assistant. For more details see https://github.com/Koenkk/zigbee2mqtt/issues/1334#issuecomment-478298221

Here is the new debug log: https://gist.github.com/Bruceforce/5a943f8ac665ef52aadddffe0f6f37a7

Koenkk, 2019-04-07:
@Bruceforce indeed something strange is going on, I added extra logging, could you provide the same logging on the latest dev branch?

Bruceforce, 2019-04-09:
@Koenkk Sorry it took a while. Did the same as before.

Debug log: https://gist.github.com/Bruceforce/ea0a48c999cbc467804ad5e0481b04d7

database.db: https://gist.github.com/Bruceforce/383e6ad87db1702e3f81e71c23d134b2

Koenkk, 2019-04-09:
@Bruceforce thanks to @didiht this should be fixed in the latest dev branch, can you verify?

Bruceforce, 2019-04-09:
Thanks to both of you. It's now working again.

smith844, 2019-04-24:
Good Afternoon all.
I have a similar problem an I am at the end of my (limited) knowledge.  
When ever the AC is removed from my 9 Tradfri LED1650R5 bulbs they will then not rejoin when AC is switched back on.
I have a CC2531 USB as coordinator in Raspi, as well as a CC2530 as a router. 
I am running a Raspi 3B+ with Hassio 91.4 and zigbee2mqtt-edge(although I tried latest stable version also.)
Interestingly the router is still pingable (as seen in log) until the power is returned to LED bulbs and then that also becomes unpingable as soon as power is returned.
The only way to get the devices back is to unplug the USB CC2531 and restart the zigbee2mqtt add on.
I have tried reflashing both CC2531 and CC2530 and repairing, I even selected a Zigbee channel to try to provide some stability, but to no avail.
The log is here [https://gist.github.com/smith844/8f699effc9974cd9e21694b8185c977f](url)
In the log above I removed power at 1:10:00 PM and turned it back on at 1:11:30 PM, you will see from the log that after the switch on the router stops responding to pings.
The configuration.yaml is here [https://gist.github.com/smith844/6afcfdbd7a04c2a9826a1507525131be](url)

Any help gratefully received.  I hoped by adding a router it would help in the cases where AC power is removed from all the devices (on the wall switch) as it would provide a stable network.  I am guessing I was wrong.........

smith844, 2019-04-24:
Sorry can't get the link to work, but the addresses are:
Log.txt
https://gist.github.com/smith844/8f699effc9974cd9e21694b8185c977f
[https://gist.github.com/smith844/8f699effc9974cd9e21694b8185c977f](url)
Configuration.yaml
https://gist.github.com/smith844/6afcfdbd7a04c2a9826a1507525131be
[https://gist.github.com/smith844/6afcfdbd7a04c2a9826a1507525131be](url)

Koenkk, 2019-04-24:
@smith844 as a quick check, can you check if this happens with the max devices firmware? (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/max_devices/CC2531)

smith844, 2019-04-24:
@Koenkk 
I reflashed it with the Max devices and I get the same results.  still pings router(CC2530) after tradfri devices are off but as soon as power to them is reapplied all stops.

smith844, 2019-04-25:
@Koenkk 
It works. I obviously didn't give it enough time to stabilise.  The state reporting is still a bit hit and miss in HA but I can live with that. Thank you for the assistant and your advancement of the add on.  It has certainly given my wife less reasons to hate my drive towards home automation :)

Koenkk, 2019-04-25:
@smith844 so the max devices firmware has fixed this issue? It would be nice if you could confirm this by flashing the max stability again, then we know the issue is in the max stability firmware.

smith844, 2019-04-25:
I will of course do that as soon as I get an opportunity, probably later tonight or tomorrow.

Koenkk, 2019-04-26:
@smith844 can you check if this has also been fixed in the latest dev firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_Home_1.2/bin

CypherMK, 2019-04-27:
I tested this with 1.31, seems to work. But I get a lot off false "offline" warnings even when my ikea bulb is on. Timeout was set to 2
I guess I should update to latest dev branch?

Bruceforce, 2019-04-27:
@CypherMK how do you verify that the offline warnings are false? A bulb can be on and offline at the same time. That's why I'm asking.

smith844, 2019-04-27:
@Koenkk I have installed the dev branch and it works about the same as the max devices.  I then relooked at max stability and max devices.  I no longer get a failed system when .i put power on after ac power out, even in max stability, but the availability and state reporting is very intermittent and inconsistent.
I have bulbs that show offline and unavailable in HA and wont accept MQTT messages that still toggle on and off, but only with a group message.  its all very strange, but more usable than it was.

CypherMK, 2019-04-27:
@Bruceforce
I used node red, and used the topic: zigbee2mqtt/(device)/availability
This gives me a offline message immediately followed by a online message, but I didn't check the ZigBee map.

CypherMK, 2019-04-28:
Updated to the dev branch and the behaviour is the same.
What marks a bulb offline? 
What I expected to achieve is, that when I switch off the wall switch, I will immediately get an offline message, After switching, I will immediately get an online message. So it would be possible to automate some stuff.

Koenkk, 2019-04-29:
@CypherMK a device is marked offline when a ping request fails. The interval of the ping request is configured by `availability_timeout: 30`, in this example every device is pinged each 30 seconds. 

So the time between bulb powerlos and bulb offline is max 30 seconds. When the device is powered on, it should become online immediately.

definitio, 2019-05-04:
I think devices should be unavailable when zigbee2mqtt isn't connected to broker, i.e. LWT should be set to `offline` for every device availability topic.

Koenkk, 2019-05-05:
@definitio zigbee2mqtt already published `offline` for every device, LWT doesn't seem to be possible as the used MQTT library only allows one LWT message (and this is already used for `zigbee2mqtt/bridge/state`).

waiweng83, 2019-05-27:
Hi, I'm new to zigbee2mqtt and i find that the availabillity of battery powered devices is pretty important especially for some sensors like smoke alarm, water leakage sensor, etc.

I know that different devices will have different report/heartbeat interval. Perhaps just make it configurable for each device? Any plan to include this feature? If not, what is the best way to monitor this in home assistant?

wimpie007, 2019-05-27:
a battery operated device will simply not answer to "pings" from the coordinator.
You can just wait and hope that the device will send a packet.... no way to tell it's dead or 'sleeping'.

waiweng83, 2019-05-27:
You do not need to ping the battery powered device to know its availability. Some of the devices will send a heartbeat periodically. For example, Xiaomi/Aqara door/window sensor will send a heartbeat together with its battery voltage around once per hour. We can do like if we do not receive any packet for more than 3 hours, mark it as dead. Of course every devices will send out heartbeat at different interval, some might not even send out any heatbeat. That's why I suggest the watchdog period to be configurable per every device.

wimpie007, 2019-05-27:
ok, i understand. Yes, that would be a nice option. (configurable 'timeout' that is 'reset' when receiving a message from the device...)

CypherMK, 2019-05-27:
That would be great. Now I constructed something similar within node red. So when my temperature sensor doesn't report within a hours, I get a pushover notification.

waiweng83, 2019-05-28:
There is a PR by another guy for this feature before (#761 ) but somehow the feature is dropped.
Anyone knows why?

waiweng83, 2019-06-18:
@ugrug is the guy comes out with the idea of attribute_report_interval.
Do you mind creating another PR for this function? I think it's really useful.

desertspider, 2019-07-18:
Any update on this? Would be create if i can see the status of battery based devices.

[53 more replies]

closed 2020-05-09 by stale[bot]