Koenkk/zigbee2mqtt #776: npm install reports npm ERR! fetch failed

niekvlessert, 2018-12-29:
When running npm install I got npm ERR! fetch failed on (I guess) all dependencies.

Just 2 of them:
npm ERR! fetch failed https://registry.npmjs.org/object-visit/-/object-visit-1.0.1.tgz
npm ERR! fetch failed https://registry.npmjs.org/objectbox/-/objectbox-0.3.0.tgz

The report stated I had bad network settings, but that was not the case, wget on those URLs works just fine.

Eventually I changed all entries in npm-shrinkwrap.json from https to http and the problems were gone. Everything works fine now. Thanks for all the effort btw!!

It's on the latest Raspbian on a Pi 1B. I followed https://koenkk.github.io/zigbee2mqtt/getting_started/running_zigbee2mqtt.html, but in there is  sudo curl -sL https://deb.nodesource.com/setup_10.x | sudo -E bash - which won't work on ArmV6. So I used the Raspbian version. Might have something to do with that... no idea.

niekvlessert, 2018-12-30:
No, or the default Raspbian installation is doing things I don't know. But since wget works fine on the same URL on https I guess it's not that.

I guess it will be hard to figure out what it is, maybe just close the ticket and when someone has this issue they can find this workaround in the issue history...

Koenkk, 2018-12-30:
It seems to be a npm issue.

closed 2018-12-30 by Koenkk