Koenkk/zigbee2mqtt #778: zigbee2mqtt:error Not connected to MQTT server!

sander816, 2018-12-30:
Domoticz beta running for a few months, and i wanted to add zigbee2mqtt
Bought a cc2531 stick, with the firmware already on it

https://koenkk.github.io/zigbee2mqtt/getting_started/running_zigbee2mqtt.html

new install today, but getting this error:
(debug log)
npm start(as pi user)
```
  zigbee2mqtt:info 12/30/2018, 12:36:55 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2018-12-30.12-36-55'
  zigbee2mqtt:info 12/30/2018, 12:36:57 PM Starting zigbee2mqtt version 1.0.1 (commit #9380bbc)
  zigbee2mqtt:info 12/30/2018, 12:36:57 PM Starting zigbee-shepherd
  zigbee2mqtt:info 12/30/2018, 12:36:58 PM zigbee-shepherd started
  zigbee2mqtt:info 12/30/2018, 12:36:58 PM Coordinator firmware version: '20181119'
  zigbee2mqtt:info 12/30/2018, 12:36:58 PM Currently 0 devices are joined:
  zigbee2mqtt:warn 12/30/2018, 12:36:58 PM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 12/30/2018, 12:36:58 PM Allowing new devices to join.
  zigbee2mqtt:warn 12/30/2018, 12:36:58 PM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 12/30/2018, 12:36:58 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 12/30/2018, 12:36:58 PM Connecting to MQTT server at mqtt://localhost:1883
  zigbee2mqtt:info 12/30/2018, 12:36:58 PM zigbee-shepherd ready
  zigbee2mqtt:error 12/30/2018, 12:37:08 PM Not connected to MQTT server!

```

sander816, 2018-12-30:
Ok. I think so? I am new to mqtt. What server do i have to use?

LarsNorgaard, 2018-12-30:
> Ok. I think so? I am new to mqtt. What server do i have to use?

Mosquitto is one most people pick.
`sudo apt-get install mosquitto`

MQTT is the middleware between zigbee2mqtt and home assistant.

sander816, 2019-01-05:
ok, done that
i don't know if this is correct though, i added aqara human body sensor?
```
  zigbee2mqtt:info 1/5/2019, 1:37:31 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-01-05.13-37-30'
  zigbee2mqtt:info 1/5/2019, 1:37:31 PM Starting zigbee2mqtt version 1.0.1 (commit #9380bbc)
  zigbee2mqtt:info 1/5/2019, 1:37:31 PM Starting zigbee-shepherd
  zigbee2mqtt:info 1/5/2019, 1:37:32 PM zigbee-shepherd started
  zigbee2mqtt:info 1/5/2019, 1:37:32 PM Coordinator firmware version: '20181119'
  zigbee2mqtt:info 1/5/2019, 1:37:32 PM Currently 1 devices are joined:
  zigbee2mqtt:info 1/5/2019, 1:37:32 PM 0x00158d00029bff93 (0x00158d00029bff93): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
  zigbee2mqtt:warn 1/5/2019, 1:37:32 PM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 1/5/2019, 1:37:32 PM Allowing new devices to join.
  zigbee2mqtt:warn 1/5/2019, 1:37:32 PM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 1/5/2019, 1:37:32 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 1/5/2019, 1:37:32 PM Connecting to MQTT server at mqtt://localhost:1883
  zigbee2mqtt:info 1/5/2019, 1:37:32 PM zigbee-shepherd ready
  zigbee2mqtt:info 1/5/2019, 1:37:32 PM Connected to MQTT server
  zigbee2mqtt:info 1/5/2019, 1:37:32 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 1/5/2019, 1:37:32 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00029bff93', payload '{"illuminance":9,"linkquality":94,"occupancy":true}'
  zigbee2mqtt:info 1/5/2019, 1:37:46 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00029bff93', payload '{"illuminance":8,"linkquality":31,"occupancy":true}'
  zigbee2mqtt:info 1/5/2019, 1:37:46 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00029bff93', payload '{"illuminance":8,"linkquality":28,"occupancy":true}'
  zigbee2mqtt:info 1/5/2019, 1:37:52 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00029bff93', payload '{"illuminance":521,"linkquality":39,"occupancy":true}'
  zigbee2mqtt:info 1/5/2019, 1:37:52 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00029bff93', payload '{"illuminance":521,"linkquality":39,"occupancy":true}'
  zigbee2mqtt:info 1/5/2019, 1:38:00 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00029bff93', payload '{"illuminance":25,"linkquality":15,"occupancy":true}'
  zigbee2mqtt:info 1/5/2019, 1:38:00 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00029bff93', payload '{"illuminance":25,"linkquality":21,"occupancy":true}'
  zigbee2mqtt:info 1/5/2019, 1:38:07 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00029bff93', payload '{"illuminance":19,"linkquality":78,"occupancy":true}'
  zigbee2mqtt:info 1/5/2019, 1:38:07 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00029bff93', payload '{"illuminance":19,"linkquality":76,"occupancy":true}'
```

closed 2019-03-13 by stale[bot]