Koenkk/zigbee2mqtt #779: log_level issue

1Joe1, 2018-12-30:
With log level info my setup is working perfectly fine. When switching to warn it seems as if no more sensor data is being sent via mqtt. Is that normal?
When permanently running in info mode I fear the constantly growing data volume, as all sensor data is permanently being logged. Or is this fear unnecessary as older data is automatically being deleted at some point?

Koenkk, 2018-12-30:
Log files are automatically cleaned up. 3 files will be kept with a maximum of 10MB per file.

1Joe1, 2019-01-02:
Thanks, that kind of solves my "problem". Though at the moment there are 10 log files in the log-directory. I'll keep an eye on it...
Still it surprises me that the log level has an effect on the sensor data that is sent via mqtt. Is that the way it is supposed to be? Does that only happen in my setup?

Koenkk, 2019-01-02:
10 log files directories with max 3 logs files inside. only last 10 directories are kept.

Log level will not affect messages send to mqtt, it will just stop logging them when you set it higher than info.

closed 2019-03-10 by stale[bot]