Koenkk/zigbee2mqtt #781: Device connected next step?

Btje1977, 2018-12-30:
I installed everything and it i looking good. Now I started to pair a device. I added a ikea gu10 light bulp and the mqtt logging showed it was connected:

Dec 30 19:57:15 HomeAssistant npm[484]:   zigbee2mqtt:info 2018-12-30 19:57:15 New device with address 0x90fd9ffffe0fc442 connected!
Dec 30 19:57:15 HomeAssistant npm[484]:   zigbee2mqtt:info 2018-12-30 19:57:15 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connec                             ted","message":"0x90fd9ffffe0fc442"}'

mqtt added the discovery options:

mqtt:
  discovery: true
  discovery_prefix: ha_mqtt
  broker: 192.168.1.100
  port: 1883
  keepalive: 60
  username: ha
  password: Juul2006!
  client_id: home-assistant-1
  birth_message:
    topic: 'hass/status'
    payload: 'online'
  will_message:
    topic: 'hass/status'
    payload: 'offline'

But now the next step. Where should the device show up to add it as an switch?
I have a separate groups yaml. should I add the example code to groups?

this code:

group:
  zigbee_group:
    view: false
    control: hidden
    name: Zigbee2mqtt
    entities:
      - input_boolean.zigbee_permit_join
      - timer.zigbee_permit_join
      - sensor.zigbee2mqtt_bridge_state
      - switch.zigbee2mqtt_main_join
      - automation.enable_zigbee_joining
      - automation.disable_zigbee_joining
      - automation.disable_zigbee_joining_by_timer
      - input_select.zigbee2mqtt_log_level
      - automation.zigbee2mqtt_log_level

And what about the other example code?

Koenkk, 2018-12-30:
It should show under Configuration -> Integrations -> MQTT

Btje1977, 2018-12-30:
Is shows configured items like nest and google cast. And indeed mqtt configuration.yaml. But the it says no "apparaten".

svrooij, 2018-12-30:
You’ll have to enable the zigbee2mqtt -> homeassistant discovery as described at 
https://koenkk.github.io/zigbee2mqtt/integration/home_assistant.html

Koenkk, 2018-12-30:
@svrooij fun to see you here, I know you from FHICT, lol :smile:

svrooij, 2018-12-30:
Used bought myself some CC2530’s want to replace all the gateways have in my home.
- Hue
- Xiaomi
And also join some ikea lights.
Possibly replace the Honeywell gateway as well but that’s a really closed system.

Btje1977, 2018-12-31:
I saw that i forgotto set hoem assistant to true:

then the logging shows:

-- Logs begin at Mon 2018-12-31 12:06:38 CET. --
Dec 31 12:08:39 HomeAssistant npm[480]:   zigbee2mqtt:info 2018-12-31 12:08:39 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Dec 31 12:08:39 HomeAssistant npm[480]:   zigbee2mqtt:warn 2018-12-31 12:08:39 Message without device!
Dec 31 12:08:39 HomeAssistant npm[480]:   zigbee2mqtt:info 2018-12-31 12:08:39 Connecting with device...
Dec 31 12:08:39 HomeAssistant npm[480]:   zigbee2mqtt:info 2018-12-31 12:08:39 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Dec 31 12:08:39 HomeAssistant npm[480]:   zigbee2mqtt:warn 2018-12-31 12:08:39 Message without device!
Dec 31 12:08:39 HomeAssistant npm[480]:   zigbee2mqtt:info 2018-12-31 12:08:39 Device incoming...
Dec 31 12:08:39 HomeAssistant npm[480]:   zigbee2mqtt:info 2018-12-31 12:08:39 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
Dec 31 12:08:39 HomeAssistant npm[480]:   zigbee2mqtt:info 2018-12-31 12:08:39 New device with address 0x000b57fffed7a57e connected!
Dec 31 12:08:39 HomeAssistant npm[480]:   zigbee2mqtt:info 2018-12-31 12:08:39 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000b57fffed7a57e"}'
Dec 31 12:08:39 HomeAssistant npm[480]:   zigbee2mqtt:info 2018-12-31 12:08:39 MQTT publish: topic 'homeassistant/light/0x000b57fffed7a57e/light/config', payload '{"brightness":true,"schema":"json","command_topic":"zigbee2mqtt/0x000b57fffed7a57e/set","state_topic":"zigbee2mqtt/0x000b57fffed7a57e","availability_topic":"zigbee2mqtt/bridge/state","name":"0x000b57fffed7a57e_light","unique_id":"0x000b57fffed7a57e_light_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000b57fffed7a57e","name":"0x000b57fffed7a57e","sw_version":"Zigbee2mqtt 1.0.1","model":"TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)","manufacturer":"IKEA"}}'

But still nothing in home assistant.

They are added to the configuratio.yaml:
devices:
  '0x90fd9ffffe0fc442':
    friendly_name: '0x90fd9ffffe0fc442'
    retain: false
  '0x000b57fffed7a57e':
    friendly_name: '0x000b57fffed7a57e'
    retain: false

Btje1977, 2019-01-07:
Found it. I changed the discovery prefix: ha_mqtt  when I removed this it will connect.

offtopic @Koenkk  Ik werk hier bij Oce met een Altran collega van jou, Marco Veltman.

closed 2019-01-07 by Btje1977

Koenkk, 2019-01-07:
@Btje1977 toevallig! Morgen ben ik ook weer bij Oce :smile: