Koenkk/zigbee2mqtt #782: Testing: binding

Koenkk, 2018-12-30:
This issue is used to gather feedback on the binding feature.

Notes:
- Documentation is available [here](https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/information/binding.md)
- Dev branch is required (edge for Hassio users)

niekvlessert, 2019-01-01:
I'm pretty new to this world, but I can help testing. Would this be a valid testing environment? Insert the CC2531 USB stick in my iMac, forward the USB port to Virtualbox, install zigbee2mqtt in a Linux VM and test. So I can keep my stable & slow RPI 1B the same.

Another question; how would I pair the Ikea remote so that I can try using it and zigbee2mqtt at the same time? Now I factory default the Ikea lamp, then it'll connect to zigbee2mqtt automatically, but when I then pair the remote, the connection with zigbee2mqtt is gone.

niekvlessert, 2019-01-03:
I've got my VM setup running, running zigbee2mqtt dev version on a VM with forwarded USB port on my iMac. Since the bulb pairs fine with this it looks ok. I couldn't pair the Ikea device, it gave: 'PM Device with modelID 'TRADFRI remote control' is not supported.'

So I found: https://github.com/Koenkk/zigbee2mqtt/issues/102#issuecomment-395855749

I added the settings to devices.js, but I couldn't find the handleReady in lib/zigbee.js, I guess this info is dated? Any tips?

Koenkk, 2019-01-04:
@niekvlessert that not needed, once paired you will be able to bind it

martinrosenauer, 2019-01-04:
@Koenkk, should we be able to pair the (5 button) Trådfri Remote now ? (I'm also getting the 'PM Device with modelID 'TRADFRI remote control' is not supported.' after resetting it as also @niekvlessert get).

Koenkk, 2019-01-04:
Yes, but the unsupported message doesnt mean you cannot use binding

martinrosenauer, 2019-01-04:
@Koenkk, ok, so is it the same that happens when I "pair" the remote with a bulb? (e.g. holding the reset button inside it pressed for 15-20 secs next to the bulb)

Koenkk, 2019-01-04:
@martinrosenauer yes

niekvlessert, 2019-01-04:
It works!

- Factory default the remote 
- Pair with zigbee2mqtt, ignore the 'Not supported'
- Factory default the bulb
- Let it pair with zigbee2mqtt
- Pair the remote with the bulb
- Now my mosquitto_pub -t 'zigbee2mqtt/0x000b57fffecb610d/set' -m '{"color_temp":350}' still works but the remote also works. :)

Interesting; I can't change the color temperature with the remote anymore, off/on and intensity work fine.

ookyn, 2019-01-04:
Hi,
Same results as @niekvlessert here : 
I can manage all features with mqtt but i can't manage color of my rgb bulb with the remote anymore .

JumpmanJunior, 2019-01-05:
I'm using an IKEA remote and RGB bulb. But it does not work for me.
* The bulb is paired with zigbee2mqtt.
* The remote is also paired with zigbee2mqtt. I have received the "Not supported" error message
* I bind the remote to the bulb: zigbee2mqtt/bridge/bind/REMOTE BULB
* I get the following messages in the log:

> 2019-1-5 03:55:08 - warn: Cannot handle MQTT message on 'zigbee2mqtt/bridge/bind/REMOTE' with data 'BULB'
> 2019-1-5 03:55:11 - info: Successfully bound cluster '5' from 0x000d6ffffe1d7895' to '0x000d6ffffe1df5f5'
> 2019-1-5 03:55:13 - info: Successfully bound cluster '6' from 0x000d6ffffe1d7895' to '0x000d6ffffe1df5f5'
> 2019-1-5 03:55:15 - info: Successfully bound cluster '8' from 0x000d6ffffe1d7895' to '0x000d6ffffe1df5f5'
> 2019-1-5 03:55:15 - warn: Don't know how to setup reporting for '5', skipping..
> 2019-1-5 03:55:15 - info: Configured reporting for cluster '6' of device '0x000d6ffffe1df5f5'
> 2019-1-5 03:55:15 - info: Configured reporting for cluster '8' of device '0x000d6ffffe1df5f5'
* So it seems that everything worked. I assume that the cluster 5 message relates to the not working colour changing feature reported by @niekvlessert and @ookyn.
* But for me the remote is not working at all. I can still control the bulb via zigbee2mqtt.
* Unbinding seems also to work:

> 2019-1-5 04:11:24 - warn: Cannot handle MQTT message on 'zigbee2mqtt/bridge/unbind/xUfo' with data 'xIkea'
> 2019-1-5 04:11:28 - info: Successfully unbound cluster '5' from 0x000d6ffffe1d7895' to '0x000d6ffffe1df5f5'
> 2019-1-5 04:11:30 - info: Successfully unbound cluster '6' from 0x000d6ffffe1d7895' to '0x000d6ffffe1df5f5'
> 2019-1-5 04:11:32 - info: Successfully unbound cluster '8' from 0x000d6ffffe1d7895' to '0x000d6ffffe1df5f5'

niekvlessert, 2019-01-05:
Are you using the dev branch?

BudBundi, 2019-01-07:
I try to bind a "Xiaomi Aqara single key wireless wall switch (single click)" with a "GL-C-008 - Gledopto Zigbee LED controller RGB + CCT / RGBW / WWCW / Dimmer"
but only get the message 
`zigbee2mqtt:error 2019-1-7 16:41:06 Failed to bind cluster '5' from 0x00158d0002523a2f' to '0x00124b00199f0348' (Error: Timed out after 10000 ms)`

Startup messages 

>   zigbee2mqtt:info 2019-1-7 16:32:07 Logging to directory: '/app/data/log/2019-01-07.15-32-07'
>   zigbee2mqtt:info 2019-1-7 16:32:11 Starting zigbee2mqtt version 1.0.1 (commit #932a66a)
>   zigbee2mqtt:info 2019-1-7 16:32:11 Starting zigbee-shepherd
>   zigbee2mqtt:info 2019-1-7 16:32:15 zigbee-shepherd started
>   zigbee2mqtt:info 2019-1-7 16:32:15 Coordinator firmware version: '20181024'
>   zigbee2mqtt:info 2019-1-7 16:32:15 Currently 7 devices are joined:
>   zigbee2mqtt:info 2019-1-7 16:32:15 Bewegungsmelder_Unten (0x00158d0002b95d67): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
>   zigbee2mqtt:info 2019-1-7 16:32:15 Bewegungsmelder_Oben (0x00158d0002b6d069): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
>   zigbee2mqtt:info 2019-1-7 16:32:15 Raumsensor_Stube (0x00158d0002c8cfe2): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
>   zigbee2mqtt:info 2019-1-7 16:32:15 Raumsensor_Ankleide (0x00158d0002c8d2c3): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
>   zigbee2mqtt:info 2019-1-7 16:32:15 Raumsensor_Badezimmer (0x00158d0002c8d155): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
>   zigbee2mqtt:info 2019-1-7 16:32:15 Licht_Badezimmer (0x00124b00199f0348): GL-C-008 - Gledopto Zigbee LED controller RGB + CCT / RGBW / WWCW / Dimmer (Router)
>   zigbee2mqtt:info 2019-1-7 16:32:15 Schalter_Badezimmer (0x00158d0002523a2f): WXKG03LM - Xiaomi Aqara single key wireless wall switch (EndDevice)
>   zigbee2mqtt:info 2019-1-7 16:32:15 Zigbee: disabling joining new devices.
>   zigbee2mqtt:info 2019-1-7 16:32:15 Connecting to MQTT server at mqtt://mosquitto
>   zigbee2mqtt:info 2019-1-7 16:32:16 zigbee-shepherd ready
>   zigbee2mqtt:info 2019-1-7 16:32:16 Connected to MQTT server
>   zigbee2mqtt:info 2019-1-7 16:32:16 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'

Is it not possible to bind the Xiaomi Switch? Or is there another problem?
Platform is Docker on a Raspberry Pi 3b+
Image is `koenkk/zigbee2mqtt:arm32v6-dev@sha256:aa87c8e89569ecd6646658b2802e8c28170e062882fd70b56501f235cb07ba9b`

Koenkk, 2019-01-07:
@BudBundi Xiaomi defines don't follow the ZCL specification, therefore they are (probably) not able to control other devices.

trekker25, 2019-01-09:
would be awesome if it could be used for groups!
Not sure if this is even possible?

As a backup would be great to have a remote that can turn on / off a couple of groups i'm going to realize next month.

Koenkk, 2019-01-09:
@trekker25 I didn't have any success with the TRADFRI dimmer, however this probably depends per device.

JumpmanJunior, 2019-01-11:
I have now tried to bind the Philips dimmer switch with an Ikea RGB bulb. I did the same as described in my post with the Tradfri remote and I also have received the same result. According to the log it works except for cluster 5 but the remote is not working. The remote still sends its events to zigbee2mqtt but is not controlling the bulb.

> 2019-1-11 03:44:31 - warn: Cannot handle MQTT message on 'zigbee2mqtt/bridge/bind/xDimmer' with data 'xIkea'
> 2019-1-11 03:44:32 - info: Successfully bound cluster '5' from 0x0017880104e680f9' to '0x000d6ffffe1df5f5'
> 2019-1-11 03:44:32 - info: Successfully bound cluster '6' from 0x0017880104e680f9' to '0x000d6ffffe1df5f5'
> 2019-1-11 03:44:33 - info: Successfully bound cluster '8' from 0x0017880104e680f9' to '0x000d6ffffe1df5f5'
> 2019-1-11 03:44:33 - warn: Don't know how to setup reporting for '5', skipping..
> 2019-1-11 03:44:33 - info: Configured reporting for cluster '6' of device '0x000d6ffffe1df5f5'
> 2019-1-11 03:44:33 - info: Configured reporting for cluster '8' of device '0x000d6ffffe1df5f5'
> 2019-1-11 03:44:49 - info: MQTT publish: topic 'zigbee2mqtt/xDimmer', payload '{"battery":100,"linkquality":23,"action":"off","last_seen":"2019-01-11T02:44:49.119Z"}'

Koenkk, 2019-01-11:
@JumpmanJunior can you do some zigbee sniffing and post the results of the on/off commands send by the remote? https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/how_tos/how_to_sniff_zigbee_traffic.md

Koenkk, 2019-01-15:
@JumpmanJunior do you know how this remote is normally used, can it be paired directly to a bulb?

JumpmanJunior, 2019-01-15:
Yes, the remote can be paired directly to a bulb. That is even the default if you buy it in a set with a bulb.

In order to exclude a problem with the bulb I have now tried to bind the Philips dimmer switch to a Philips bulb (instead of the Ikea bulb I used before). But the results are the same, it is not working.

Koenkk, 2019-01-16:
@JumpmanJunior in the log I see that the remote reports that the binding is OK, however ZCL Onoff is still send to `0x0000`. Perhaps it can only be bound to one device, by default zigbee2mqtt will bind it to the coordinator.

- Can you replace https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L725 and https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L726 with `unbind`
- Start zigbee2mqtt and verify that it logs `Successfully configured...`
- Put `return` after https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L722
- Bind it to the bulb again.

JumpmanJunior, 2019-01-17:
OK, this worked. Now the remote is controlling the bulb. But I was not receiving the button presses in zigbee2mqtt anymore. After executing the unbind command it immediately stopped controlling the bulb and the button presses where showing up in zigbee2mqtt again.

I also tried the same with the Tradfri remote which was integrated via the coordinator_group but I had no success with this one. But there is also no "Successfully configured" message for this remote.

Koenkk, 2019-01-18:
@JumpmanJunior it seems that this remote can only be bound to one devices, so either to coordinator or he bulb. About the TRADFRI remote, it doesn't allow to be bound (also experienced this). What you can do is add the bulbs to the same group as the one of the TRADFRI remote.

f-bader, 2019-01-25:
Hi, binding worked as long as I woke up the TRADFRI wireless dimmer (ICTC-G-1) before sending the MQTT message
But I receive an error about reporting for one cluster

> 2019-1-25 19:28:12 - info: Successfully bound cluster '6' from 0x000b57fffe9292a7' to '0x0017880110348471'
> 2019-1-25 19:28:12 - info: Successfully bound cluster '8' from 0x000b57fffe9292a7' to '0x0017880110348471'
> 2019-1-25 19:28:12 - info: Configured reporting for cluster '8' of device '0x0017880110348471'
> 2019-1-25 19:28:42 - error: Failed to setup reporting for cluster '6' of device '0x0017880110348471'

f-bader, 2019-01-25:
One thing thats a bit annoying that when I use the dimmer the statechange of  the target is not published. Only the statechange of the dimmer. This makes the integration in home assistant a bit tricky

Koenkk, 2019-01-26:
@f-bader that's because the reporting setup fails (see your log). cluster 6 is the on/off cluster, Does changing brightness trigger a state update?

nikko82, 2019-01-27:
Hi. Thanks for this great tool!

I've successfully set up binding between an "IKEA TRADFRI wireless dimmer (ICTC-G-1)" and a "Xiaomi Aqara smart LED bulb (ZNLDP12LM)", but after pairing the dimmer doesn't seem to report to the bridge any more (not a big issue) and when I dim I get lots of MQTT messages with missing brightness level and a few with level 0 and 1. A while after dimming I get a message with brightness 75.

`Jan 27 22:07:28 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:07:28 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","brightness":1,"color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:07:28 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:07:28 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:07:28 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:07:28 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","brightness":0,"color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:07:28 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:07:28 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:07:28 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:07:28 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:07:28 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:07:28 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:07:28 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:07:28 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:07:28 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:07:28 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:09:12 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:09:12 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","brightness":1,"color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:09:12 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:09:12 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:09:13 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:09:13 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","brightness":0,"color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:09:13 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:09:13 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:09:13 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:09:13 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","brightness":1,"color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:09:13 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:09:13 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:09:13 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:09:13 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","brightness":0,"color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:09:13 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:09:13 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","brightness":0,"color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:10:10 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:10:10 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:10:10 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:10:10 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:10:10 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:10:10 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:10:10 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:10:10 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:10:11 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:10:11 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
Jan 27 22:10:11 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:10:11 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","color_temp":454,"linkquality":36,"color":{"x":0.507,"y":0.418}}'
[... cut ...]
Jan 27 22:28:42 raspberrypi npm[27006]:   zigbee2mqtt:info 1/27/2019, 10:28:42 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002d71dd1', payload '{"state":"ON","brightness":75,"color_temp":454,"linkquality":39,"color":{"x":0.507,"y":0.418}}'
`

In the beginning of the log I was dimming the light and got lots of messages with missing brightness or level 0 or one. Then 1:15 later it reported level 75 (which is incorrect, it was dimmed to 100%).
Then I dimmed down, and the same thing happened (I see one message with level 4) and 2 minutes later it reported 75 again.

Is there anything I can do about it? Is it a bug, or is that maybe how the Ziaomi-bulb behaves?

f-bader, 2019-01-28:
> @f-bader that's because the reporting setup fails (see your log). cluster 6 is the on/off cluster, Does changing brightness trigger a state update?

No it does not. I will try to rebind and check if cluster 6 is successful setup

milakov, 2019-01-30:
Hi, is fw 20181024 sufficient to try binding feature?

filoor, 2019-02-07:
Hi,

I'm taring to bind my hue dimmer to a group of lights.
This is what i'm getting 
`zigbee2mqtt:info 2019-2-7 09:19:07 Successfully bound cluster '5' from 0x001788010219009c' to '1'
  zigbee2mqtt:info 2019-2-7 09:19:07 Successfully bound cluster '6' from 0x001788010219009c' to '1'
  zigbee2mqtt:info 2019-2-7 09:19:08 Successfully bound cluster '8' from 0x001788010219009c' to '1'
  zigbee2mqtt:info 2019-2-7 09:19:08 Successfully bound cluster '5' from 0x001788010219009c' to '1'
  zigbee2mqtt:info 2019-2-7 09:19:09 Successfully bound cluster '6' from 0x001788010219009c' to '1'
  zigbee2mqtt:error 2019-2-7 09:19:09 Zigbee publish to device '0x001788010239775d', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:info 2019-2-7 09:19:09 Successfully bound cluster '8' from 0x001788010219009c' to '1'
  zigbee2mqtt:info 2019-2-7 09:19:10 Successfully bound cluster '5' from 0x001788010219009c' to '1'
  zigbee2mqtt:info 2019-2-7 09:19:10 Successfully bound cluster '6' from 0x001788010219009c' to '1'
  zigbee2mqtt:info 2019-2-7 09:19:11 Successfully bound cluster '8' from 0x001788010219009c' to '1'`

But lights wil not go on if i click on the buttons

nikko82, 2019-02-07:
I just updated to the latest dev version hoping that the new reporting-fix would fix the reporting-issues for my Aqara bulb, but still after unbinding, removing and adding back the bulb and binding again I still have the same issue. When I dom with the IKEA dimmer thu bulb either doesn't report brightness at all or it reports only 0 and 1. Any idea why this happens?

Koenkk, 2019-02-07:
@nikko82 did you add `report: true` for the device that you want reporting for? (https://koenkk.github.io/zigbee2mqtt/configuration/device_specific_configuration.html)

sylarevan, 2019-02-09:
Hi. I just tried to bind my Hue dimmer switch with one Hue bulb. Here is what I get in the log when binding:
```
Feb 09 14:41:47 raspberrypi npm[29218]:   zigbee2mqtt:warn 2/9/2019, 2:41:47 PM Cannot handle MQTT message on 'zigbee2mqtt/bridge/bind/hue-switch-chambre' with data 'hue-light-chambre'
Feb 09 14:42:37 raspberrypi npm[29218]:   zigbee2mqtt:error 2/9/2019, 2:42:37 PM Failed to bind cluster '5' from 0x00178801021b66c2' to '0x0017880102246f61' (Error: Timed out after 10000 ms)
Feb 09 14:43:03 raspberrypi npm[29218]:   zigbee2mqtt:info 2/9/2019, 2:43:03 PM Successfully bound cluster '6' from 0x00178801021b66c2' to '0x0017880102246f61'
Feb 09 14:43:04 raspberrypi npm[29218]:   zigbee2mqtt:info 2/9/2019, 2:43:04 PM Successfully bound cluster '8' from 0x00178801021b66c2' to '0x0017880102246f61'
Feb 09 14:43:04 raspberrypi npm[29218]:   zigbee2mqtt:info 2/9/2019, 2:43:04 PM Configured reporting for cluster '6' of device '0x0017880102246f61'
Feb 09 14:43:04 raspberrypi npm[29218]:   zigbee2mqtt:info 2/9/2019, 2:43:04 PM Configured reporting for cluster '8' of device '0x0017880102246f61'
```

But the remote is still sending data to z2m. It seems you proposed a solution to this before:

> @JumpmanJunior in the log I see that the remote reports that the binding is OK, however ZCL Onoff is still send to `0x0000`. Perhaps it can only be bound to one device, by default zigbee2mqtt will [...]

Is it still something which is required to make the binding ok between a Hue switch and bulb?

sylarevan, 2019-02-09:
Well, because of the `(Error: Timed out after 10000 ms)` above I tried another time to bind the switch with the bulb but I've waken up the switch before. I had a slightly different log:
```
zigbee2mqtt:warn 2/9/2019, 5:39:54 PM Cannot handle MQTT message on 'zigbee2mqtt/bridge/bind/hue-switch-chambre' with data 'hue-light-chambre'
zigbee2mqtt:info 2/9/2019, 5:39:57 PM MQTT publish: topic 'zigbee2mqtt/hue-switch-chambre', payload '{"battery":96.5,"linkquality":31}'
zigbee2mqtt:info 2/9/2019, 5:39:57 PM Successfully bound cluster '5' from 0x00178801021b66c2' to '0x0017880102246f61'
zigbee2mqtt:info 2/9/2019, 5:39:58 PM Successfully bound cluster '6' from 0x00178801021b66c2' to '0x0017880102246f61'
zigbee2mqtt:info 2/9/2019, 5:39:58 PM Successfully bound cluster '8' from 0x00178801021b66c2' to '0x0017880102246f61'
zigbee2mqtt:warn 2/9/2019, 5:39:58 PM Don't know how to setup reporting for '5', skipping..
zigbee2mqtt:info 2/9/2019, 5:39:58 PM Configured reporting for cluster '6' of device '0x0017880102246f61'
zigbee2mqtt:info 2/9/2019, 5:39:58 PM Configured reporting for cluster '8' of device '0x0017880102246f61'
```

Still, both devices are not bound.

JumpmanJunior, 2019-02-10:
With the newest version binding a Philips dimmer switch to a Philips bulb works without problems. Unbinding is also working. But as the switch can only be bound to one device I tried to activate reporting for the bulb. `report: true` is set in the configuration but zigbee2mqtt does not receive any updates from the bulb.

sylarevan, 2019-02-10:
@JumpmanJunior How did you confirm that the binding was effective? Despite the messages `Successfully bound cluster '5'`(and clusters 6 and 8) above, if I turn off z2m, the switch does not actually command my bulb...

Koenkk, 2019-02-10:
Please report the reporting issues here: https://github.com/Koenkk/zigbee2mqtt/issues/1064

JumpmanJunior, 2019-02-10:
@sylarevan If I turn of zigbee2mqtt the switch can still control the bulb.

nikko82, 2019-02-11:
> @nikko82 did you add `report: true` for the device that you want reporting for? (https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/configuration/device_specific_configuration.md)

I have now set up reporting, and according to the log it's set up successfully

> 2/11/2019, 11:59:02 AM - debug: Setup reporting for 0x00158d0002d6f79b - genOnOff - onOff
2/11/2019, 11:59:02 AM - debug: Successfully setup reporting for 0x00158d0002d6f79b - genOnOff - onOff
2/11/2019, 11:59:02 AM - debug: Setup reporting for 0x00158d0002d6f79b - genLevelCtrl - currentLevel
2/11/2019, 11:59:02 AM - debug: Successfully setup reporting for 0x00158d0002d6f79b - genLevelCtrl - currentLevel
2/11/2019, 11:59:02 AM - debug: Setup reporting for 0x00158d0002d6f79b - lightingColorCtrl - colorTemperature
2/11/2019, 11:59:02 AM - debug: Successfully setup reporting for 0x00158d0002d6f79b - lightingColorCtrl - colorTemperature
2/11/2019, 11:59:02 AM - debug: Setup reporting for 0x00158d0002d6f79b - lightingColorCtrl - currentX
2/11/2019, 11:59:03 AM - debug: Successfully setup reporting for 0x00158d0002d6f79b - lightingColorCtrl - currentX
2/11/2019, 11:59:03 AM - debug: Setup reporting for 0x00158d0002d6f79b - lightingColorCtrl - currentY
2/11/2019, 11:59:03 AM - debug: Successfully setup reporting for 0x00158d0002d6f79b - lightingColorCtrl - currentY

but it's still not working. These are the log low events when I dim with the dimmer

> 2/11/2019, 12:14:42 PM - debug: Received zigbee message of type 'cmdMoveWithOnOff' with data '{"cid":"genLevelCtrl","data":{"movemode":0,"rate":195}}' of device 'TRADFRI wireless dimmer' (0xd0cf5efffe2fb1$
2/11/2019, 12:14:42 PM - debug: Received zigbee message of type 'cmdMoveWithOnOff' with data '{"cid":"genLevelCtrl","data":{"movemode":0,"rate":195}}' of device 'TRADFRI wireless dimmer' (0xd0cf5efffe2fb1$
2/11/2019, 12:14:42 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genLevelCtrl","data":{"61440":124218168,"currentLevel":0}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f79b)
2/11/2019, 12:14:42 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"61440":124218168}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f79b)
2/11/2019, 12:14:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d6f79b', payload '{"state":"ON","color_temp":153,"linkquality":120}'
2/11/2019, 12:14:42 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genLevelCtrl","data":{"61440":124218169,"currentLevel":0}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f79b)
2/11/2019, 12:14:42 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"61440":124218169}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f79b)
2/11/2019, 12:14:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d6f79b', payload '{"state":"ON","color_temp":153,"linkquality":120}'
2/11/2019, 12:14:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffe2fb15c', payload '{"battery":87,"linkquality":152,"brightness":242}'
2/11/2019, 12:14:42 PM - debug: Received zigbee message of type 'cmdStopWithOnOff' with data '{"cid":"genLevelCtrl","data":{}}' of device 'TRADFRI wireless dimmer' (0xd0cf5efffe2fb15c)
2/11/2019, 12:14:42 PM - debug: Received zigbee message of type 'cmdStopWithOnOff' with data '{"cid":"genLevelCtrl","data":{}}' of device 'TRADFRI wireless dimmer' (0xd0cf5efffe2fb15c)
2/11/2019, 12:14:42 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genLevelCtrl","data":{"61440":124218172,"currentLevel":0}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f79b)
2/11/2019, 12:14:42 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"61440":124218172}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f79b)
2/11/2019, 12:14:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d6f79b', payload '{"state":"ON","color_temp":153,"linkquality":120}'
2/11/2019, 12:14:42 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genLevelCtrl","data":{"61440":124218173,"currentLevel":0}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f79b)
2/11/2019, 12:14:42 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"61440":124218173}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f79b)
2/11/2019, 12:14:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d6f79b', payload '{"state":"ON","color_temp":153,"linkquality":120}'
2/11/2019, 12:14:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffe2fb15c', payload '{"battery":87,"linkquality":152,"brightness":242}'

Btw. if I dim from Home Assistant, the bulb reports back with the new level.

way2many, 2019-02-11:
@JumpmanJunior Same as my Hue suspension Fair. No reporting.

My "binding" was tricky. Nothing worked at all. Had to reset all the things a dozen time until I gave up and had the idea to separate both from the network for the moment. So, I first paired the dimmer with the light and then the light joined zigbee2mqtt automatically (not my intention to do that). I can now control my hue light with zigbee2mqtt and the hue dimmer switch but reporting is not working. The hue dimmer switch is also absent on the network. Will try tonight to pair the dimmer with the rest of the network and see what results I get from this.

nanosonde, 2019-02-13:
> 
> 
> About the TRADFRI remote, it doesn't allow to be bound (also experienced this). What you can do is add the bulbs to the same group as the one of the TRADFRI remote.
@Koenkk How do I do this? Just add the entry "coordinator_group:" to the target device in the configuration?

nanosonde, 2019-02-13:
Ok.
I have just read it again here: https://www.zigbee2mqtt.io/information/groups.html
The Ikea Remote and my Ikea Driver are working together using groups while being paired to zigbee2mqtt.

antti365, 2019-02-16:
@Koenkk Hi, this is me again ... :)
I try to connect the Ikea dimmer with the Ikea bulb:
2/17/2019, 12:25:17 AM - debug: Received MQTT message on 'zigbee2mqtt/bridge/bind/Ikea_Dimmer' with data 'Ikea_Bulb'
2/17/2019, 12:25:17 AM - debug: binding cluster '6' from 0x000b57fffe2b3c08' to '0x000b57fffe8d1c04'
2/17/2019, 12:25:17 AM - warn: Cannot handle MQTT message on 'zigbee2mqtt/bridge/bind/Ikea_Dimmer' with data 'Ikea_Bulb'
2/17/2019, 12:26:07 AM - error: Failed to bind cluster '6' from 0x000b57fffe2b3c08' to '0x000b57fffe8d1c04' (Error: Timed out after 10000 ms)
2/17/2019, 12:26:07 AM - debug: binding cluster '8' from 0x000b57fffe2b3c08' to '0x000b57fffe8d1c04'
2/17/2019, 12:26:57 AM - error: Failed to bind cluster '8' from 0x000b57fffe2b3c08' to '0x000b57fffe8d1c04' (Error: Timed out after 10000 ms)

Same with bundles Ikea Switch - Ikea Outlet and Ikea Switch - Ikea Bulb

Koenkk, 2019-02-16:
Did you wake the dimmer while executing the commands?

antti365, 2019-02-16:
@Koenkk Yes of course

BudBundi, 2019-02-17:
> The Ikea Remote and my Ikea Driver are working together using groups while being paired to zigbee2mqtt. I have just added a group with the ID equal to the ID of the Ikea remote control.

What are you meaning by that. Do yu sniff the group of the remot e.g. 0x0123 and than in the config  `groups:
  '0x0123':
    friendly_name: IKEA`
?
And the remote controlls the devices added to the group?

BudBundi, 2019-02-18:
> What are you meaning by that. Do yu sniff the group of the remot e.g. 0x0123 and than in the config `groups: '0x0123': friendly_name: IKEA`
> ?
> And the remote controlls the devices added to the group?

For other people who want to try it, the group id you sniff must convert to dec.

nanosonde, 2019-02-18:
@BudBundi 
Yes, I have just created a group with the same ID as I have found during sniffing. 
I have put the ID as a decimal value as you have just found out. ;)
Then, I have added the Ikea driver to the group with `zigbee2mqtt/bridge/group/[GROUP_FRIENDLY_NAME]/add` and payload `DEVICE_FRIENDLY_NAME ` of the Ikea driver.

antti365, 2019-02-23:
@Koenkk I checked in the latest version (commit # 1fda02a), a binding of light bulb and dimmer Ikea is working successfully, but the switch with the light bulb, though binded, but the light bulb is not controlled by the switch.

filoor, 2019-02-28:
> @Koenkk I checked in the latest version (commit # [1fda02a](https://github.com/Koenkk/zigbee2mqtt/commit/1fda02ae7c79b11b7a81d5a46004c64c25ada9a5)), a binding of light bulb and dimmer Ikea is [...]

Hi @Koenkk I'm having the same issue with the last dev firmware (CC2531ZNP-Prod_20190223.zip).
If i bind a ikea dimmer (ICTC-G-1) it works but when i bind a hue dimmer (324131092621)I cant control the light but i see the values in my log of the remote.

Koenkk, 2019-02-28:
@filoor I don't think this is due to the firmware, did it work with a previous firmware?

My guess it that the Hue dimmer probably sends command to the TRADFRI bulb which it doesn't understand.

filoor, 2019-02-28:
@Koenkk 
Thanks for your replay, it did I had before firmware version: '20181224' and it did work perfectly.
the bulb that i connect ti is a HUE  not a TRADFRI. Sorry I was not clear about that.

I have some 'TRADFRI' remotes connected to a Hue bulb's and that works great when i bind them.

Koenkk, 2019-02-28:
I don't see how the firmware could change this, reflashing the coorinator also shouldn't affect the configured binding

filoor, 2019-02-28:
@Koenkk 

This are my log if i bind 
`  zigbee2mqtt:debug 2019-2-28 20:53:38 binding cluster '6' from 0x00178801103e53ce' to '0x0017880100dd283b'
  zigbee2mqtt:info 2019-2-28 20:53:38 Successfully bound cluster '6' from 0x00178801103e53ce' to '0x0017880100dd283b'
  zigbee2mqtt:debug 2019-2-28 20:53:38 binding cluster '8' from 0x00178801103e53ce' to '0x0017880100dd283b'
  zigbee2mqtt:info 2019-2-28 20:53:38 Successfully bound cluster '8' from 0x00178801103e53ce' to '0x0017880100dd283b'
  zigbee2mqtt:info 2019-2-28 20:53:38 Configured reporting for cluster '6' of device '0x0017880100dd283b'
  zigbee2mqtt:info 2019-2-28 20:53:38 Configured reporting for cluster '8' of device '0x0017880100dd283b'`

**ANd when i try to control i will get a erro becouse of the brightness**

`  zigbee2mqtt:debug 2019-2-28 20:56:33 Received zigbee message of type 'cmdOn' with data '{"cid":"genOnOff","data":{}}' of device 'RWL021' (0x00178801103e53ce)
  zigbee2mqtt:info 2019-2-28 20:56:33 MQTT publish: topic 'zigbee2mqtt/remote logeerkamer', payload '{"battery":60.5,"linkquality":120,"action":"on"}'
  zigbee2mqtt:debug 2019-2-28 20:56:34 Received MQTT message on 'zigbee2mqtt/logeerkamer/set' with data '{"state": "ON", "brightness": 0}'
  zigbee2mqtt:debug 2019-2-28 20:56:34 Skipping 'state' because of 'brightness'
  zigbee2mqtt:info 2019-2-28 20:56:34 Zigbee publish to device '0x0017880100dd283b', genLevelCtrl - moveToLevelWithOnOff - {"level":0,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2019-2-28 20:56:34 MQTT publish: topic 'zigbee2mqtt/logeerkamer', payload '{"state":"ON"}'`

**SO i press on the brightness button to increase it**

`  zigbee2mqtt:debug 2019-2-28 20:58:56 Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":121}}' of device 'RWL021' (0x00178801103fc0d5)
  zigbee2mqtt:info 2019-2-28 20:58:56 MQTT publish: topic 'zigbee2mqtt/remote keuken', payload '{"battery":60.5,"linkquality":36}'
  zigbee2mqtt:debug 2019-2-28 20:58:59 Received zigbee message of type 'cmdStep' with data '{"cid":"genLevelCtrl","data":{"stepmode":0,"stepsize":30,"transtime":9}}' of device 'RWL021' (0x00178801103e53ce)
  zigbee2mqtt:info 2019-2-28 20:58:59 MQTT publish: topic 'zigbee2mqtt/remote logeerkamer', payload '{"battery":73.5,"linkquality":118,"action":"up-press","brightness":255}'
  zigbee2mqtt:debug 2019-2-28 20:58:59 Received MQTT message on 'zigbee2mqtt/logeerkamer/set' with data '{"state": "ON", "brightness": 0}'
  zigbee2mqtt:debug 2019-2-28 20:58:59 Skipping 'state' because of 'brightness'
  zigbee2mqtt:info 2019-2-28 20:58:59 Zigbee publish to device '0x0017880100dd283b', genLevelCtrl - moveToLevelWithOnOff - {"level":0,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2019-2-28 20:59:00 MQTT publish: topic 'zigbee2mqtt/logeerkamer', payload '{"state":"ON"}'`

I hope its clear let me know if you need more info.

Again Thank you for you replay

Koenkk, 2019-03-01:
I don't understand you at all, what I see in the logs is:
- 1. You press the brightness button on the remote
- 2. MQTT messages that brightness has been changed of remote
- 3. MQTT message to change brightness of corresponding bulb
- 4. Set brightness of corresponding bulb.

As far as I can see you are not using binding here, you respond to the remote control brightness up action manually (3). However, when binding is used, this shouldn't be needed.

filoor, 2019-03-01:
Sorry I was confused i had a automation running that triggered the bulb when i pressed the on button.

but it see 
Successfully bound cluster to 6 and
Successfully bound cluster to 8
When i bind the hue dimmer 

i don't see the 5 is that maybe the problem

Koenkk, 2019-03-03:
Looking at the logs, everything looks OK. To find out what is really happening you need to sniff the traffic, can you do that? (http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

BafHS, 2019-03-05:
Hello, can someone help me, to put some light into the darkness?
First of all many thanks for the great work and the time that was invested for this great product.
I start my first project:
Raspberry Pi  `Linux FHEM 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux`
 with Docker `Docker version 18.09.0, build 4d60db4`
List of container
```IMAGE                        COMMAND                  PORTS</br>                                                
koenkk/zigbee2mqtt:arm32v6   "./run.sh"                 
fhem/fhem:latest             "/entry.sh start"        
elzekool/rpi-nodered         "/usr/bin/entry.sh /…"   0.0.0.0:1880->1880/tcp 
hypriot/rpi-mysql            "/entrypoint.sh mysq…"   0.0.0.0:3306->3306/tcp, 0.0.0.0:33060->33060/tcp
pascaldevink/rpi-mosquitto   "/bin/sh -c '/usr/sb…"   0.0.0.0:1883->1883/tcp, 0.0.0.0:9001->9001/tcp
portainer/portainer          "/portainer"             0.0.0.0:9000->9000/tcp
```
Start: Zigbee2MQTT
```  zigbee2mqtt:info 2019-3-4 18:47:47 Logging to directory: '/home/pi/fhem-docker/xiaomibridge/logs/2019-03-04.18-47-47'
  zigbee2mqtt:info 2019-3-4 18:47:48 Starting zigbee2mqtt version 1.1.1 (commit #ff02e54)
  zigbee2mqtt:info 2019-3-4 18:47:48 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-3-4 18:47:49 zigbee-shepherd started
  zigbee2mqtt:info 2019-3-4 18:47:49 Coordinator firmware version: '20190109'
  zigbee2mqtt:info 2019-3-4 18:47:49 Currently 0 devices are joined:
  zigbee2mqtt:warn 2019-3-4 18:47:49 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2019-3-4 18:47:49 Allowing new devices to join.
  zigbee2mqtt:warn 2019-3-4 18:47:49 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2019-3-4 18:47:49 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-3-4 18:47:49 Connecting to MQTT server at mqtt://mqtt:1883
  zigbee2mqtt:info 2019-3-4 18:47:50 zigbee-shepherd ready
  zigbee2mqtt:info 2019-3-4 18:47:50 Connected to MQTT server
  zigbee2mqtt:info 2019-3-4 18:47:50 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 2019-3-4 18:48:19 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-3-4 18:47:47 Logging to directory: '/home/pi/fhem-docker/xiaomibridge/logs/2019-03-04.18-47-47'
  zigbee2mqtt:info 2019-3-4 18:47:48 Starting zigbee2mqtt version 1.1.1 (commit #ff02e54)
  zigbee2mqtt:info 2019-3-4 18:47:48 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-3-4 18:47:49 zigbee-shepherd started
  zigbee2mqtt:info 2019-3-4 18:47:49 Coordinator firmware version: '20190109'
  zigbee2mqtt:info 2019-3-4 18:47:49 Currently 0 devices are joined:
```
Start pairing find the remote:
```
  zigbee2mqtt:info 2019-3-4 18:49:23 Device incoming...
  zigbee2mqtt:info 2019-3-4 18:49:23 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 2019-3-4 18:49:23 New device with address 0xd0cf5efffec0bb59 connected!
  zigbee2mqtt:info 2019-3-4 18:49:23 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0xd0cf5efffec0bb59"}'
  zigbee2mqtt:info 2019-3-4 18:49:24 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0xd0cf5efffec0bb59","type":"EndDevice","model":"E1524","friendly_name":"0xd0cf5efffec0bb59"}]}'
  zigbee2mqtt:info 2019-3-4 18:49:40 Connecting with device...
  zigbee2mqtt:info 2019-3-4 18:49:40 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2019-3-4 18:49:40 Connecting with device...
  zigbee2mqtt:info 2019-3-4 18:49:40 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2019-3-4 18:49:40 Connecting with device...
  zigbee2mqtt:info 2019-3-4 18:49:40 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:warn 2019-3-4 18:49:41 Message without device!
````
Start pairing find the bulb
````  zigbee2mqtt:info 2019-3-4 18:49:44 Device incoming...
  zigbee2mqtt:info 2019-3-4 18:49:44 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 2019-3-4 18:49:44 New device with address 0xd0cf5efffe12348a connected!
  zigbee2mqtt:info 2019-3-4 18:49:44 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0xd0cf5efffe12348a"}'
  zigbee2mqtt:info 2019-3-4 18:49:47 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0xd0cf5efffe12348a","type":"Router","model":"LED1545G12","friendly_name":"0xd0cf5efffe12348a"}]}'
  zigbee2mqtt:info 2019-3-4 18:50:37 Device incoming...
  zigbee2mqtt:info 2019-3-4 18:50:37 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
````
Define my Device
```  zigbee2mqtt:info 2019-3-4 18:55:07 Successfully renamed - 0xd0cf5efffe12348a to Gaestezimmer_Lampe
  zigbee2mqtt:info 2019-3-4 18:55:07 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0xd0cf5efffe12348a","type":"Router","model":"LED1545G12","friendly_name":"Gaestezimmer_Lampe"},{"ieeeAddr":"0xd0cf5efffec0bb59","type":"EndDevice","model":"E1524","friendly_name":"0xd0cf5efffec0bb59"}]}'
[... cut ...]
  zigbee2mqtt:info 2019-3-4 19:17:15 MQTT publish: topic 'zigbee2mqtt/Gaestezimmer_Lampe', payload '{"state":"OFF","brightness":169,"color_temp":320,"color_mode":2,"color":{"x":0.46,"y":0.411}}'
````
Remote Control not working? Where is my failure???
I plan the final enviroment ikea Tradfri and Xiaomi Devices, like pir sensor
At the moment i  have no idee...
Design question, if my FHEM not available, I can not turn any lights? Or is there a backup solution, say the controllers directly pair with the lamp and switch the lamp without FHEM?
Does it make sense to pick up the Tradfri Gateway? 
It's about the WAF ... i dont need trouble..
thx a lot
regards fom Berlin

Koenkk, 2019-03-05:
@BafHS it partially fails to bind:

```
  zigbee2mqtt:error 2019-3-4 19:14:12 Failed to bind cluster '5' from 0xd0cf5efffec0bb59' to '0xd0cf5efffe12348a' (Error: Timed out after 10000 ms)
```

You need to make sure your remote is awake at the time you execute the binding command (by pressing a some random buttons on it).

BafHS, 2019-03-06:
@Koenkk  thx that okay, but when i push a buttom on the E1524 the bulb nothing receive.
```` zigbee2mqtt:warn 2019-3-6 07:39:24 Cannot handle MQTT message on 'zigbee2mqtt/bridge/bind/0xd0cf5efffec0bb59' with data '0xd0cf5efffe12348a'
  zigbee2mqtt:info 2019-3-6 07:39:25 Successfully bound cluster '5' from 0xd0cf5efffec0bb59' to '0xd0cf5efffe12348a'
  zigbee2mqtt:warn 2019-3-6 07:39:27 Cannot handle MQTT message on 'zigbee2mqtt/bridge/bind/0xd0cf5efffec0bb59' with data '0xd0cf5efffe12348a'
  zigbee2mqtt:info 2019-3-6 07:39:27 Successfully bound cluster '6' from 0xd0cf5efffec0bb59' to '0xd0cf5efffe12348a'
  zigbee2mqtt:info 2019-3-6 07:39:29 Successfully bound cluster '8' from 0xd0cf5efffec0bb59' to '0xd0cf5efffe12348a'
  zigbee2mqtt:warn 2019-3-6 07:39:29 Don't know how to setup reporting for '5', skipping..
  zigbee2mqtt:info 2019-3-6 07:39:31 Successfully bound cluster '5' from 0xd0cf5efffec0bb59' to '0xd0cf5efffe12348a'
  zigbee2mqtt:info 2019-3-6 07:39:33 Successfully bound cluster '6' from 0xd0cf5efffec0bb59' to '0xd0cf5efffe12348a'
  zigbee2mqtt:info 2019-3-6 07:39:35 Successfully bound cluster '8' from 0xd0cf5efffec0bb59' to '0xd0cf5efffe12348a'
  zigbee2mqtt:warn 2019-3-6 07:39:35 Don't know how to setup reporting for '5', skipping..
  zigbee2mqtt:info 2019-3-6 07:39:35 Configured reporting for cluster '6' of device '0xd0cf5efffe12348a'
  zigbee2mqtt:info 2019-3-6 07:39:35 Configured reporting for cluster '8' of device '0xd0cf5efffe12348a'
  zigbee2mqtt:info 2019-3-6 07:39:35 Configured reporting for cluster '6' of device '0xd0cf5efffe12348a'
  zigbee2mqtt:info 2019-3-6 07:39:36 Configured reporting for cluster '8' of device '0xd0cf5efffe12348a'
````
Only Cluster '5'
But my first try, i pair only the E1524. I dont now why?
````  zigbee2mqtt:info 2019-3-6 07:34:23 New device with address 0xd0cf5efffe12348a connected!
  zigbee2mqtt:info 2019-3-6 07:34:23 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0xd0cf5efffe12348a"}'
  zigbee2mqtt:warn 2019-3-6 07:34:23 Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 2019-3-6 07:34:23 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
````

Koenkk, 2019-03-06:
I just realised you are using the E1524, this is a special case, please see the [OP for how to bind this device](https://github.com/Koenkk/zigbee2mqtt/issues/782#issue-394902886)

[274 more replies]

closed 2020-09-10 by github-actions[bot]