Koenkk/zigbee2mqtt #786: Invalid checksum for motion sensor RTCGQ11LM

peterforeman, 2018-12-31:
I have three RTCGQ11LM's paired with my CC2530. Running latest coordinator firmware and zigbee2mqtt 1.0.1. After pairing they never report motion ("occupancy"), only a message every hour or so.

Debugging gives me this:

```
  cc-znp Error: Invalid checksum
  cc-znp     at CcZnp._parseMtIncomingData (/app/node_modules/cc-znp/lib/ccznp.js:323:19)
  cc-znp     at Unpi.parseMtIncomingData (/app/node_modules/cc-znp/lib/ccznp.js:92:18)
  cc-znp     at emitOne (events.js:116:13)
  cc-znp     at Unpi.emit (events.js:211:7)
  cc-znp     at Dissolve.<anonymous> (/app/node_modules/unpi/index.js:107:14)
  cc-znp     at emitOne (events.js:116:13)
  cc-znp     at Dissolve.emit (events.js:211:7)
  cc-znp     at Dissolve.<anonymous> (/app/node_modules/dissolve-chunks/index.js:73:29)
  cc-znp     at emitNone (events.js:106:13)
  cc-znp     at Dissolve.emit (events.js:208:7) +8s
```
(complete log: https://pastebin.com/aPHPbNgF)

Tried re-pairing, flashing CC2530 again etc.etc.. Any ideas?

Koenkk, 2019-01-01:
Could you post your log with `log_level: debug` ? (https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html)

chaertl1, 2019-01-11:
I've got the same problem with an Aqara motion sensor

Koenkk, 2019-01-11:
How is the CC2530 connected?

Koenkk, 2019-01-12:
Which pins did you connect?

chaertl1, 2019-01-12:
CC2530  CP210X
VCC       3V3 
P02        TXD
P01        RXD
GND      GND

peterforeman, 2019-01-12:
Same for me. Attached with USB<>Serial CP210X with same pins. Everything worked fine, except for the motion sensor which reported wrong checksums.

Koenkk, 2019-01-12:
I will check this once I receive my CP210x

chaertl1, 2019-01-13:
Good morning.
I also tested with a CH340 USB->serial converter with the same results.
I will try to find a controller that supports rtscts.

chaertl1, 2019-01-21:
@peterforeman 
Can you reopen this topic please.
It looks like more people are having problems with the human body sensors on a cc2530

wimpie007, 2019-02-03:
got the samen issue.
Body sensor RTCGQ11LM stops working after upgrading.
Sensor reports data, and is seen in the log, but "Device with modelID 'undefined' is not supported" show up.... Sensor useless on this moment.
Both in 'stable' and in edge version (hassio)

closed 2019-04-11 by stale[bot]