Koenkk/zigbee2mqtt #788: Color ignored when specifying brightness in automation with Hue color bulb

kiwinol, 2018-12-31:
When turning on my hue lights with an automation in homeassistant if I provide both brightness and color as parameters color is ignored.

Example:
```
 - alias: Adam Light Switch
   trigger:
     platform: mqtt
     topic: 'zigbee2mqtt/Adams_Switch'
   condition:
     condition: and
     conditions:
      - condition: template
        value_template: "{{ 'single' == trigger.payload_json.click}}"
      - condition: state
        entity_id: light.0x0017880110460c0f_light
        state: 'off'
   action:
     - service: homeassistant.turn_on
       entity_id: light.0x0017880110460c0f_light
       data:
         hs_color: [16.471,100]
         brightness: 255
```
See log here:
[https://pastebin.com/wMAiA8xw](url)

The only way around this I have found is to to turn on the light with the turn_on service then set the color afterwards as a secondary action.

Koenkk, 2019-01-01:
@khzd do you also have hue bulbs? (as this works as expected on my innr bulb)

khzd, 2019-01-02:
Hi
no I have GLEDOPTO (008), for this I get yhis "warning" Jan 01 19:21:00 ubuntu59 npm[15784]: zigbee2mqtt:debug 1/1/2019, 7:21:00 PM Skipping 'state' because of 'brightness'
so I cannot changed it, it work before 1.0.1 if I'm not wrong

Koenkk, 2019-01-05:
Can you verify it worked in an earlier version?

kiwinol, 2019-01-05:
It worked with the last version before 1.x upgrade.

Koenkk, 2019-01-06:
Would you mind posting the logging before the 1.x upgrade?

AnhDuc85, 2019-01-07:
it's working on my end (can set color and brightness at once), running latest-dev and current firmware, here's my automation: https://pastebin.com/zTgVATC4

kiwinol, 2019-01-08:
@AnhDuc85 I have just tested with the latest edge version of:
```
https://github.com/danielwelch/hassio-zigbee2mqtt
```
And it still doesnt work for me.

Are you using HASSIO?

Also which firmware are you running? I am using:
```
"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20181024}
```

philhawthorne, 2019-01-17:
I'm also experincing this with Hue bulbs running the latest firmware and `dev` branch. (All npm_modules also refreshed).

Looking at the converters, I am not sure if the color is actually being sent when power on is called. See here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/toZigbee.js#L98

I think that needs to be updated to send the colour the bulb should turn on to

Koenkk, 2019-01-18:
@philhawthorne the color is send by a separate converter, if you look at the zigbee messages being send, you see that `genOnOff` and `lightingColorCtrl` messages are send. The problem is that the bulb doesn't execute the `lightingColorCtrl` cmd.

philhawthorne, 2019-01-21:
Cheers @Koenkk 

I'll test this out tonight and confirm!

closed 2019-01-22 by Koenkk