Koenkk/zigbee2mqtt #790: No serial interface on USB Zigbee sniffer with CC2530

untitledlt, 2019-01-01:
I know it's not zigbee2mqtt related but i got stuck in an early stage.
I have flashed my USB sniffer with CC2530 as per http://ptvo.info/how-to-select-and-flash-cc2530-144/ instructions. 
Flashing went without any errors but i'm not sure what i should do next.
Device does not list serial interface (tried ubuntu and os x). 
Is USB for flashing only?
Device i'm using: https://www.aliexpress.com/item/ZigBee-CC2531-USB-Dongle/1830760873.html?spm=2114.search0104.3.186.74ca6ec0pjSnhC&ws_ab_test=searchweb0_0,searchweb201602_2_10065_10068_10890_319_10546_317_10548_10696_453_10084_454_10083_10618_10307_10820_10303_537_536_10059_10884_10887_100031_321_322_10103-10890,searchweb201603_51,ppcSwitch_0&algo_expid=944b3f07-4922-4e16-b005-1570f6258d6a-27&algo_pvid=944b3f07-4922-4e16-b005-1570f6258d6a

untitledlt, 2019-01-02:
what do you mean `debug firmware`? 
I tried `CC2530ZNP-Pro-Secure_LinkKeyJoin.hex`, `CC2530ZNP-SB.hex` from Z-Stack Home Automation SDK and few others what i could find. 
Is there a separate debug version?

untitledlt, 2019-01-02:
@dossidr thanks for an answer. There are 5 pins on module but they are not labeled and i can't find a pinout online. Do you know which pins are TX/RX?

dossidr, 2019-01-02:
Why do you need RX/TX? The Stick self has an USB Slot. The USB you plugged in your Linux Server an see "dmesg" == /dev/ttyUSB or /dev/ttyACM0 after the successfully flashed Firmware. These 5 Pins you only need to be flashed the Firmware with CC-Debugger direct. I think that is your misunderstanding how to use the sniffer USB stick?! Best regards

dossidr, 2019-01-02:
You can flash your Device directy with CC-Debugger. Connect you can see page 9 *pdf File
[Zigbee_CC2530_CC2531.pdf](https://github.com/Koenkk/zigbee2mqtt/files/2721681/Zigbee_CC2530_CC2531.pdf)

sehraf, 2019-01-02:
`SmartRF04EB` is the chip used to flash the actual CC253X.. does this USB stick combine both chips (usually you have _two_ sticks, one zigbee stick and one debugger/flasher stick). To me it looks like you have both on one board and USB is connected to the debugger/flasher.

untitledlt, 2019-01-02:
CC2531 has built-in USB support but i have an older CC2530 without USB. 
This module has c8051f320 chip which adds USB support and and flasher/programmer. Also this module has 5 pins header but it's not labeled. First pin should be VCC and GND is the last one. Don't know about the rest but all of them have 3.2V so I don't know which is TX/RX if at all.

sehraf, 2019-01-02:
On [this](https://nl.aliexpress.com/item/CC2530-USB-serial-port-zigbeeOTA-support-Bootloader-znp-sniffer/32808298773.html) device the pins are 3,3V, RST, P2.1 ❔, P2.2 ❔ and GND as seen [here](https://ae01.alicdn.com/kf/HTB1WEfAQVXXXXcbXpXXq6xXFXXXz/110912934/HTB1WEfAQVXXXXcbXpXXq6xXFXXXz.jpg) ... which is not as helpfull as expected ...

maxic62, 2019-01-03:
Hello Guys,
I'm facing the same trouble here with my Pi 3
I've flashed my CC2531 USB EMK with this firmware :
https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2531/bin/CC2531ZNP-Prod_20181024.zip
It works great with my computer on ubuntu with zigbee2mqtt (can paire Xiaomi sensors and get informations from them) but not with my raspberry pi 3
No /dev/ACM0 appears!

info:
uname -a
`
Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux`

lsusb
` 
Bus 001 Device 007: ID 0451:16a8 Texas Instruments, Inc.
Bus 001 Device 004: ID 1058:25a2 Western Digital Technologies, Inc.
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

dmesg
`
...
[58981.343316] usb 1-1.3: new full-speed USB device number 7 using dwc_otg
[58981.492274] usb 1-1.3: New USB device found, idVendor=0451, idProduct=16a8
[58981.492289] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[58981.492297] usb 1-1.3: Product: TI CC2531 USB CDC
[58981.492305] usb 1-1.3: Manufacturer: Texas Instruments
[58981.492312] usb 1-1.3: SerialNumber: __0X00124B0018ED2204
`

ls -la /dev/ttyA*
`
crw-rw---- 1 root dialout 204, 64 Jan  2 21:42 /dev/ttyAMA0
`

Thanks for your help ;)

maxic62, 2019-01-04:
Hello guys,
Something new on my side. I got a new raspberry PI3B+ and fresh rasbian install. All works now !

Something goes wrong on my previous raspberry + raspbian installation:
lsmod output is empty on the bad one, it means the cdc_acm module cannot be loaded correctly !
Now i have to figure out why my /lib/modules/ does not match with my kernel version

ls /lib/modules/
` 
4.14.71+  4.14.71-v7+  4.14.74+  4.14.74-v7+
`

uname -a
`Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux`

do I have to update my current kernel ? Is it risky for the system ?

Thanks

maxic62, 2019-01-08:
Hello guys,
problem solved, finally a simple` apt update, apt upgrade ` has synchronized my kernel and /lib/modules folder. Both are with the latest kernel and finally my CC2531 works very well 👍

trancefreak77, 2019-03-03:
@maxic62: I have the same USB dongle as you have. Did you get it to work with a raspberry pi running linux? I have the same problem that no tty device is available when plugging in the dongle.

untitledlt, 2019-03-03:
I've finished buying CC2531

trancefreak77, 2019-03-03:
What do you mean?
My dongle is 2530.

untitledlt, 2019-03-04:
I wasn't able to make it work on CC2530 and bought CC2531.
Flashing was smooth and everything works now.

trancefreak77, 2019-03-04:
@untitledlt: Thanks for your reply! Looks like I also have to use the 2531 instead of 2530. Cheers!

closed 2019-05-10 by stale[bot]