Koenkk/zigbee2mqtt #791: Docker image, config file not used?

greynoO, 2019-01-01:
I am trying to run the latest docker image on a Raspberry PI 3B+. Everything was done as described here:

https://koenkk.github.io/zigbee2mqtt/information/docker.html

First of all, the --device command does nothing for me, the created configuration.yaml still has `/dev/ttyACM0` but I want `/dev/tty/USB0`.

Even after I modify the config accordingly, I still get this error:

```
{"message":"Error: No such file or directory, cannot open /dev/ttyACM0","stack":"Error: Error: No such file or directory, cannot open /dev/ttyACM0"}
```

Any help would be appreciated! Thanks!

greynoO, 2019-01-01:
Was only an issue when using as standalone installation not with hass.io

Koenkk, 2019-01-01:
`--device` is a docker command, it's not related to zigbee2mqtt. 

`--device` is need to map the device inside the docker image. This device than has to be specified in `configuration.yaml`.

greynoO, 2019-01-01:
Thanks, maybe add 

`--device=/dev/ttyACM0`: Location of CC2531 USB sniffer (as defined in configuration.yaml)

to make that clear. Maybe I was just too tired already :)

closed 2019-01-01 by greynoO