Koenkk/zigbee2mqtt #792: IKEA TRADFRI wireless dimmer (ICTC-G-1) How to get the Battery level ?

antti365, 2019-01-01:
Hi! A few days ago I started using zigbee2mqtt with IKEA devices. I noticed that in the MQTT line, along with the dimmer status, the battery charge level is not transmitted. Is there any way to add it there? I used to have a native gateway and, as far as I remember, there was an indication of battery charge, or am I mistaken?

RaveGun, 2019-01-08:
The is an deal ongoing to the a kit consisting of lightbulb and a dimmer switch at IKEA and I ordered 2 sets. I will report back after I get them if it behaves the same.

Do you run ZigBee2Mqtt in a docker container?

antti365, 2019-01-09:
Hi, 
No, I don’t use docker. Zigbee2mqtt is launched via systemd on Rpi3 running Rasbian Stretch. I just thought it was logical that all devices on batteries should report the status of the battery in some way, in this case I only get the dimmer value and link quality. 
Thanks for the answer.

Koenkk, 2019-01-09:
Can you update the section of the TRADFRI dimmer in `devices.js` to 

```js
    {
        zigbeeModel: ['TRADFRI wireless dimmer'],
        model: 'ICTC-G-1',
        vendor: 'IKEA',
        description: 'TRADFRI wireless dimmer',
        supports: 'brightness [0-255], quick rotate for instant 0/255',
        fromZigbee: [
            fz.ICTC_G_1_move, fz.ICTC_G_1_moveWithOnOff, fz.ICTC_G_1_stop, fz.ICTC_G_1_stopWithOnOff,
            fz.ICTC_G_1_moveToLevelWithOnOff,
        ],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.bind('genLevelCtrl', coordinator, cb),
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, 3, 0, cb),
            ];
            execute(device, actions, callback);
        },
    },
```

antti365, 2019-01-09:
I updated the TRADFRI dimmer section in devices.js as you specified, turned the dimmer several times and started zigbee2mqtt.
Result:
Jan 10 00:20:54 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:20:54 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-01-10.00-20-54'
Jan 10 00:20:55 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:20:55 Starting zigbee2mqtt version 1.0.1 (commit #9380bbc)
Jan 10 00:20:55 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:20:55 Starting zigbee-shepherd
Jan 10 00:20:56 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:20:56 zigbee-shepherd started
Jan 10 00:20:56 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:20:56 Coordinator firmware version: '2018090201'
Jan 10 00:20:56 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:20:56 Currently 2 devices are joined:
Jan 10 00:20:56 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:20:56 Entrance_Lamp (0x000b57fffe33d833): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
Jan 10 00:20:56 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:20:56 Entrance_Dimmer (0x000b57fffe2b3c08): ICTC-G-1 - IKEA TRADFRI wireless dimmer (EndDevice)
Jan 10 00:20:56 raspberrypi3 npm[28929]:   zigbee2mqtt:warn 2019-1-10 00:20:56 `permit_join` set to  `true` in configuration.yaml.
Jan 10 00:20:56 raspberrypi3 npm[28929]:   zigbee2mqtt:warn 2019-1-10 00:20:56 Allowing new devices to join.
Jan 10 00:20:56 raspberrypi3 npm[28929]:   zigbee2mqtt:warn 2019-1-10 00:20:56 Set `permit_join` to `false` once you joined all devices.
Jan 10 00:20:56 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:20:56 Zigbee: allowing new devices to join.
Jan 10 00:20:56 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:20:56 Connecting to MQTT server at mqtt://localhost
Jan 10 00:20:56 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:20:56 zigbee-shepherd ready
Jan 10 00:20:56 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:20:56 Connected to MQTT server
Jan 10 00:20:56 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:20:56 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Jan 10 00:20:56 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:20:56 MQTT publish: topic 'zigbee2mqtt/Entrance_Lamp', payload '{"state":"OFF","brightness":1}'
Jan 10 00:21:13 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:21:13 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":255,"linkquality":39}'
Jan 10 00:21:16 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:21:16 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":39}'
Jan 10 00:21:17 raspberrypi3 npm[28929]:   zigbee2mqtt:error 2019-1-10 00:21:17 Failed to configure Entrance_Dimmer 0x000b57fffe2b3c08
Jan 10 00:21:18 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:21:18 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":255,"linkquality":39}'
Jan 10 00:21:20 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:21:20 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":255,"linkquality":39}'
Jan 10 00:21:20 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:21:20 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":255,"linkquality":39}'
Jan 10 00:21:22 raspberrypi3 npm[28929]:   zigbee2mqtt:info 2019-1-10 00:21:22 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":39}'

Returned to the old version devices.js file.
Result:
Jan 10 00:22:54 raspberrypi3 npm[30785]:   zigbee2mqtt:info 2019-1-10 00:22:54 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-01-10.00-22-54'
Jan 10 00:22:55 raspberrypi3 npm[30785]:   zigbee2mqtt:info 2019-1-10 00:22:55 Starting zigbee2mqtt version 1.0.1 (commit #9380bbc)
Jan 10 00:22:55 raspberrypi3 npm[30785]:   zigbee2mqtt:info 2019-1-10 00:22:55 Starting zigbee-shepherd
Jan 10 00:22:56 raspberrypi3 npm[30785]:   zigbee2mqtt:info 2019-1-10 00:22:56 zigbee-shepherd started
Jan 10 00:22:56 raspberrypi3 npm[30785]:   zigbee2mqtt:info 2019-1-10 00:22:56 Coordinator firmware version: '2018090201'
Jan 10 00:22:56 raspberrypi3 npm[30785]:   zigbee2mqtt:info 2019-1-10 00:22:56 Currently 2 devices are joined:
Jan 10 00:22:56 raspberrypi3 npm[30785]:   zigbee2mqtt:info 2019-1-10 00:22:56 Entrance_Lamp (0x000b57fffe33d833): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
Jan 10 00:22:56 raspberrypi3 npm[30785]:   zigbee2mqtt:info 2019-1-10 00:22:56 Entrance_Dimmer (0x000b57fffe2b3c08): ICTC-G-1 - IKEA TRADFRI wireless dimmer (EndDevice)
Jan 10 00:22:56 raspberrypi3 npm[30785]:   zigbee2mqtt:warn 2019-1-10 00:22:56 `permit_join` set to  `true` in configuration.yaml.
Jan 10 00:22:56 raspberrypi3 npm[30785]:   zigbee2mqtt:warn 2019-1-10 00:22:56 Allowing new devices to join.
Jan 10 00:22:56 raspberrypi3 npm[30785]:   zigbee2mqtt:warn 2019-1-10 00:22:56 Set `permit_join` to `false` once you joined all devices.
Jan 10 00:22:56 raspberrypi3 npm[30785]:   zigbee2mqtt:info 2019-1-10 00:22:56 Zigbee: allowing new devices to join.
Jan 10 00:22:56 raspberrypi3 npm[30785]:   zigbee2mqtt:info 2019-1-10 00:22:56 Connecting to MQTT server at mqtt://localhost
Jan 10 00:22:56 raspberrypi3 npm[30785]:   zigbee2mqtt:info 2019-1-10 00:22:56 zigbee-shepherd ready
[... cut ...]
Jan 10 00:23:26 raspberrypi3 npm[30785]:   zigbee2mqtt:info 2019-1-10 00:23:26 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":255,"linkquality":42}'
Jan 10 00:23:26 raspberrypi3 npm[30785]:   zigbee2mqtt:info 2019-1-10 00:23:26 Succesfully configured Entrance_Dimmer 0x000b57fffe2b3c08
Jan 10 00:23:27 raspberrypi3 npm[30785]:   zigbee2mqtt:info 2019-1-10 00:23:27 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":42}'

Did I do something wrong or a mistake in the code? Thank you very much for your help!

antti365, 2019-01-10:
Hi, set the dev branch, the result has changed, but in my opinion not quite the way it should ... :)

Jan 10 21:43:58 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:43:58 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-01-10.21-43-57'
Jan 10 21:43:59 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:43:59 Starting zigbee2mqtt version 1.0.1 (commit #6dd60c0)
Jan 10 21:43:59 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:43:59 Starting zigbee-shepherd
Jan 10 21:44:00 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:44:00 zigbee-shepherd started
Jan 10 21:44:00 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:44:00 Coordinator firmware version: '2018090201'
Jan 10 21:44:00 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:44:00 Currently 2 devices are joined:
Jan 10 21:44:00 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:44:00 Entrance_Lamp (0x000b57fffe33d833): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
Jan 10 21:44:00 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:44:00 Entrance_Dimmer (0x000b57fffe2b3c08): ICTC-G-1 - IKEA TRADFRI wireless dimmer (EndDevice)
Jan 10 21:44:00 raspberrypi3 npm[32552]:   zigbee2mqtt:warn 2019-1-10 21:44:00 `permit_join` set to  `true` in configuration.yaml.
Jan 10 21:44:00 raspberrypi3 npm[32552]:   zigbee2mqtt:warn 2019-1-10 21:44:00 Allowing new devices to join.
Jan 10 21:44:00 raspberrypi3 npm[32552]:   zigbee2mqtt:warn 2019-1-10 21:44:00 Set `permit_join` to `false` once you joined all devices.
Jan 10 21:44:00 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:44:00 Zigbee: allowing new devices to join.
Jan 10 21:44:00 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:44:00 Connecting to MQTT server at mqtt://localhost
Jan 10 21:44:00 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:44:00 zigbee-shepherd ready
Jan 10 21:44:00 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:44:00 Connected to MQTT server
Jan 10 21:44:00 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:44:00 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Jan 10 21:44:01 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:44:01 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":18,"last_seen":"2019-01-10T19:44:01.021Z"}'
Jan 10 21:44:02 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:44:02 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":255,"linkquality":18,"last_seen":"2019-01-10T19:44:02.276Z"}'
Jan 10 21:44:02 raspberrypi3 npm[32552]:   zigbee2mqtt:error 2019-1-10 21:44:02 Failed to configure Entrance_Dimmer (0x000b57fffe2b3c08) ('Error: Profile: 49246 is not supported.')
Jan 10 21:44:02 raspberrypi3 npm[32552]:   zigbee2mqtt:info 2019-1-10 21:44:02 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":18,"last_seen":"2019-01-10T19:44:02.907Z"}'

antti365, 2019-01-10:
I used the old base and configuration file? Maybe I needed to pair the device again?

Koenkk, 2019-01-10:
Yes, can you try repairing?

antti365, 2019-01-12:
Hi, sorry, I wanted to clarify where we left off? Is there a problem on my side or am I waiting for the code to be amended?

Koenkk, 2019-01-13:
@antti365 I have to make some more changes

Koenkk, 2019-01-14:
Can you update to the dev branch and try again?

antti365, 2019-01-14:
Yes of course. Is it enough to stop zigbee2mqtt, delete / node_modules and repeat npm install?

Koenkk, 2019-01-14:
you also need to update zigbee2mqtt itself (git pull)

antti365, 2019-01-14:
Yes, I already tried and made a complete update, now like this ...
Jan 14 20:04:00 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:04:00 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-01-14.20-04-00'
Jan 14 20:04:01 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:04:01 Starting zigbee2mqtt version 1.0.1 (commit #9605e72)
Jan 14 20:04:01 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:04:01 Starting zigbee-shepherd
Jan 14 20:04:02 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:04:02 zigbee-shepherd started
Jan 14 20:04:02 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:04:02 Coordinator firmware version: '2018090201'
Jan 14 20:04:02 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:04:02 Currently 2 devices are joined:
Jan 14 20:04:02 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:04:02 Entrance_Lamp (0x000b57fffe33d833): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
Jan 14 20:04:02 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:04:02 Entrance_Dimmer (0x000b57fffe2b3c08): ICTC-G-1 - IKEA TRADFRI wireless dimmer (EndDevice)
Jan 14 20:04:02 raspberrypi3 npm[2231]:   zigbee2mqtt:warn 2019-1-14 20:04:02 `permit_join` set to  `true` in configuration.yaml.
Jan 14 20:04:02 raspberrypi3 npm[2231]:   zigbee2mqtt:warn 2019-1-14 20:04:02 Allowing new devices to join.
Jan 14 20:04:02 raspberrypi3 npm[2231]:   zigbee2mqtt:warn 2019-1-14 20:04:02 Set `permit_join` to `false` once you joined all devices.
Jan 14 20:04:02 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:04:02 Zigbee: allowing new devices to join.
Jan 14 20:04:02 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:04:02 Connecting to MQTT server at mqtt://localhost
Jan 14 20:04:02 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:04:02 zigbee-shepherd ready
Jan 14 20:04:02 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:04:02 Connected to MQTT server
Jan 14 20:04:02 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:04:02 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Jan 14 20:04:52 raspberrypi3 npm[2231]:   zigbee2mqtt:error 2019-1-14 20:04:52 Failed to configure Entrance_Dimmer (0x000b57fffe2b3c08) ('Error: Timed out after 10000 ms')
Jan 14 20:05:09 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:05:09 Zigbee publish to device '0x000b57fffe33d833', genLevelCtrl - moveToLevelWithOnOff - {"level":254,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
Jan 14 20:05:09 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:05:09 MQTT publish: topic 'zigbee2mqtt/Entrance_Lamp', payload '{"state":"ON"}'
Jan 14 20:05:11 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:05:11 Zigbee publish to device '0x000b57fffe33d833', genLevelCtrl - moveToLevelWithOnOff - {"level":0,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
Jan 14 20:05:11 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:05:11 MQTT publish: topic 'zigbee2mqtt/Entrance_Lamp', payload '{"state":"ON"}'
Jan 14 20:05:12 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:05:12 MQTT publish: topic 'zigbee2mqtt/Entrance_Lamp', payload '{"state":"ON","brightness":1,"last_seen":"2019-01-14T18:05:12.218Z"}'
Jan 14 20:05:45 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:05:45 Device incoming...
Jan 14 20:05:45 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:05:45 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
Jan 14 20:05:55 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:05:55 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":15,"last_seen":"2019-01-14T18:05:55.065Z"}'
Jan 14 20:05:55 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:05:55 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":255,"linkquality":15,"last_seen":"2019-01-14T18:05:55.666Z"}'
Jan 14 20:05:56 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:05:56 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":15,"last_seen":"2019-01-14T18:05:56.351Z"}'
Jan 14 20:06:21 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:06:21 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":25,"linkquality":15,"last_seen":"2019-01-14T18:06:21.249Z"}'
Jan 14 20:06:22 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:06:22 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":21,"linkquality":15,"last_seen":"2019-01-14T18:06:22.992Z"}'
Jan 14 20:06:25 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:06:25 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":21,"linkquality":15,"last_seen":"2019-01-14T18:06:25.865Z"}'
Jan 14 20:06:26 raspberrypi3 npm[2231]:   zigbee2mqtt:info 2019-1-14 20:06:26 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":15,"last_seen":"2019-01-14T18:06:26.291Z"}'

Koenkk, 2019-01-14:
Did you wake the device when starting zigbee2mqtt?

antti365, 2019-01-14:
Yes, turned the dimmer several times. Now I will try again to reset the dimmer, connect it to zigbee2mqtt and then tell about the results.

antti365, 2019-01-14:
Once again, I completely reinstalled zigbee2mqtt, created a new database, pulled out, inserted a stick, pressed the button closer to usb, dropped the dimmer and connected it to zigbee2mqtt, the error changed:
Jan 14 20:52:47 raspberrypi3 npm[13615]:   zigbee2mqtt:info 2019-1-14 20:52:47 Coordinator firmware version: '2018090201'
Jan 14 20:52:47 raspberrypi3 npm[13615]:   zigbee2mqtt:info 2019-1-14 20:52:47 Currently 0 devices are joined:
Jan 14 20:52:47 raspberrypi3 npm[13615]:   zigbee2mqtt:warn 2019-1-14 20:52:47 `permit_join` set to  `true` in configuration.yaml.
Jan 14 20:52:47 raspberrypi3 npm[13615]:   zigbee2mqtt:warn 2019-1-14 20:52:47 Allowing new devices to join.
Jan 14 20:52:47 raspberrypi3 npm[13615]:   zigbee2mqtt:warn 2019-1-14 20:52:47 Set `permit_join` to `false` once you joined all devices.
Jan 14 20:52:47 raspberrypi3 npm[13615]:   zigbee2mqtt:info 2019-1-14 20:52:47 Zigbee: allowing new devices to join.
Jan 14 20:52:47 raspberrypi3 npm[13615]:   zigbee2mqtt:info 2019-1-14 20:52:47 Connecting to MQTT server at mqtt://localhost
Jan 14 20:52:48 raspberrypi3 npm[13615]:   zigbee2mqtt:info 2019-1-14 20:52:48 zigbee-shepherd ready
Jan 14 20:52:48 raspberrypi3 npm[13615]:   zigbee2mqtt:info 2019-1-14 20:52:48 Connected to MQTT server
Jan 14 20:52:48 raspberrypi3 npm[13615]:   zigbee2mqtt:info 2019-1-14 20:52:48 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Jan 14 20:53:32 raspberrypi3 npm[13615]:   zigbee2mqtt:info 2019-1-14 20:53:32 Connecting with device...
Jan 14 20:53:32 raspberrypi3 npm[13615]:   zigbee2mqtt:info 2019-1-14 20:53:32 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
… more messages
Jan 14 20:54:03 raspberrypi3 npm[13615]:   zigbee2mqtt:info 2019-1-14 20:54:03 Connecting with device...
Jan 14 20:54:03 raspberrypi3 npm[13615]:   zigbee2mqtt:info 2019-1-14 20:54:03 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Jan 14 20:54:03 raspberrypi3 npm[13615]:   zigbee2mqtt:info 2019-1-14 20:54:03 Device incoming...
Jan 14 20:54:03 raspberrypi3 npm[13615]:   zigbee2mqtt:info 2019-1-14 20:54:03 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
Jan 14 20:54:03 raspberrypi3 npm[13615]:   zigbee2mqtt:info 2019-1-14 20:54:03 New device with address 0x000b57fffe2b3c08 connected!
Jan 14 20:54:03 raspberrypi3 npm[13615]:   zigbee2mqtt:info 2019-1-14 20:54:03 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000b57fffe2b3c08"}'
Jan 14 20:54:07 raspberrypi3 npm[13615]: /opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/af.js:269
Jan 14 20:54:07 raspberrypi3 npm[13615]:             throw e;
Jan 14 20:54:07 raspberrypi3 npm[13615]:             ^
Jan 14 20:54:07 raspberrypi3 npm[13615]: TypeError: "value" argument is out of bounds
Jan 14 20:54:07 raspberrypi3 npm[13615]:     at checkInt (buffer.js:1185:11)
Jan 14 20:54:07 raspberrypi3 npm[13615]:     at Buffer.writeUInt16LE (buffer.js:1243:5)
Jan 14 20:54:07 raspberrypi3 npm[13615]:     at Concentrate.write_number (/opt/zigbee2mqtt/node_modules/concentrate/index.js:63:21)
Jan 14 20:54:07 raspberrypi3 npm[13615]:     at Concentrate.<anonymous> (/opt/zigbee2mqtt/node_modules/concentrate/index.js:42:29)
Jan 14 20:54:07 raspberrypi3 npm[13615]:     at Array.forEach (<anonymous>)
Jan 14 20:54:07 raspberrypi3 npm[13615]:     at Concentrate.result (/opt/zigbee2mqtt/node_modules/concentrate/index.js:38:13)
Jan 14 20:54:07 raspberrypi3 npm[13615]:     at FoundPayload.frame (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:117:20)
Jan 14 20:54:07 raspberrypi3 npm[13615]:     at Object.zcl.frame (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/zcl.js:66:78)
Jan 14 20:54:07 raspberrypi3 npm[13615]:     at Object.af.zclFoundation (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/af.js:263:25)
Jan 14 20:54:07 raspberrypi3 npm[13615]:     at ZShepherd._foundation (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:660:20)
Jan 14 20:54:07 raspberrypi3 npm[13615]: npm ERR! code ELIFECYCLE
Jan 14 20:54:07 raspberrypi3 npm[13615]: npm ERR! errno 1
Jan 14 20:54:07 raspberrypi3 npm[13615]: npm ERR! zigbee2mqtt@1.0.1 start: `node index.js`
Jan 14 20:54:07 raspberrypi3 npm[13615]: npm ERR! Exit status 1
Jan 14 20:54:07 raspberrypi3 npm[13615]: npm ERR!
Jan 14 20:54:07 raspberrypi3 npm[13615]: npm ERR! Failed at the zigbee2mqtt@1.0.1 start script.
Jan 14 20:54:07 raspberrypi3 npm[13615]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Jan 14 20:54:07 raspberrypi3 npm[13615]: npm ERR! A complete log of this run can be found in:
Jan 14 20:54:07 raspberrypi3 systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Jan 14 20:54:07 raspberrypi3 systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Jan 14 20:54:07 raspberrypi3 systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jan 14 20:54:08 raspberrypi3 systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Jan 14 20:54:08 raspberrypi3 systemd[1]: Stopped zigbee2mqtt ZigBee2MQTT Bridge.
:(((

Koenkk, 2019-01-15:
Should be fixed in the latest dev branch

antti365, 2019-01-16:
Well, yes, now there are no errors at start-up and configuration, BUT there is also no battery level ... :)
Jan 16 14:22:49 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:49 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-01-16.14-22-49'
Jan 16 14:22:50 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:50 Starting zigbee2mqtt version 1.0.1 (commit #28e3288)
Jan 16 14:22:50 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:50 Starting zigbee-shepherd
Jan 16 14:22:51 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:51 zigbee-shepherd started
Jan 16 14:22:51 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:51 Coordinator firmware version: '2018090201'
Jan 16 14:22:51 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:51 Currently 1 devices are joined:
Jan 16 14:22:51 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:51 0x000b57fffe2b3c08 (0x000b57fffe2b3c08): ICTC-G-1 - IKEA TRADFRI wireless dimmer (EndDevice)
Jan 16 14:22:51 raspberrypi3 npm[21194]:   zigbee2mqtt:warn 2019-1-16 14:22:51 `permit_join` set to  `true` in configuration.yaml.
Jan 16 14:22:51 raspberrypi3 npm[21194]:   zigbee2mqtt:warn 2019-1-16 14:22:51 Allowing new devices to join.
Jan 16 14:22:51 raspberrypi3 npm[21194]:   zigbee2mqtt:warn 2019-1-16 14:22:51 Set `permit_join` to `false` once you joined all devices.
Jan 16 14:22:51 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:51 Zigbee: allowing new devices to join.
Jan 16 14:22:51 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:51 Connecting to MQTT server at mqtt://localhost
Jan 16 14:22:51 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:51 zigbee-shepherd ready
Jan 16 14:22:51 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:51 Connected to MQTT server
Jan 16 14:22:51 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:51 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Jan 16 14:22:51 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:51 MQTT publish: topic 'zigbee2mqtt/0x000b57fffe2b3c08', payload '{"brightness":0,"linkquality":55,"last_seen":"2019-01-16T12:22:51.939Z"}'
Jan 16 14:22:52 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:52 MQTT publish: topic 'zigbee2mqtt/0x000b57fffe2b3c08', payload '{"brightness":255,"linkquality":55,"last_seen":"2019-01-16T12:22:52.355Z"}'
Jan 16 14:22:53 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:53 MQTT publish: topic 'zigbee2mqtt/0x000b57fffe2b3c08', payload '{"brightness":255,"linkquality":55,"last_seen":"2019-01-16T12:22:53.017Z"}'
Jan 16 14:22:53 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:53 MQTT publish: topic 'zigbee2mqtt/0x000b57fffe2b3c08', payload '{"brightness":255,"linkquality":55,"last_seen":"2019-01-16T12:22:53.888Z"}'
Jan 16 14:22:54 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:54 MQTT publish: topic 'zigbee2mqtt/0x000b57fffe2b3c08', payload '{"brightness":0,"linkquality":55,"last_seen":"2019-01-16T12:22:54.377Z"}'
Jan 16 14:22:54 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:54 Succesfully configured 0x000b57fffe2b3c08 (0x000b57fffe2b3c08)
Jan 16 14:22:55 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:22:55 MQTT publish: topic 'zigbee2mqtt/0x000b57fffe2b3c08', payload '{"brightness":0,"linkquality":55,"last_seen":"2019-01-16T12:22:55.229Z"}'
Jan 16 14:23:01 raspberrypi3 npm[21194]:   zigbee2mqtt:info 2019-1-16 14:23:01 MQTT publish: topic 'zigbee2mqtt/0x000b57fffe2b3c08', payload '{"brightness":0,"linkquality":55,"last_seen":"2019-01-16T12:23:01.802Z"}'

Koenkk, 2019-01-16:
@antti365 it can take up to one day for the device to report it's battery status.

antti365, 2019-01-16:
Great, then we will test! Thank you so much for your work! I will definitely report the results. 👍

GSanCap, 2019-01-16:
Some Battery Powered Devices take some time to report the Battery level...patience ;)

antti365, 2019-01-16:
I try, but patience is not enough ... :) Therefore, a few questions:
1. In what form should the answer come - together with the level of dimming or separately? That is, to get it, is it necessary to periodically turn the dimmer?
2. In the database.db file I found the following value in the dimmer properties, if I understand correctly, should something like this come? ("batteryPercentageRemaining": 60)
3. Does the receipt of a message depend on the battery charge? Maybe it will not come if the level is 60%?

Koenkk, 2019-01-16:
1. Seperatly, no need to turn the dimmer.
2. Yes, that's the message that will reported by the device, zigbee2mqtt converts it to `{"battery": 60}`
3. AFAIK not

antti365, 2019-01-17:
Changed the specified file and restarted zigbee2mqtt, waiting for 3 hours now, I understand correctly?

antti365, 2019-01-17:
Yeah, it turns out 3 seconds ... :) everything comes now !!!
Jan 17 09:38:47 raspberrypi3 npm[7926]:   zigbee2mqtt:info 2019-1-17 09:38:47 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":0,"last_seen":"2019-01-17T07:38:47.376Z","battery":60}'

antti365, 2019-01-17:
Should I now return the modified value back to MAX?

Koenkk, 2019-01-17:
Yes, can you monitor if you also receive updates then?

antti365, 2019-01-19:
Set the value in MAX, restarted zigbee2mqtt and waited ... This is what we have in the log file at the moment:

2019-1-17 13:30:09 - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-01-17.13-30-09'
2019-1-17 13:30:10 - info: Starting zigbee2mqtt version 1.0.1 (commit #28e3288)
2019-1-17 13:30:10 - info: Starting zigbee-shepherd
2019-1-17 13:30:11 - info: zigbee-shepherd started
2019-1-17 13:30:11 - info: Coordinator firmware version: '2018090201'
2019-1-17 13:30:11 - info: Currently 1 devices are joined:
2019-1-17 13:30:11 - info: Entrance_Dimmer (0x000b57fffe2b3c08): ICTC-G-1 - IKEA TRADFRI wireless dimmer (EndDevice)
2019-1-17 13:30:11 - warn: `permit_join` set to  `true` in configuration.yaml.
2019-1-17 13:30:11 - warn: Allowing new devices to join.
2019-1-17 13:30:11 - warn: Set `permit_join` to `false` once you joined all devices.
2019-1-17 13:30:11 - info: Zigbee: allowing new devices to join.
2019-1-17 13:30:11 - info: Connecting to MQTT server at mqtt://localhost
2019-1-17 13:30:12 - info: zigbee-shepherd ready
2019-1-17 13:30:12 - info: Connected to MQTT server
2019-1-17 13:30:12 - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
2019-1-17 13:30:12 - info: MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":5,"linkquality":0,"last_seen":"2019-01-17T11:27:28.349Z","battery":60}'
2019-1-17 13:30:29 - info: MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":255,"linkquality":84,"last_seen":"2019-01-17T11:30:29.982Z","battery":60}'
2019-1-17 13:30:30 - info: MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":84,"last_seen":"2019-01-17T11:30:30.529Z","battery":60}'
2019-1-17 13:30:34 - info: MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":52,"last_seen":"2019-01-17T11:30:34.289Z","battery":60}'
2019-1-17 13:30:37 - info: Succesfully configured Entrance_Dimmer (0x000b57fffe2b3c08)
2019-1-18 06:52:26 - info: Device incoming...
2019-1-18 06:52:26 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'

More than two days have passed, it seems to me that MAX does not work as it should ... Although I will continue monitoring, should he (dimmer) ever answer ... Or are there other ideas?

Koenkk, 2019-01-19:
I guess the value is too big, could you try with `maxRepIntval: 3600` (one hour)

antti365, 2019-01-19:
Yes, now I will change the value to 3600. But as others think, what is the optimal time to monitor the battery? Maybe once a day will be enough?

Koenkk, 2019-01-19:
@antti365 I agree, but this doesn't seem to work (MAX value = 18 hours)

Koenkk, 2019-01-19:
@antti365 that shouldn't make a difference (in the end the value is the same).

antti365, 2019-01-19:
With a value of 3600 there are no problems, everything works. The only thing that time for some reason is shifted by 2 minutes each time.
Jan 19 15:15:11 raspberrypi3 npm[23678]:   zigbee2mqtt:info 2019-1-19 15:15:11 Succesfully configured Entrance_Dimmer (0x000b57fffe2b3c08)
Jan 19 15:15:21 raspberrypi3 npm[23678]:   zigbee2mqtt:info 2019-1-19 15:15:21 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":6,"linkquality":92,"last_seen":"2019-01-19T13:15:21.573Z","battery":60}'
Jan 19 15:15:21 raspberrypi3 npm[23678]:   zigbee2mqtt:info 2019-1-19 15:15:21 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":8,"linkquality":92,"last_seen":"2019-01-19T13:15:21.861Z","battery":60}'
Jan 19 15:15:33 raspberrypi3 npm[23678]:   zigbee2mqtt:info 2019-1-19 15:15:33 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":92,"last_seen":"2019-01-19T13:15:33.842Z","battery":60}'
Jan 19 16:17:09 raspberrypi3 npm[23678]:   zigbee2mqtt:info 2019-1-19 16:17:09 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":99,"last_seen":"2019-01-19T14:17:09.248Z","battery":60}'
Jan 19 17:19:07 raspberrypi3 npm[23678]:   zigbee2mqtt:info 2019-1-19 17:19:07 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":99,"last_seen":"2019-01-19T15:19:07.230Z","battery":60}'
Jan 19 18:21:05 raspberrypi3 npm[23678]:   zigbee2mqtt:info 2019-1-19 18:21:05 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":99,"last_seen":"2019-01-19T16:21:05.256Z","battery":60}'
Jan 19 19:23:03 raspberrypi3 npm[23678]:   zigbee2mqtt:info 2019-1-19 19:23:03 MQTT publish: topic 'zigbee2mqtt/Entrance_Dimmer', payload '{"brightness":0,"linkquality":99,"last_seen":2019-01-19T17:23:03.336Z","battery":60}'

So what do we do next?

Koenkk, 2019-01-19:
Can you try to find the mas value?

antti365, 2019-01-19:
Of course, went to find... :)

antti365, 2019-01-24:
Maybe you have something with the battery? I've been testing for more than a week, this has never happened before ....

Bruceforce, 2019-01-24:
It's the battery that was delivered in the package and it's only a few days old. I think maybe the 3 second delay caused some confusion?

Have you already found a working maximum value @antti365

antti365, 2019-01-25:
Still finding, the fact is that you have to wait for quite large intervals, at the moment exactly the working interval is 15 hours (54000 seconds) and definitely NOT working - 17 hours 61200), I think that soon I will find this magic number. .. :)

Koenkk, 2019-01-25:
@antti365 I think that is close enough, would you mind making a PR to update this value? (https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L8)

antti365, 2019-01-25:
@Koenkk With pleasure, if you tell me how to do it right ... :)

Koenkk, 2019-01-25:
Click the pencil/edit icon on the file and than create a pull request: https://help.github.com/articles/creating-a-pull-request/

RaveGun, 2019-02-14:
Hello,
I've hooked up the a dimmer to my HA a couple of weeks ago and it worked fine at the beginning, but I did not use it for anything else, it just sat there.

Three days ago I tried the dimmer but there was no update.
So, I switched the battery and it worked fine again. I said that maybe the default IKEA battery was weak or of a poor quality.

Yesterday, I moved my CC2531 coordinator to my HA RPi where I set up the Edge add-on.
Now I just so that I am getting the battery status but it looks like this:
[image]

This looks like the battery has drained.
Are the pinging issues with the battery powered devices still active on the Edge add-on?

Thank you.

Koenkk, 2019-02-14:
Closing this issue as the battery information is now provided.

@RaveGun battery powered devices are not pinged

closed 2019-02-14 by Koenkk

gbilic, 2019-11-14:
@RaveGun, I experience the same, the battery seems to be drained when the device is paired with zigbee2mqtt.