Koenkk/zigbee2mqtt #794: About "connect P04, P06 and P20 to GND"

fred-c1, 2019-01-02:
This is not an issue but more a FYI or TBC/To Be Confirmed

In very short: it seems that with the lastest firmware(s???), "connect P04, P06 and P20 to GND" is not needed anymore.

For memo, "connect P04, P06 and P20 to GND" comes from this page:
https://koenkk.github.io/zigbee2mqtt/information/cc_sniffer_devices.html
in "CC2530 with RFX2401" section

******************************** This is how I came to the conclusion that "connect P04, P06 and P20 to GND" is not needed anymore.

Hardware:
I don't have a GBAN unit
I have ZB600+XCore2530. Like the GBAN unit, this hardware is based on CC2530+RFX2401
On ZB600 I removed all the jumpers except UART RX and TX
Therefore the CC2530 P04, P06 and P20 pins are floating

Firmware:
https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2530_CC2591/bin/CC2530_CC2591ZNP-Prod_20181123.zip

I have done several reboot and power cycling => good startup and operation of Zigbee2MQTT

******************************** Below is an attempt to clarify the original "connect P04, P06 and P20 to GND" requirement

The orginal firmware of "https://github.com/Koenkk/Z-Stack-firmware" is ti.com ZNP provided in ti.com Z-stack.
This original firmware is designed for ti.com hardware: SmartRF05EB + CC2530EM

On ti.com board:
P0.4 = EM_UART_CTS    = serial port "Clear To Send" (CC2530 input)
P0.5 = EM_UART_RTS    = serial port "Request To Send" (CC2530 output)
(P0.6 = EM_JOY_LEVEL   = "Voltage level from joystick, indicating position")
P2.0 = EM_JOY_MOVE   =  "Positive edge when the joystick is moved"

Back to "connect P04, P06 and P20 to GND":

About "P06": I believe that this is a typo and should be P05
In fact many post refer to P05 instead of P06

About "P04" and "P05"
I believe that you don't need to connect these pins to GND anymore because
RTS/CTS is disabled in the new CC2530 firmware
+
RTS/CTS is disabled in configuration.yaml (rtscts: false)

About "P20"
I don't why apparently, you don't need to connect it to GND anaymore

********************************
wish 1: maybe someone else could confirm that "connect P04, P06 and P20 to GND" is not needed anymore.

wish 2: it would be great to have a reliable method to upgrade (/replace) the firmware (/coordinator) without having to repair all the devices. My attempt to do this failed (I lost some devices).

Happy new year !

MikeVRT, 2019-01-04:
> maybe someone else could confirm that "connect P04, P06 and P20 to GND" is not needed anymore
on my CC2530 its NEEDED. without that - didn't not work.

ykreal, 2019-01-18:
Thanks for the information.

Hardware:
CC2530 with RFX2401(GBAN unit) didn't connect P04, P06 and P20 to GND

configuration.yaml:
rtscts: false

Firmware:
coordinator: CC2530_CC2591ZNP-Prod_20181123.zip (work)
router: CC2530_CC2591_router_2018_09.zip (work)

When startup Zigbee2MQTT everytime, I got a warning message.
But I think it's okay.
`warn: No converter available for 'CC2530.ROUTER' with cid 'genOnOff', type 'devChange' and data '{"cid":"genOnOff","data":{"onOff":0}}'`

desertspider, 2019-01-27:
I can confirm no soldering is needed anymore.

enboig, 2019-02-04:
I am flashing CC2530 with RFX2401 as a coordinator, but it don't work. Flashing appear to be ok:
```
~/src/cc-tool-gban$ sudo ./cc-tool -v -e -w  CC2530ZNP-Prod.hex 
  Programmer: SmartRF04DD
  Target: CC2530
  Erasing flash...
  Completed       
  Writing flash (241 KB)...
  Completed (8.35 s.)
  Verifying flash...
  Completed (3.26 s.)
```
and it is detected 
```
[ 2082.847883] usb 1-5: new full-speed USB device number 18 using xhci_hcd
[ 2082.997889] usb 1-5: New USB device found, idVendor=10c4, idProduct=ea60
[ 2082.997895] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2082.997900] usb 1-5: Product: CP2102 USB to UART Bridge Controller
[ 2082.997904] usb 1-5: Manufacturer: Silicon Labs
[ 2082.997907] usb 1-5: SerialNumber: 0001
[ 2082.999079] cp210x 1-5:1.0: cp210x converter detected
[ 2083.000862] usb 1-5: cp210x converter now attached to ttyUSB0
```
But fails to start:
```
  zigbee2mqtt:info 2019-2-4 23:47:20 Logging to directory: '/home/user/src/zigbee2mqtt/data/log/2019-02-04.23-47-20'
  zigbee2mqtt:info 2019-2-4 23:47:20 Starting zigbee2mqtt version 1.1.1 (commit #4fbb666)
  zigbee2mqtt:info 2019-2-4 23:47:20 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-2-4 23:47:23 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  zigbee2mqtt:info 2019-2-4 23:48:23 Starting zigbee-shepherd
  zigbee2mqtt:error 2019-2-4 23:48:26 Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2019-2-4 23:48:26 Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2019-2-4 23:48:26 Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/home/user/src/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:277:13)\n    at CcZnp.emit (events.js:189:13)\n    at Timeout.<anonymous> (/home/user/src/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
```
I tried with ```rtscts: false```
Am I missin something? I am using firmware CC2530_CC2591ZNP-Prod_20190109.zip

avanc, 2019-03-25:
If this is confirmed, the documentation should be updated:
https://www.zigbee2mqtt.io/information/cc_sniffer_devices.html

fred-c1, 2019-04-06:
@Koenkk , can you help to update the docuentation?
https://www.zigbee2mqtt.io/information/cc_sniffer_devices.html

Koenkk, 2019-04-07:
@fred-c1  or @avanc could you make a pr to update https://github.com/Koenkk/zigbee2mqtt.io/blob/master/information/cc_sniffer_devices.md ?

fred-c1, 2019-04-10:
@Koenkk , I don't now much about github and PRs.
I made an attempt...      Please review.

Koenkk, 2019-04-11:
@fred-c1 is this the expected change? https://github.com/Koenkk/zigbee2mqtt.io/pull/61/files

fred-c1, 2019-04-11:
@Koenkk    yes this is correct

Koenkk, 2019-04-11:
Great, thanks!

closed 2019-06-10 by Koenkk

rsurgiewicz, 2019-07-23:
Hi,
Today I've received GBAN2530-S and flashed newest firmware 22530+
As docs and you Guys mentioned soldering with new firmware is obsolete.

Unfortunatelly it did not work with zigbee2mqtt the (I was getting
`zigbee2mqtt:info 7/23/2019, 8:35:49 PM Starting zigbee-shepherd
  zigbee2mqtt:error 7/23/2019, 8:35:56 PM Error while starting zigbee-shepherd! (Error: request timeout)
  zigbee2mqtt:error 7/23/2019, 8:35:56 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 7/23/2019, 8:35:56 PM Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:269:22)\n    at Object.onceWrapper (events.js:273:13)\n    at CcZnp.emit (events.js:182:13)\n    at Timeout.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:248:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
`
of course I set `rtscts: false`
After reading comment https://github.com/Koenkk/zigbee2mqtt/issues/794#issuecomment-460447872 
I've decided to dig in past commits to find 
[CC2530_CC2591ZNP-Prod_20190109.zip](https://github.com/Koenkk/Z-Stack-firmware/blob/56b8e4eb89482d0a3ec914cc4407b6c69c5a5595/coordinator/CC2530_CC2591/bin/CC2530_CC2591ZNP-Prod_20190109.zip) and flashed device.
It worked at a glance.

I don't know why but for me newest firmware (Jun 28) didn't work.

tcrichton, 2019-10-28:
FYI, flashing a GBAN2530-S from aliexpress using coordinator firmware without soldering worked just fine for me using cc-tool and this firmware https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2530_CC2591_DEFAULT_20190608.zip

I was surprised that when I replaced my CC2531 with this GBAN2530-S zigbee2mqtt picked up without re-pairing (just changing the serial device and rtscts

Thank you all for your hard work in testing and making this so simple,

Tristan.

MichaelKorn, 2020-04-05:
I can confirm tcrichton: I also flashed a GBA2530-S from aliexpress without soldering. Before I had a CC2531 and I only changed the "COM port name" in the ioBroker. I then had a few problems and had to restart the zigbee instance but re-pairing was not needed.
Overall, everything works. Only the range is worse than with my CC2531 (I soldered an antenna to the CC2531)!

epicurean70, 2020-04-29:
Sorry, But how did you flash the GB2530 S without soldering? Just plug it directly into the usb port of the PC?

fred-c1, 2020-04-29:
@epicurean70:   Some people just hold the GB2530 5-pin connector "in place" during the flashing process

tcrichton, 2020-04-30:
@epicurean70 I can confirm I held the header pins in place during the flashing process.