Koenkk/zigbee2mqtt #795: Müller Licht Tint

frauhottelmann, 2019-01-03:
Appaerently Müller Licht, a German comany, is now producing Zigbee bulbs and remotes:
https://www.mueller-licht.de/produktserien/tint/
[image]
Aldi (north starting on the 21st, south starting on the 28th) will carry them at the end of January. I will try to get a set and add support for it. The set is the one with the big remote, even though I would be more interested in the small one :D

Koenkk, 2019-01-03:
Aldi selling smart bulbs! Wondering if they will also do this in the Netherlands lol.

Chrischi-, 2019-01-04:
@frauhottelmann 
rumour has it that the set or the individual lamps will not be cheaper than the Hue or similar. In my opinion the Hue lamps are always preferable in such a case (more expensive or equally expensive).

d33mha, 2019-01-04:
From the official press release:

> - tint Smart Light Starter Set inklusive Fernbedienung, LED-Birne 
> oder LED-Reflektor, 29,99 Euro
> - tint Smart Light Erweiterungsset weiß, LED-Birne, LED-Kerze oder
> LED-Reflektor, 14,99 Euro
> - tint Smart Light Erweiterungsset color, LED-Birne, LED-Kerze 
> oder LED-Reflektor, 19,99 Euro

frauhottelmann, 2019-01-04:
They are already on amazon Germany: https://www.amazon.de/stores/page/F5C0BCAB-8D9B-4E31-9E57-849B988C39C5?ingress=2&visitId=c20ad902-ec8a-49f8-bc6b-b673294a9cbb&ref_=bl_dp_s_web_7089465031
Prices are OK, but not great compared to the other possibilities like Hue or IKEA. And they are all RA=80, which isn't alle that great either.

Koenkk, 2019-01-04:
IKEA TRADFRI bulbs indeed seem to be a better choiche. To be honest, I dont see why you would buy this.

d33mha, 2019-01-04:
> And they are all RA=80, which isn't alle that great either.

That's the reason why I am still using old fashioned lightbulbs or halogen with smart plugs in my house.

Chrischi-, 2019-01-04:
@frauhottelmann
get this: Philips Hue White Wireless Dimming Kit, E27 LED Lampe inkl. Dimmschalter, dimmbar, warmweißes Licht, steuerbar via App, kompatibel mit Amazon Alexa (Echo, Echo Dot) https://www.amazon.de/dp/B0148NMVRE/ref=cm_sw_r_cp_api_i_K81lCbNAHFR4F

The kit is better than the Müller and sometimes even available for 19,99€. I bought the kit recently and sold the E27 led-bulb for 11€ on eBay KA.

ozbfat, 2019-01-05:
The Müller/Tint remote seems to have a lot of buttons which could be of use. Seems to use also Zigbee for communication... maybe worth a try?

The RGB bulb with 806  Lumen for 19,99€ sounds also quite cheap to me. Which alternatives are comparable?  Only confusing thing to me is that in specs it says 'no dimming'. Do they refer to some external dimming application or it doesn't have any internal dimming function?

Some specs:

Abmessung | 60x120 mm
-- | --
Nettogewicht | 116 g
Gehäusematerial | PBT weiß
Sockel | E27
Entsorgungspflichtig | ElektroG
Elektrische Daten |  
Betriebsspannung | 220-240 V
Nominelle Leistungsaufnahme | 9,5 W
Bemessungs-Leistungsaufnahme | 9,5 W
Nominelle Lebensdauer | 25.000 h
Bemessungs-Lebensdauer | 25.000 h
Netzfrequenz | 50/60Hz
Stromstärke | 55 mA
Stromart | AC
Energieeffizienzklasse | A+
Leistungsfaktor | 0,5
Gewichteter Energieverbrauch | 10kWh/1000h
max. Zündzeit | 0,5 s
Schaltzyklen bis Ausfall | 100.000
Vorzeitige Ausfallrate 1000h | 5%
Lebensdauerfaktor 6000h | 0,9
maximale Leerlaufleistung | 0,5 W
**Dimmbarkeit | nein**
Lichttechnische Daten |  
Nomineller Lichtstrom | 806 lm
Lichtstrom Maximalwert | 880 lm
Bemessungslichtstrom | 806 lm
Farbtemperatur | 2700
Lichtfarbe | white+color (1800-6500K+RGB)
Farbwiedergabeindex | Ra≥80
Farbkonsistenz | 6
Leistung Lichtstrom-Äq.Glühlampe | 60 W

zIEEE74, 2019-01-05:
Aldi sold them in Italy at the end of December, at an amazingly discounted price.
I bought 3 lamps: E14, E27 and GU10.
I dissected the E14 and created a Teardown Page on FB.
Processor is JN5169 from NXP, like in Xiaomi IOT.
Personally I prefer the standard Cortex M in Ikea lamps: far more pins exposed on the Pcb, but this is probably far better for sensors, since it can survive 2 years on cr2032 lithium batteries.

rtreffer, 2019-01-13:
The thing that caught my attentions was RGB E14 and GU10. See https://www.aldi-nord.de/angebote/angebote-mo-21-01/smart-light-erweiterungslampe-color-1000816.article.html

At 20€ each that sounds pretty interesting as there aren't many E14/GU10 zigbee RGB lamps AFAIK.

sehraf, 2019-01-19:
for the record, i get some E14 rgb bulbs.

andronaut42, 2019-01-26:
hi all,

I understand correctly, the devices should not be supported? - what a shame
But I would be interested, because I got three pieces as a gift.
I'm not a programmer, but maybe I can help to support them?
I've already tried the manual "How to support new devices", but I have not really succeeded.
Maybe someone likes to do it with me or help me with it?

regards Joachim

sehraf, 2019-01-26:
Not sure how you come to this conclusion, but let me cheer you up: they are already supported in the dev branch

andronaut42, 2019-01-26:
Sorry I interpreted that wrong
but great!
Thank you for the clarification

Carbon459, 2019-01-26:
Does the remote work, too?

sonntam, 2019-01-29:
How do you pair them? Do I have to use the remote or should I see activity in the log as soon as I turn a bulb on?

HobbyIngenieur, 2019-01-29:
> How do you pair them? Do I have to use the remote or should I see activity in the log as soon as I turn a bulb on?

It should pair automatically. My understanding is, that until earlier today the devices weren't really supported. Mine showed up as "ZBT-ExtendedColor" but werent usable at all. A few minutes ago I updated to the latest version of the bridge and now the bulb shows up as "404000/404005/404012" and accepts commands.

sonntam, 2019-01-30:
@HobbyIngenieur Thanks, it works now. I had to get the Zigbee stick closer to the lamp.

Have you found out how to activate the "scenes" that are available on the remot control, i.e. sunset, night, bonfire etc.?

ozbfat, 2019-01-30:
I can't pair it. I have to use zigbee2mqtt edge on HA and activate permit joint of course and then just switch on the bulb right? The bulb is just next to the zigbee stick and permit join is active but it is not being und discovered, what am I doing wrong?  I have the rgb E27 Version without the remote.

ozbfat, 2019-01-30:
Now I managed to connect but getting this failure :

zigbee2mqtt:info 2019-1-30 23:46:23 Zigbee publish to device '0x00158d00031d0fe1', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null zigbee2mqtt:error 2019-1-30 23:46:24 Zigbee publish to device '0x00158d00031d0fe1', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.

Diggen85, 2019-02-02:
Has anyone tried to get the Remote working?
It pairs as "ZBT-Remote-ALL-RGBW" but does not report anything. Maybe it behaves the same as the Ikea Tradfri remote? - Control Groups only

MoskitoHorst, 2019-02-11:
I also managed to connect the remote. But I get no data. Wath can I do, to Analyse?

Koenkk, 2019-02-12:
Can you sniff te traffic of the remote while pressing buttons on it? https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_sniff_zigbee_traffic.html

sander816, 2019-02-12:
i bought two e14 muller rgbw lights, but i have a problem with them:
one registers as `"ieeeAddr":"0xd0cf5efffef919e6","type":"Router","model":"404000/404005/404012","friendly_name":"0xd0cf5efffef919e6"`
But then in domoticz, i cannot find it in the devices list

the other one i cannot get registered at all: when i switch it on it colorcycles and starts in white mode, that indicates coupling mode, but i am not receiving it anymore.  It used to work earlier this week

MoskitoHorst, 2019-02-12:
It was hard work for me, but here is the logfile: I pressed the following keys (the numbers are the lines in the log window);
130 On/Off
148 Color wheel
179 warm
189 cold
200 dark
210 light
216 Szene 1-6
273 main switch
[image]

[tintRemote.zip](https://github.com/Koenkk/zigbee2mqtt/files/2857556/tintRemote.zip)

I hope, it is what you Need.

Koenkk, 2019-02-13:
It's the same as the [TRADFRI 5 button remote control](http://www.zigbee2mqtt.io/getting_started/pairing_devices.html#ikea-tradfri-remote-control-e1524). In your case the group is `0x4004`. 

You should add the device here: https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/coordinatorGroup.js#L8 and then add `coordinator_group: 0x4004` for this device.

behave49, 2019-02-13:
I got them to work on my domiticz. On/off is going oke. But when I change color a couple of times they stop responding. I have to reboot my pi to get them working again. 
Any idea's

MoskitoHorst, 2019-02-14:
I just created pull requests for the first steps.
Here on-off and brightness are working.

behave49, 2019-02-14:
on/off/brightness works well. Till i change the colors. After 4 or 5 times it stops working.
I used to know the command to see what the mqttbroker is doing but lost my comand list ;-(

sander816, 2019-02-14:
How did you get it working in domoticz?
I cannot find them in my device list

Maximus48p, 2019-02-14:
Able to pair a bulb in Hass (44062) but not the remote. Any suggestions how? @MoskitoHorst maybe?

behave49, 2019-02-14:
> 
> 
> How did you get it working in domoticz?
> I cannot find them in my device list

Make sure you got a working broker and the zigbee2mqtt in domiticz, pair de bulp ( just turn the light on/off couple of times) and then you should see it in your device's

MoskitoHorst, 2019-02-14:
For the bulb, you have to switch them on and off 5 times, 1 second per cycle (About).
For Pairing the remote: I did pull request for a few files. And then i pressed the button above the batteries About 10 to 20 seconds. I tried several times. And I restartet zigbee2mqtt after this. Then it was paired.

After this, I had to enter the groupID, same method as for IKEA. Id was 0x4004 in my case, do not know, if this is constant. And the 3 Levels (lower Button on remote with the 3 bulbs on it) seem to have different IS's

sander816, 2019-02-14:
That's what i did, zigbee2mqtt, mosquitto, and domoticz mqtt client
I see it in domoticz log, along with other devices, but i dont see them in devices

All other zigbee works fine

behave49, 2019-02-14:
push the green arrow then they show up in youre device list

sander816, 2019-02-14:
i know, i cannot push the green arrow, because they are not visible in my domoticz devices

sander816, 2019-02-14:
i see this in domoticz:
```
2019-02-14 15:49:56.092 (zigbee network) Zigbee2mqtt bridge is online 
2019-02-14 15:49:56.092 (zigbee network) Joining new devices is enabled on the zigbee bridge 
2019-02-14 15:49:56.144 (zigbee network) Received available devices list from bridge 
2019-02-14 15:49:56.144 (zigbee network) Device RTCGQ11LM 0x00158d00029bff93 (0x00158d00029bff93) 
2019-02-14 15:49:56.144 (zigbee network) Device MCCGQ11LM 0x00158d000252a53d (0x00158d000252a53d) 
2019-02-14 15:49:56.144 (zigbee network) Device MFKZQ01LM 0x00158d000278e797 (0x00158d000278e797) 
2019-02-14 15:49:56.144 (zigbee network) Device MCCGQ01LM 0x00158d00028b6b2f (0x00158d00028b6b2f) 
2019-02-14 15:49:56.145 (zigbee network) Device 404000/404005/404012 0xd0cf5efffeebda06 (0xd0cf5efffeebda06) 
2019-02-14 15:49:56.145 (zigbee network) Device 404000/404005/404012 0xd0cf5efffef919e6 (0xd0cf5efffef919e6) 
2019-02-14 15:49:56.145 (zigbee network) Device ZNCZ02LM 0x00158d000268d9e6 (0x00158d000268d9e6) 
2019-02-14 15:49:56.145 (zigbee network) Device JTYJ-GD-01LM/BW 0x00158d0002ad12a0 (0x00158d0002ad12a0) 
```

but these two are not in mydevices:
`Device 404000/404005/404012 `

MoskitoHorst, 2019-02-14:
Even after many tries, I only get reactions/ logfile entries for 3 Buttons: On/Off and brightness + and brightness -.
Does anyone have a hint?

behave49, 2019-02-16:
Does anyone know the command when you logged in ssh to show the activity from the broker? I lost my command list ;-(

behave49, 2019-02-16:
> i see this in domoticz:
>
> ```
> 2019-02-14 15:49:56.092 (zigbee network) Zigbee2mqtt bridge is online
> 2019-02-14 15:49:56.092 (zigbee network) Joining new devices is enabled on the zigbee bridge [...]

That is your'e logfile. Look in youre device list , not hardware but device

[image]

sander816, 2019-02-16:
I know, they are NOT in my devices.
I see them in the log, but not in the devices list. I see all my Xiaomi devices in the devices list, but not the Muller lights

sander816, 2019-02-16:
I have no 404 devices in my list
When i filter in 404, there are no devices

rbrose, 2019-03-07:
I can pair my ZBT-ExtendedColor without problems. But if is change any paramter the lamp going off.
I try set state to true or set Brightness or set color ... always the lamp going simple off. anybody an idea?

Maximus48p, 2019-03-07:
[https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1209](url)

closed 2019-05-13 by stale[bot]

BNBaron, 2019-08-12:
Hi everyone, 
I've recently purchased one of the E14 tint bulbs, thinking I could pair it to my existing Ikea Tradfri system. I was able to pair the bulb with the hockeypuck remote, but it would give an error in the app. So now I'm able to control the bulb using the remote, but not using the app. 
Does anyone now a fix?

Regards

RubenKelevra, 2022-05-30:
Just want to let you guys know, there's a newer generation available at Aldi. I bought a couple of units. They seem to work, but sometimes not correctly identified by Zigbee2MQTT (issues are created).

They upped to RA >90 so are now equal to Ikea Tradfri.

Prices are higher, but they usually don't get sold right away and may become cheaper when they lie around for a couple of weeks. :)

JelleKoster, 2023-12-28:
@RubenKelevra Thanks for sharing, I have the same pairing issue with their switch. Which issue are you referring to? I can then continue there.

RubenKelevra, 2023-12-28:
Hey @JelleKoster,

I think there's a misunderstanding. I haven't had any issue using them.

Just the human readable label, like "Model number: xyz" was wrong, due to the manufacturer using the same internal ID for more than one model of lamps.

I bought like 25 bulbs by them in the end. Sadly looks like I got a bad batch and they nearly all failed. They lost Zigbee connection and after power cycle they show the rainbow pattern of pairing - but won't actually pair.

The company however have sent me replacements for all failed bulbs - no questions asked. And the new bulbs seems to work fine so far.

If you got a device which is not yet working with Zigbee2MQTT, please create a new ticket, explaining what issue you're experiencing.

The goal is to support all features of all Zigbee devices. :)

Here's the general howto:

https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Best regards

Ruben