Koenkk/zigbee2mqtt #798: WSDCGQ11LM Sanitized/ignored values

jeromelaban, 2019-01-03:
Using release 1.0.1, and a `Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)`, I'm sometimes getting this kind of values: 

```
 zigbee2mqtt:info 2019-1-3 15:35:45 MQTT publish: topic
'zigbee2mqtt/0x00158d00XXXXXXXX', 
payload '{
"temperature":-100,
"linkquality":60,
"humidity":90.62,
"pressure":1017.63,
"battery":13,
"voltage":2875,
"device":{"ieeeAddr":"0x00158d00XXXXXXXX'",
"friendlyName":"0x00158d00XXXXXXXX'","type":"EndDevice","nwkAddr":XXXXX,
"manufId":4151,"manufName":"LUMI","powerSource":"Battery",
"modelId":"lumi.weather","status":"online"}}'
```

This kind of value is obviously incorrect, and not in the range of the possible sensor values. Is there a way to ignore values such as those ?

Thanks !

Koenkk, 2019-01-04:
Perhaps its a bug, could you provide the log when receiving such a value while running with `log_level: debug`?

aep, 2019-01-06:
unsure if this is the same issue.  i'm getting a weird temperature range:  

{"temperature":29.91,"linkquality":120,"humidity":94.92}

  zigbee2mqtt:debug 1/6/2019, 1:30:06 PM Received zigbee message of type 'attReport' with data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2757}}' of device 'lumi.sens' (0x00158d000.......)

jeromelaban, 2019-01-06:
@Koenkk I still trying to get a debug log, but as always in those random cases, I'm have yet to get a repro log... :)

jeromelaban, 2019-01-07:
There it is, with an invalid humidity value as well:

```  
zigbee2mqtt:debug 2019-1-6 23:47:58 Received zigbee message of type 'attReport' with data '
{"cid":"msRelativeHumidity",
"data":{"measuredValue":65436}
}' of device 'lumi.weather' (0x00158dXXXXXXXXX)

zigbee2mqtt:debug 2019-1-6 23:48:18 Received zigbee message of type 'attReport' with data '
{"cid":"msTemperatureMeasurement",
"data":{"measuredValue":-10000}
}' of device 'lumi.weather' (0x00158dXXXXXXXXX)

zigbee2mqtt:info 2019-1-6 23:48:18 MQTT publish: topic 'zigbee2mqtt/0x00158dXXXXXXXXX', 
payload '{
"temperature":-100,
"linkquality":63,
"humidity":654.36,
"pressure":1030,
"battery":15,
"voltage":2905,
"device":{"ieeeAddr":"0x00158dXXXXXXXXX",
"friendlyName":"0x00158dXXXXXXXXX",
"type":"EndDevice",
"nwkAddr":XXXXX,
"manufId":4151,
"manufName":"LUMI",
"powerSource":"Battery",
"modelId":"lumi.weather",
"status":"online"
}}'
```

Koenkk, 2019-01-08:
Should be fixed in the dev branch.

lolorc, 2019-03-04:
I'm not that sure about that :)
```
Mar 04 03:47:49 zigbee2mqtt/aqara_06 {"battery":86,"voltage":2975,"temperature":-100,"humidity":654.36,"pressure":983.6,"linkquality":80,"last_seen":1551667669052,"elapsed":10413}        
Mar 04 03:47:59 zigbee2mqtt/aqara_06 {"temperature":19.32,"linkquality":80,"last_seen":1551667679874,"elapsed":10822}                                                               
Mar 04 03:47:59 zigbee2mqtt/aqara_06 {"humidity":51.45,"linkquality":80,"last_seen":1551667679885,"elapsed":11}                                                                     
Mar 04 03:47:59 zigbee2mqtt/aqara_06 {"pressure":983,"linkquality":80,"last_seen":1551667679900,"elapsed":15}
```

temperature: -100
humidity: 654.36

this is the 1st time it happens, I'm really curious to know if this is really a problem with bad report from the sensor or bad handling by some part of the z2m stack.
I've just upgraded to dev yesterday.

```
  serialport/stream binding.read finished +23ms                                                                                                                                                                                                                                                                                                                                      │
  cc-znp { sof: 254,                                                                                                                                                                                                                                                                                                                                                                 │
  cc-znp   len: 28,                                                                                                                                                                                                                                                                                                                                                                  │
  cc-znp   type: 'AREQ',                                                                                                                                                                                                                                                                                                                                                             │
  cc-znp   subsys: 'AF',                                                                                                                                                                                                                                                                                                                                                             │
  cc-znp   cmd: 'incomingMsg',                                                                                                                                                                                                                                                                                                                                                       │
  cc-znp   payload:                                                                                                                                                                                                                                                                                                                                                                  │
  cc-znp    { groupid: 0,                                                                                                                                                                                                                                                                                                                                                            │
  cc-znp      clusterid: 1026,                                                                                                                                                                                                                                                                                                                                                       │
  cc-znp      srcaddr: 17435,                                                                                                                                                                                                                                                                                                                                                        │
  cc-znp      srcendpoint: 1,                                                                                                                                                                                                                                                                                                                                                        │
  cc-znp      dstendpoint: 1,                                                                                                                                                                                                                                                                                                                                                        │
[... cut ...]
  zigbee2mqtt:debug 3/4/2019, 3:47:59 AM Received zigbee message of type 'devChange' with data '{"cid":"msRelativeHumidity","data":{"measuredValue":5145}}' of device 'lumi.weather' (aqara_06)                                                                                                                                                                                      │
  zigbee2mqtt:debug 3/4/2019, 3:47:59 AM Received zigbee message of type 'devChange' with data '{"cid":"msPressureMeasurement","data":{"16":9838}}' of device 'lumi.weather' (aqara_06)
```

Koenkk, 2019-03-04:
@lolorc your case should also be fixed now.

lolorc, 2019-03-04:
Thanks, still I'm wondering if this fixed or hidden :-)
In other words, is that something other zigbee gateway implementations do need as well ?

Koenkk, 2019-03-05:
@lolorc it's hidden, the only way to find out if something needs to be fixed in zigbee2mqtt/zigbee-shepherd is to capture a faulty zigbee message and check what temperature/humidity/pressure is in there. In this way we can check if the error is in the parsing by zigbee-shepherd or if it is actually reported by the sensor.

closed 2019-05-04 by Koenkk