Koenkk/zigbee2mqtt #802: Philips Hue motion sensor not supported

henriz, 2019-01-04:
Hello,

I have a working setup, and am trying to add our philips hue motion sensors to it.

They are on the list of supported devices.

However, If I bind it (which it does fine), I get this:
 zigbee2mqtt:info 1/4/2019, 5:48:03 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:warn 1/4/2019, 5:48:03 PM Message without device!
  zigbee2mqtt:info 1/4/2019, 5:48:03 PM Device incoming...
  zigbee2mqtt:info 1/4/2019, 5:48:03 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 1/4/2019, 5:48:03 PM New device with address 0x0017880102030d92 connected!
  zigbee2mqtt:info 1/4/2019, 5:48:03 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x0017880102030d92"}'
  zigbee2mqtt:info 1/4/2019, 5:48:18 PM MQTT publish: topic 'zigbee2mqtt/0x0017880102030d92', payload '{"battery":100,"linkquality":42,"last_seen":"2019-01-04T17:48:18.745Z"}'
  zigbee2mqtt:info 1/4/2019, 5:48:18 PM MQTT publish: topic 'zigbee2mqtt/0x0017880102030d92', payload '{"battery":100,"linkquality":39,"last_seen":"2019-01-04T17:48:18.899Z","occupancy":true}'
  zigbee2mqtt:info 1/4/2019, 5:48:20 PM MQTT publish: topic 'zigbee2mqtt/0x0017880102030d92', payload '{"battery":100,"linkquality":42,"last_seen":"2019-01-04T17:48:20.142Z","occupancy":true,"temperature":20.1}'
  zigbee2mqtt:info 1/4/2019, 5:48:20 PM Succesfully configured 0x0017880102030d92 0x0017880102030d92
  zigbee2mqtt:info 1/4/2019, 5:48:20 PM MQTT publish: topic 'zigbee2mqtt/0x0017880102030d92', payload '{"battery":100,"linkquality":42,"last_seen":"2019-01-04T17:48:20.431Z","occupancy":true,"temperature":20.1,"illuminance":7982}'
  zigbee2mqtt:warn 1/4/2019, 5:48:20 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 1/4/2019, 5:48:20 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html

The device in question is the philips motion sensor, it also has light and temperature sensors on-board.

I'm talking about this product:

https://www2.meethue.com/nl-nl/p/hue-bewegingssensor/8718696743171

On the back, modelid: 9290012607

Koenkk, 2019-01-04:
This happens when the device has not fully been paired, can you try repairing?

henriz, 2019-01-04:
I have no try re-pairing the device like 10 times, all with the same result:

` zigbee2mqtt:warn 1/4/2019, 6:42:06 PM Message without device!
  zigbee2mqtt:warn 1/4/2019, 6:42:07 PM Message without device!
  zigbee2mqtt:warn 1/4/2019, 6:42:07 PM Message without device!
  zigbee2mqtt:info 1/4/2019, 6:42:08 PM MQTT publish: topic 'zigbee2mqtt/0x001788010212678e', payload '{"battery":100,"linkquality":34,"last_seen":"2019-01-04T18:42:08.261Z","occupancy":true,"temperature":21.96,"illuminance":14539}'
  zigbee2mqtt:warn 1/4/2019, 6:42:08 PM Message without device!
  zigbee2mqtt:warn 1/4/2019, 6:42:08 PM Message without device!
  zigbee2mqtt:warn 1/4/2019, 6:42:09 PM Message without device!
  zigbee2mqtt:warn 1/4/2019, 6:42:09 PM Message without device!
  zigbee2mqtt:info 1/4/2019, 6:42:11 PM Succesfully configured 0x001788010212678e 0x001788010212678e
  zigbee2mqtt:warn 1/4/2019, 6:42:11 PM Message without device!
  zigbee2mqtt:warn 1/4/2019, 6:42:12 PM Message without device!
  zigbee2mqtt:info 1/4/2019, 6:42:13 PM MQTT publish: topic 'zigbee2mqtt/0x001788010212678e', payload '{"battery":100,"linkquality":39,"last_seen":"2019-01-04T18:42:13.162Z","occupancy":true,"temperature":21.96,"illuminance":14523}'
  zigbee2mqtt:warn 1/4/2019, 6:42:13 PM Message without device!
  zigbee2mqtt:warn 1/4/2019, 6:42:14 PM Message without device!
  zigbee2mqtt:warn 1/4/2019, 6:42:15 PM Message without device!
  zigbee2mqtt:warn 1/4/2019, 6:42:16 PM Message without device!
  zigbee2mqtt:warn 1/4/2019, 6:42:17 PM Message without device!
  zigbee2mqtt:info 1/4/2019, 6:42:17 PM Device incoming...
  zigbee2mqtt:info 1/4/2019, 6:42:17 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 1/4/2019, 6:42:18 PM MQTT publish: topic 'zigbee2mqtt/0x001788010212678e', payload '{"battery":100,"linkquality":36,"last_seen":"2019-01-04T18:42:18.050Z","occupancy":true,"temperature":21.96,"illuminance":14506}'
  zigbee2mqtt:info 1/4/2019, 6:42:20 PM MQTT publish: topic 'zigbee2mqtt/0x001788010212678e', payload '{"battery":100,"linkquality":39,"last_seen":"2019-01-04T18:42:20.371Z","occupancy":false,"temperature":21.96,"illuminance":14506}'
  zigbee2mqtt:info 1/4/2019, 6:42:20 PM MQTT publish: topic 'zigbee2mqtt/0x001788010212678e', payload '{"battery":100,"linkquality":36,"last_seen":"2019-01-04T18:42:20.426Z","occupancy":false,"temperature":21.96,"illuminance":14523}'
  zigbee2mqtt:warn 1/4/2019, 6:42:22 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 1/4/2019, 6:42:22 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:info 1/4/2019, 6:42:22 PM MQTT publish: topic 'zigbee2mqtt/0x001788010212678e', payload '{"battery":100,"linkquality":42,"last_seen":"2019-01-04T18:42:22.074Z","occupancy":false,"temperature":21.96,"illuminance":14523}'`

henriz, 2019-01-04:
For those interested, I added this sensor to the domoticz plugin for Zigbee2Mqqt and created a pullrequest, which has already been merged into main branch. In other words, the plugin now also support this sensor.

See:
https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/pull/59