Koenkk/zigbee2mqtt #803: Adding support for Q-INIGO, Neuhaus Lighting Group

seppi91, 2019-01-04:
Hi,
I just came across this project and tried to integrate some lights with my CC2531 based zigbee USB stick. I'm using Q-INIGO (based on zigbee protocol) ceiling lamps from [Paul Neuhaus](https://www.paul-neuhaus.de/shop/de/led-deckenleuchte-smart-home-alexa-tauglich-100-424-11.html). I followed the `How to support new devices` docs and tried to pair the light (router) with the usual pairing process (turn lights 5 times on/off) but had no success. I also tried pairing the remote control (end device) to the stick, but also resulted in no success.

Both pairing processes are logged by the zigbee2mqtt but obviously the pairing was never successful. I also added some configuration to my devices.json which hides the "unsupported device" logs but doesn't change the outcome. [devices.json](https://hastebin.com/qobizabifu.css)

I would be grateful for some hints on how to integrate these things. [logging output](https://hastebin.com/ocibemumoh.coffeescript) [database.db] (https://hastebin.com/aqewivasah.json)

Koenkk, 2019-01-06:
Looking at your `database.db` it seems that `NLG-CCT light ` paired successfully.

You should be able to turn it on/off with the following in `devices.js`.

```js
    {
        zigbeeModel: ['NLG-CCT light '],
        model: '100.424.11',
        vendor: 'Paul Neuhaus',
        description: 'Q-INIGO LED ceiling light',
        supports: generic.light_onoff_brightness_colortemp.supports,
        fromZigbee: generic.light_onoff_brightness_colortemp.fromZigbee,
        toZigbee: generic.light_onoff_brightness_colortemp.toZigbee,
    },
```

seppi91, 2019-01-08:
Hi Koenkk,
thank you I've updated my devices.js as you described above. The pairing seems to work - at least it says "device connected" in the log.
Nevertheless the device seems to not properly announce itself, as zigbee2mqtt is not able to fetch the device descriptor
`PM Cannot get the Node Descriptor of the Device: 0x00158d0001da03b0 (Error: Timed out after 10000 ms)`

I flushed the database.db and tried to repair the device with DEBUG=* option. Please finde the log here https://hastebin.com/awiyezifub.pl. Is there anything that I can do to mitigate this? Also zigbee2mqtt seems not to publish into the homeassistant mqtt topic, at least I cannot see any messages there when subscribing with my mqtt client.

Koenkk, 2019-01-09:
Home assistant integration requires a change in a different file, let's first try to control it.

Does it turn on/off when sending a message to `zigbee2mqtt/0x00158d0001da03b0/set` with payload `ON`?

seppi91, 2019-01-09:
Alirght, I sent a mqtt message as described, but then zigbee2mqtt crashes immediately. I posted the logs here https://hastebin.com/ayewuyucub.pl

database.db https://hastebin.com/xucedigone.json

Koenkk, 2019-01-10:
Can you try with:

```js
    {
        zigbeeModel: ['NLG-CCT light '],
        model: '100.424.11',
        vendor: 'Paul Neuhaus',
        description: 'Q-INIGO LED ceiling light',
        supports: generic.light_onoff_brightness_colortemp().supports,
        fromZigbee: generic.light_onoff_brightness_colortemp().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colortemp().toZigbee,
    },
```

Koenkk, 2019-01-11:
Don't worry about the home assistant part, it will work when it's officially supported by zigbee2mqtt.

For the remote, can you try:

```js
{
        zigbeeModel: ['NLG-remote '],
        model: 'todo',
        description: 'Paul Neuhaus remote',
        supports: '',
        vendor: 'Paul Neuhaus',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

Afterwards, repair the device, wait 10 seconds, press all buttons on the remote a few times and post the logging here.

seppi91, 2019-01-14:
Hi Koenkk, I adapted the remote configuration as you described and re-paired the device. Here are the logs, https://hastebin.com/uwuxuwojov.makefile 
Regards

Koenkk, 2019-01-15:
No messages are send by the remote yet, meaning it is a bit more complicated to add support for this device. You would need to find out what is send by the remote by doing some zigbee sniffing (https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/how_tos/how_to_sniff_zigbee_traffic.md)

msadeghz, 2019-01-20:
I have a same issue. I have a device (thermostat, HA1.2 profile) and see just "Cannot get the Node Descriptor of the Device: 0x02698dff00158d00 (Error: Timed out after 10000 ms)" message. 
is there any way to know the data transferred between zigbee2mqtt and ZNP on serial connection during pairing stage?

Koenkk, 2019-01-21:
@msadeghz yes, by running with `DEBUG=* npm start`

closed 2019-03-29 by stale[bot]

karia001, 2020-04-03:
@seppi91 how were you able to pair the q-inigo with the cc2531? i even cannot pair the device so i am still stuck on step 1:-(

seppi91, 2020-04-06:
Hi @karia001 sorry for the late reply. Yes I was able to pair the device. I just enabled the discovery mode by setting `permit_join: true` and then initiated the pairing by switching the light on-off multiple times. 
I did not change anything else. Maybe you may try to enable DEBUG logs and attach them here ...

karia001, 2020-04-06:
hi @seppi91  my light and cc2531 are 5 meters apart,  is this perhaps te reason why it's not pairing.

seppi91, 2020-04-06:
I dont think so. My distance is probably even bigger and theres a wall in
between.

karia001, 2020-04-06:
Hi @seppi91 
tried several times to join, but still unsuccessful
here is my log file, hope you can help me figure out what i am doing wrong. if i look in the file i don't see any messages form the q-inigo

info  2020-04-06 17:34:07: Logging to console and directory: '/share/zigbee2mqtt/log/2020-04-06.17-34-07' filename: log.txt
info  2020-04-06 17:34:09: Starting zigbee2mqtt version 1.12.0 (commit #unknown)
info  2020-04-06 17:34:09: Starting zigbee-herdsman...
info  2020-04-06 17:34:13: zigbee-herdsman started
info  2020-04-06 17:34:13: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
info  2020-04-06 17:34:13: Currently 5 devices are joined:
info  2020-04-06 17:34:13: 0x00158d0003038afb (0x00158d0003038afb): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
info  2020-04-06 17:34:13: 0x00158d000301a2e6 (0x00158d000301a2e6): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
info  2020-04-06 17:34:13: 0x00158d0003017dd2 (0x00158d0003017dd2): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
info  2020-04-06 17:34:13: 0x00158d0003017e5e (0x00158d0003017e5e): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
info  2020-04-06 17:34:13: 0x00158d00030ade0e (0x00158d00030ade0e): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
warn  2020-04-06 17:34:13: `permit_join` set to  `true` in configuration.yaml.
warn  2020-04-06 17:34:13: Allowing new devices to join.
warn  2020-04-06 17:34:13: Set `permit_join` to `false` once you joined all devices.
info  2020-04-06 17:34:13: Zigbee: allowing new devices to join.
info  2020-04-06 17:34:13: Connecting to MQTT server at mqtt://core-mosquitto
info  2020-04-06 17:34:15: Connected to MQTT server
info  2020-04-06 17:34:15: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2020-04-06 17:34:15: MQTT publish: topic 'zigbee2mqtt/0x00158d0003038afb', payload '{"battery":100,"voltage":3005,"linkquality":92,"contact":true}'
info  2020-04-06 17:34:16: MQTT publish: topic 'zigbee2mqtt/0x00158d000301a2e6', payload '{"battery":100,"voltage":3022,"linkquality":36}'
info  2020-04-06 17:34:16: MQTT publish: topic 'zigbee2mqtt/0x00158d0003017dd2', payload '{"battery":97,"voltage":2995,"linkquality":28,"occupancy":false}'
info  2020-04-06 17:34:16: MQTT publish: topic 'zigbee2mqtt/0x00158d0003017e5e', payload '{"battery":100,"voltage":3025,"linkquality":65,"occupancy":false}'
info  2020-04-06 17:34:16: MQTT publish: topic 'zigbee2mqtt/0x00158d00030ade0e', payload '{"battery":86,"voltage":2975,"linkquality":60}'
info  2020-04-06 17:34:16: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.12.0","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'
info  2020-04-06 17:34:16: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0003038afb/contact/config', payload '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","state_topic":"zigbee2mqtt/0x00158d0003038afb","json_attributes_topic":"zigbee2mqtt/0x00158d0003038afb","name":"0x00158d0003038afb_contact","unique_id":"0x00158d0003038afb_contact_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003038afb"],"name":"0x00158d0003038afb","sw_version":"Zigbee2mqtt 1.12.0","model":"MiJia door & window contact sensor (MCCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-04-06 17:34:16: MQTT publish: topic 'homeassistant/sensor/0x00158d0003038afb/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d0003038afb","json_attributes_topic":"zigbee2mqtt/0x00158d0003038afb","name":"0x00158d0003038afb_battery","unique_id":"0x00158d0003038afb_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003038afb"],"name":"0x00158d0003038afb","sw_version":"Zigbee2mqtt 1.12.0","model":"MiJia door & window contact sensor (MCCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-04-06 17:34:16: MQTT publish: topic 'homeassistant/sensor/0x00158d0003038afb/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d0003038afb","json_attributes_topic":"zigbee2mqtt/0x00158d0003038afb","name":"0x00158d0003038afb_linkquality","unique_id":"0x00158d0003038afb_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003038afb"],"name":"0x00158d0003038afb","sw_version":"Zigbee2mqtt 1.12.0","model":"MiJia door & window contact sensor (MCCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-04-06 17:34:16: MQTT publish: topic 'homeassistant/sensor/0x00158d000301a2e6/click/config', payload '{"icon":"mdi:toggle-switch","value_template":"{{ value_json.click }}","state_topic":"zigbee2mqtt/0x00158d000301a2e6","json_attributes_topic":"zigbee2mqtt/0x00158d000301a2e6","name":"0x00158d000301a2e6_click","unique_id":"0x00158d000301a2e6_click_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d000301a2e6"],"name":"0x00158d000301a2e6","sw_version":"Zigbee2mqtt 1.12.0","model":"MiJia wireless switch (WXKG01LM)","manufacturer":"X [... cut ...]
debug 2020-04-06 17:42:05: Received Zigbee message from '0x00158d0003017dd2', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
info  2020-04-06 17:42:05: MQTT publish: topic 'zigbee2mqtt/0x00158d0003017dd2', payload '{"battery":97,"voltage":2995,"linkquality":23,"occupancy":true}'
info  2020-04-06 17:43:35: MQTT publish: topic 'zigbee2mqtt/0x00158d0003017dd2', payload '{"battery":97,"voltage":2995,"linkquality":23,"occupancy":false}'
debug 2020-04-06 17:44:09: Saving state to file /share/zigbee2mqtt/state.json

seppi91, 2020-04-10:
Sorry @karia001 but I really don't know what's wrong. As far as I can see there are no messages arriving at your receiver. Either there is something wrong with your light or maybe it's some incompatibility with your CC2531.

What I would do in your case is
- start from a clean configuration (moving the state.json and database.db) into another directory and restart pairing process for your Q-INIGO Light (debug enabled)
- If no pairing process is initiated I would try to follow https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html and check if you can see some messages from the light. If so I would ask @Koenkk if he may help in interpreting the values. Maybe something changed in the Q-INIGO firmware which is causing troubles on your side.

karia001, 2020-04-18:
Hi @seppi91 i have my Q-INIGO light working now, i just needed to bring my CC2531 closer to the light (around 1 m was enough), its now in my HA en its working correctly. Only not able to work with the q-remote anymore but that's not a big issue.
Thank you for your help. 😄